INTI-CMNB/KiBot #580: exclude_filter: '_none' does not working for csv bom output

honzakutil, 2024-02-13:
I have following configuration:
```yaml
outputs:
  # Bill of Materials - in csv format
  - name: 'bom-csv'
    comment: "Bill of Materails - csv format"
    type: bom
    dir: 'outputs/variant_${board_name}${board_rev}_${act_variant}_prints${variant_prints}'
    options:
      output: '${board_name}${board_rev}_${act_variant}_prints${variant_prints}_BOM.%x'
      use_aux_axis_as_origin: true
      exclude_filter: '_none'
      ref_separator: ', '
      csv:
        hide_pcb_info: true
        hide_stats_info: true
        separator: ';'
      columns:
        - field: Quantity Per PCB
          name: Qty
        - field: References
          name: Reference(s)
        - Value
        - Footprint
        - Description
```
After running KiBot with this configuration, all components with JP* (JP201, JP202 ...) are missing in the output csv file. I believe, that filter `_mechanical` is turned on by default for this type of output. I hoped, that I can disable it by using `exclude_filter: '_none'`, but it seems not to be working. Or, maybe I misunderstood the documentation?

If I replace solderjumper symbol (Jumper:SolderJumper_2_Open) with ordinary resistor symbol (Device:R) it works as expeted - parts are listed in the output. Component symbol library/name seems to be the trigger for this. I also tried renaming JP201->R201 and remove content of component description field, but none of these two worked.

Is there any way, how to make KiBot to include the solderjumper parts in the bom output? Or is it a bug and can it be fixed?

I am using KiBot 1.6.3, running in docker desktop on windows. KiBot instalation checker protocol follows: 
[esc5-tx1b_prints01_info.txt](https://github.com/INTI-CMNB/KiBot/files/14268496/esc5-tx1b_prints01_info.txt)

set-soft, 2024-02-13:
Please post a small example showing the problem. Without it I can't check what is excluding the components. Also: run kibot using `-vvvv` to get information on when JP* is removed. You can also try using 1.6.4.

honzakutil, 2024-02-14:
Update to 1.6.4 did not solve the issue.

According to `kibot -vv` output filter `_mechanical` is used even if KiBot is explicitely told to use `_none` by `exclude_filter: '_none'` config entry. Interestingly, if I use `exclude_filter: '_only_smd'` everything works, as expected. Unfortunatelly, this is not solution for me, since in real project I have also THT components.

Here is part of output with `exclude_filter: '_none'` config entry:
```
- 'Pick and PLace - using bom type' (pnp-csv) [bom]
DEBUG:Output destination: /workdir/outputs/variant_jptest0_default_prints00 (kibot - kiplot.py:439)
DEBUG:Board loaded (kibot - kiplot.py:232)
DEBUG:Filters reset (kibot - fil_base.py:194)
DEBUG:Applying filter `_mechanical` to exclude (kibot - fil_base.py:187)
DEBUG:Excluding 'JP1': Field 'part' (SolderJumper_2_Open) matched 're.compile('solder.*jump', re.IGNORECASE)' (kibot - fil_generic.py:170)
DEBUG:Excluding 'JP3': Field 'part' (SolderJumper_2_Open) matched 're.compile('solder.*jump', re.IGNORECASE)' (kibot - fil_generic.py:170)
DEBUG:Excluding 'R2': Field 'part' (SolderJumper_2_Open) matched 're.compile('solder.*jump', re.IGNORECASE)' (kibot - fil_generic.py:170)
DEBUG:Applying filter `_kibom_dnf_Config` to fitted (kibot - fil_base.py:205)
DEBUG:Applying filter `_kibom_dnc_Config` to fixed (kibot - fil_base.py:215)
DEBUG:Applying KiBoM style variants `default` (kibot - var_kibom.py:106)
```

And here is the same part, but with `exclude_filter: '_only_smd'` config entry:
```
- 'Pick and PLace - using bom type' (pnp-csv) [bom]
DEBUG:Output destination: /workdir/outputs/variant_jptest0_default_prints00 (kibot - kiplot.py:439)
DEBUG:Board loaded (kibot - kiplot.py:232)
DEBUG:Filters reset (kibot - fil_base.py:194)
DEBUG:Applying filter `_only_smd` to exclude (kibot - fil_base.py:187)
DEBUG:Applying filter `_kibom_dnf_Config` to fitted (kibot - fil_base.py:205)
DEBUG:Applying filter `_kibom_dnc_Config` to fixed (kibot - fil_base.py:215)
DEBUG:Applying KiBoM style variants `default` (kibot - var_kibom.py:106)
```

Here are example project files, that can be used to replicate the problem. Parts *JP1*, *R2*, *JP3* are expected to be found in the bom output, when `exclude_filter: '_none'` config entry is used [jptest.zip](https://github.com/INTI-CMNB/KiBot/files/14276245/jptest.zip).

set-soft, 2024-02-14:
76b622d788360cd24de6da6bb70e45f00e65734f should fix it (with 1d8ec5264089e07553980fd4413dc662e18d06a4 fix of the fix ;-)

The *_none* worked inside a variant, but didn't work used directly.

Let me know if the *dev* works for you.

closed 2024-02-14 by set-soft

honzakutil, 2024-02-15:
I pulled the image ghcr.io/inti-cmnb/kicad7_auto:dev_1.6.5-1d8ec52_k7.0.10_d12.1 and it is working now. Thanks for fixing the issue.

Once the version 1.6.5 is released, I can pull the :latest and  will get the fix too, am I correct?