INTI-CMNB/KiBot #581: [HOW] to execute custom bom export script

OSHW-Rico, 2024-02-14:
**What do you want to achieve?**

For our production process we have some custom scripts which are normally executed via the schematics editor/tools/create BOM. Is it possible to do that from kibot? I did not see a "custom scripts" section or similar.

PS: Heads up the the creators and maintainers of this project! I love how the kicad community enhances and reuses the work of others to make a more pleasend experience for everyone involved!!!

set-soft, 2024-02-14:
I think this is more or less the same as #109 

So the answer is not. You can always run the scripts before or after KiBot.

What are these scripts doing that can't be done using the internal BoM?

OSHW-Rico, 2024-02-14:
Yeah I can for sure, but I would need to be in the correct python environment, things which are handled automatically when done from the UI I'd guess.

It's primary use is to generate the pick and place files. We have a database library where each part is a symbol for itself and carries some information for the assembly process. In the end it's just some boolean logic and a text export.

The workflow is back from kicad 5 where it was not possible to access symbol custom field values from the pcb, so it was necessary to to it from the schematic and lookup the position in the PCB. Maybe that has changed since.

Rico

⁣Get BlueMail for Android ​

set-soft, 2024-02-15:
With KiBot you can generate pick & place files from the BoM output, there are some examples to generate XYRS files.

closed 2024-03-04 by set-soft