INTI-CMNB/KiBot #585: [BUG] KiBot ignores 'on_board no' during drc

oliv3r, 2024-02-18:
When validating parity between schematic and pcb (using `update_xml`), KiBot will state that a component is in the schematic, but not in the PCB. However, if the component is marked as 'not in PCB' (using `on_board no`) this of course expected.

```
- Running the DRC
- Updating BoM in XML format
ERROR:BT1 found in schematic, but not in PCB (kibot.gs - gs.py:812)
ERROR:Net `GND` missing PCB connection/s: BT1 pin 2 (kibot.gs - gs.py:812)
ERROR:Net `VCC` missing PCB connection/s: BT1 pin 1 (kibot.gs - gs.py:812)
```

Find attached a minimal test project.
[not_on_board.tar.gz](https://github.com/INTI-CMNB/KiBot/files/14323996/not_on_board.tar.gz)

set-soft, 2024-03-04:
The above patch should solve the problem.
Thanks for reporting!

closed 2024-03-04 by set-soft

oliv3r, 2024-03-04:
Any thoughts/plans to make this available on 1.6.4? With the current versioning strategy and without release branches this will probably be hard.

set-soft, 2024-03-05:
> Any thoughts/plans to make this available on 1.6.4?

Nope. You can use *dev* code if you hit a bug in the current *stable*.

oliv3r, 2024-03-07:
@set-soft fair enough; any plan on when to release 1.6.5? I'd rather stick to stable builds from now on, or rather, I'd stick to tagged versions to give more predictability.

oliv3r, 2024-03-07:
> Hi @oliv3r !
> 
> I think 1.6.5 will be released in the near future because it should add KiCad 8 support.

Cool! Though may I recommend to do the 1.6.5 release as your last KiCad 7x release and then consider a slightly different versioning scheme for 8 and beyond, where you could introduce release branches? E.g. 1.7 for the initial release branch (where the first version of that branch would be v1.8.0). This would allow for cherry-picking bugfixes onto v1.8.0 in the future as well.

(or, use the 'major' number to track kicad releases, e.g. v2.x.x would be KiCad 8 based, v3.x.x would be kicad 9 based. Could even consider making the jump to v8.x.x!)

I have a (gitlab only) action that automate the whole release-branch process. The readme should explain a few things in this repo https://gitlab.com/ci-includes/pipeline-deployers/-/tree/master/templates/release-brancher and I'm sure someone made a github equivalent. Though this can still be done very much manually as well. I did it for years manually. The repo itself also shows the 'release branch structure' (though not very many releases exist yet, as its small, simple and quite reliable!).

> 
> Note that _dev_ images are quite reliable, of course they can introduce small differences, but in general they can be used for production, if you can tolerate small glitches.

My issue is not so much the quality of the 'dev' releases, but the fact that it's a moving target. For example not too long ago, you made change with the `dnp_filter` and `re_transform` (or whatever they where called, can't remember) that broke stuff all over the place. It's fine to break stuff and force us users to upgrade; but I'd rather have a little bit more control here.

set-soft, 2024-03-21:
The release numbers I use are oriented to show how compatible is the new release with the previous.

If you find an incompatibility in the *dev* images please report it.

commit 2a0596e3 (2024-03-04): [Update_XML][Fixed] Components excluded from the board reported anyways

Closes #585