INTI-CMNB/KiBot #586: [HOW] Filter DNP components from the position files

andreasWallner, 2024-02-19:
From the docs and https://github.com/INTI-CMNB/KiBot/issues/392#issuecomment-1460447474 I thought that components that have the DNP flag set in KiCad should be hidden from the bom/output files. I would like to generate position files for our production partners that don't include those components to prevent confusion.

What I tried until now, the latter ones were just tries at workarounds:
 - explicitly set the global `kicad_dnp_applied: true` (just to check, I understood this should be the default anyways)
 - a manual filter, but `exclude_not_in_bom` and `exclude_not_on_board` are for the other flags and not for DNP
 - try `dnf_filter: _kicost_dnp` 

**Do you have some PCB/Schematic to use as example?**
Just a single resistor with DNP enabled: [test.zip](https://github.com/INTI-CMNB/KiBot/files/14335061/test.zip)

**Do you have some configuration file (.kibot.yaml) that you are using?**
```yaml
kibot:
  version: 1

global:
  kicad_dnp_applied: true

outputs:
- name: basic_position_ASCII
  comment: Components position for Pick & Place
  type: position
  dir: Position
  options:
    format: ASCII
    only_smd: false
- name: basic_position_CSV
  comment: Components position for Pick & Place
  type: position
  dir: Position
  options:
    format: CSV
    only_smd: false
```

**Environment (please complete the following information):**
I'm running KiBot locally using the current (7cdb8c7dfbc0) container via `podman run --rm -it -v .:/workspace ghcr.io/inti-cmnb/kicad7_auto:latest /bin/bash -c 'cd /workspace; kibot'` (with KiBot 1.9.4, KiCad 7.10)

set-soft, 2024-03-04:
Hi @andreasWallner !

Note that *dnf_filter: _kibom_dnf_Config* is actually working, and is using the DNP flag from KiCad. 

```
### Module positions - created on Mon 19 Feb 2024 16:55:10  ###
### Printed by KiBot
## Unit = mm, Angle = deg.
## Side : top
# Ref   Val   Package   PosX   PosY   Rot   Side
## End
```

But also note that current KiCad versions has a dedicated flag for position files, in your case is disabled:

[image]

```csv
Ref,Val,Package,PosX,PosY,Rot,Side
"R1","R","R_0201_0603Metric",99.6000,-86.6750,0.0000,top
```

Enabling it removes the component from the position files:

[image]

```csv
Ref,Val,Package,PosX,PosY,Rot,Side
```

closed 2024-03-04 by set-soft

andreasWallner, 2024-03-06:
Thanks for pointing out the difference!

Would there currently be a way to filter those in kibot? Keeping the DNP flag and the "exclude from position files" in sync is a possible error source that's going to bite us at some point I fear...

set-soft, 2024-03-07:
Hi @andreasWallner !

What's wrong with using something like `dnf_filter: _kibom_dnf_Config`?