INTI-CMNB/KiBot #587: KiCad 8 unknown fonts effect attribute - bold, italic

honzakutil, 2024-02-26:
When I run KiCad 8 with kibot with project saved in Kicad 7, everything works correctly. If I open the project in KiCad 8.0 and save it, Kibot prints following error:
```
ERROR:While loading `/workdir/gstest.kicad_sch` (kibot.gs - gs.py:812)
ERROR:rectangle uuid is not a string `Symbol('0682e566-f157-46c2-a3c0-8a57344f4211')` (kibot.gs - gs.py:812)
```
If I remove the rectangle, Kibot prints another error:
```
ERROR:While loading `/workdir/gstest.kicad_sch` (kibot.gs - gs.py:812)
ERROR:Unknown font effect attribute `[Symbol('bold'), Symbol('yes')]` (kibot.gs - gs.py:812)
```
If I change text from bold to nrmal, Kibot prints another error:
```
ERROR:While loading `/workdir/gstest.kicad_sch` (kibot.gs - gs.py:812)
ERROR:Unknown font effect attribute `[Symbol('italic'), Symbol('yes')]` (kibot.gs - gs.py:812)
```
If I change text from italic to normal, Kibot runs without errors.

There is file format change between kicad v7 and v8, I guess this has something to do with that.

Could you please fix KiBot to work with schematics, where bold or italic font is used? Quickfix for me now is just not to use bold, italic (and rectangels), but finding all of them in the real schematic is quite time-consuming.

There is [gstest.zip](https://github.com/INTI-CMNB/KiBot/files/14405076/gstest.zip) with files to demonstrate described bug. It is tiny git repo with three commits, showing that kibot is working for v7 files, not working for v8 files and is working for v8, when rectangle is removed and bold/italic is not used. Kibot outputs are also zipped inside. 
I am running kibot on windows machine using docker and wsl. This is command I use to start the docker from wsl command line: `docker run --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY --env NO_AT_BRIDGE=1 --volume="./:/workdir:rw" ghcr.io/inti-cmnb/kicad_auto_test:1.6.5_k8.0.0_d12.5_b3.5`

honzakutil, 2024-02-27:
Unerstood, have a good time!

set-soft, 2024-03-05:
Hi @honzakutil !

I tried your example with the current *nightly* docker image and it worked without problems. The issues you mention are fixed in e19629406f30ba0c994af91c5d60fd3fd95ac018.

Note that more work is needed, but the above mentioned problems are already fixed.

honzakutil, 2024-03-06:
I tried to use ghcr.io/inti-cmnb/kicad_auto_test:nightly and error is still there. I guess that changes made to kibot itself was not reflected in the docker image yet. Unfortunatelly, installing and using kibot different way than pulling docker image is far beyond my knowledge.

Is there any easy way, how to use the newest kibot in the docker image currently available, or shall I just wait for docker image, which contains the fix?

honzakutil, 2024-03-06:
I think so. I use docker and wsl on windows, I start the image in the wsl using following command: `docker run --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY --env NO_AT_BRIDGE=1 --volume="./:/workdir:rw" ghcr.io/inti-cmnb/kicad_auto_test:nightly`. 

I deleted all images except the `ghcr.io/inti-cmnb/kicad_auto_test:1.6.5_k8.0.0_d12.5_b3.5` and `ghcr.io/inti-cmnb/kicad7_auto_full:dev` and tried it again now and problem still persists.

Inerestingly, if I check kicad version now (by `kicad-cli --version` in docker command line) it reports 8.0.0. When I did it this morning, before posting previous comment, I am almost sure that it reported version 8.99.

set-soft, 2024-03-06:
Hi @honzakutil !

I think the above patch will fix the bold/italic issue, but be patient, isn't in docker images yet.

set-soft, 2024-03-22:
Ok, docker images for *dev* should support it.

closed 2024-03-22 by set-soft