INTI-CMNB/KiBot #588: [BUG] "Missing sheet instance for"

andersruneson, 2024-02-27:
**Describe the bug**
Not sure this is a bug but more a request for help. When running KiBot I get the following errors but I have no idea what's wrong:
```
- 'Top PCB 2D render in PNG format' (basic_pcbdraw_png_top) [pcbdraw]
- 'Bottom PCB 2D render in PNG format' (basic_pcbdraw_png_bottom) [pcbdraw]
- 'Various DXF files' (pcb_dxf) [dxf]
- '3D model in STEP format' (basic_step_pcb) [step]
ERROR:While loading `/builds/Parts/EA/EA00192/source/EA00192.kicad_sch` (kibot.gs - gs.py:812)
ERROR:Missing sheet instance for `/8c38455b-e47e-4fa2-9a66-b2f6ffdd4c7d/00000000-0000-0000-0000-0000653a835e/00000000-0000-0000-0000-000065336a26` (kibot.gs - gs.py:812)
```

When running verbose mode:

```
- '3D model in STEP format' (basic_step_pcb) [step]
DEBUG:SCH title: `Overview` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/can_transciever.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/PSU.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/raw_inpfilt.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/raw_inpfilt.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/buck_conv.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/iso_flyback.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/supercap.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/buck_3V3.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/RPI_CM4.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/sd_card.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/debug_uart.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
DEBUG:SCH company: `None` (kibot - v6_sch.py:1845)
DEBUG:Loading sheet from /builds/Parts/EA/EA00192/source/ethernet.kicad_sch (kibot - v6_sch.py:2154)
DEBUG:SCH title: `None` (kibot - v6_sch.py:1842)
DEBUG:SCH date: `${DATE}` (kibot - v6_sch.py:1843)
DEBUG:SCH revision: `${EA}-${EArevLong}` (kibot - v6_sch.py:1844)
[... 62 lines cut ...]
    sch.sch = sch.load_sheet(project, fname, self)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/kicad/v6_sch.py", line 1655, in load_sheet
    sheet.load(os.path.join(parent_dir, self.file), project, parent_obj)
  File "/usr/lib/python3/dist-packages/kibot/kicad/v6_sch.py", line 2298, in load
    sch.sch = sch.load_sheet(project, fname, self)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/kicad/v6_sch.py", line 1662, in load_sheet
    raise SchError('Missing sheet instance for `{}`'.format(p_path))
ERROR:While loading `/builds/Parts/EA/EA00192/source/EA00192.kicad_sch` (kibot.gs - gs.py:812)
ERROR:Missing sheet instance for `/8c38455b-e47e-4fa2-9a66-b2f6ffdd4c7d/00000000-0000-0000-0000-0000653a835e/00000000-0000-0000-0000-000065336a26` (kibot.gs - gs.py:812)
```

**To Reproduce**
I would be able to send a design, but I might have to "clean" it due to customer security.

**Expected behavior**
No such errors. :)

**Screenshots**

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? Yes, self hosted GitLab
- Using a docker image? Yes, KiBot 1.6.4 dev

closed 2024-02-29 by andersruneson

andersruneson, 2024-03-05:
Hopefully a bug in an earlier Ki7-release, but who knows. 
We do have one other bug that we've seen two times, where there is a unconnected trace reported in the DRC in KiBot but not in KiCad. Next time I see that I will try to break out that board to see if we can find the issue.
Thanks

EricAndrechek, 2024-10-18:
I have tried running this on an internal company repo - one branch with KiCAD 7 and kept getting this issue. Tried it on the KiCAD 8 branch and I am still having the same one. Any advice is appreciated!

set-soft, 2024-10-18:
Are you sure the file was saved using KiCad 8? Try opening the schematic and saving it with KiCad 8