INTI-CMNB/KiBot #589: [BUG] FileNotFoundError with `add_link_id: true` in KiBot diff output

pasrom, 2024-03-04:
**Bug Description**

When enabling `add_link_id` and using the `basic_diff_pcb` or `basic_diff_sch` output, an error occurs indicating a missing file. The error message is:

```
FileNotFoundError: [Errno 2] No such file or directory: 'bms-c1-diff_pcb.pdf' -> '/Users/roman/git/bms-c1/output/diff/bms-c1-diff_pcb_3bd7b67(origin/main)-471e8aa(HEAD).pdf'
```
‎ 
**Steps to Reproduce**

This issue was encountered with the following `.kibot.yaml` configuration: [kibot config](https://github.com/e-battery-systems/bms-c1-tests/blob/0872333201eb474a360a61ace3a2084189801fbc/kicad/diff.kibot.yaml#L1-L17) (PR: https://github.com/e-battery-systems/bms-c1-tests/pull/6).

**Expected Behavior**

The expected behavior is consistent with the functionality in GH-action tag `v2_k7_1_6_3`, where the described error does not occur ([GH action run](https://github.com/LibreSolar/bms-c1/actions/runs/7644088779/job/20827584104)).
Image used: `ghcr.io/inti-cmnb/kicad7_auto_full:latest@sha256:cd92486f6f295074ce4b7d09bc3a4d39b271fa31235600bfa70400c958aea6f6`
```
🤖 KiBot GitHub Action v2 🚀

KiBot: 1.6.3
KiCad: 7.0.9~ubuntu23.04.1
Debian: 12.1
KiAuto: 2.2.8
KiCost 1.1.18
iBoM: 2.8.1
```

**Environment Details**

- Execution context: CI/CD workflow, specifically [this job](https://github.com/e-battery-systems/bms-c1-tests/actions/runs/8146184921/job/22264123732).
- GitHub action tag used: `INTI-CMNB/KiBot@v2_k7`. 

Image used: `ghcr.io/inti-cmnb/kicad7_auto_full:latest@sha256:bb5e5319e5ba22d560e28a69b89c8cfd587474b9574314f3f678dcbff376f621`
```
🤖 KiBot GitHub Action v2 🚀

KiBot: 1.6.4
KiCad: 7.0.10~ubuntu23.04.1
Debian: 12.1
KiAuto: 2.2.8
KiCost 1.1.18
iBoM: 2.8.1
```

**Additional Information**

This bug may have been introduced in recent commits, particularly [[Diff][Fixed] Problems when trying to use an already existent worktree](https://github.com/INTI-CMNB/KiBot/commit/8a69f0349cf41a57e68a5e1d5b2cb250bb6ede95) and [[Diff][Fixed] Worktree's can leave junk branches](https://github.com/INTI-CMNB/KiBot/commit/8b58759f3d1534483b9975ea30dcac72b1e458eb), which impacted how branches are specified (`origin/main` vs. `main`) and led to link creation failures when `add_link_id` is `true`. The issue arises from a modification in branch specification requirements; where previously referring to a branch simply as `main` was adequate, it is now necessary to explicitly use `origin/main`. As a result, utilizing `origin/main` in combination with `add_link_id: true` disrupts the link creation process.

set-soft, 2024-03-05:
Hi @pasrom !

The above patch should solve the problem. It will change *origin/main* to *origin_main*.

You can try using *v2_dk7* instead of *v2_k7*

Note: When I tried your repo locally I was able to use either *origin/main* and *main*. But this could be related to what git uses as default source.

closed 2024-03-05 by set-soft

pasrom, 2024-03-06:
@set-soft, thanks for the quick fix!

Tried the `v2_dk7` tag and it worked like a charm.

Thanks again!

commit d6618801 (2024-03-05): [Diff][Fixed] problems when using things like "origin/main"

- and add_link_id

Fixes #589