INTI-CMNB/KiBot #590: [FEATURE] Better logging for corrupted kicad configs

FrankvVeelen, 2024-03-11:
When KiBot has crashed sometimes the kicad config files are empty, but the backup is in pre_script backups. Right now when retrying kibot, there is no clear error reporting, but instead a json load error in eeschema, which tries to load an empty json file. A simple check if the read json is valid with a clear error would be nice.

set-soft, 2024-03-21:
I'm not sure how do you get this. I guess the problem is that you are not retrying the same operation.

The tool altering the config, and creating the back-ups, is KiAuto. If you run KiAuto and it detects the back-ups you get:

```
ERROR:PCBnew config back-up found (~/.config/kicad/7.0/pcbnew.json.pre_script) (pcbnew_do.kiauto.file_util - file_util.py:210) (kibot - kiplot.py:124)
ERROR:It could contain your pcbnew configuration, rename it to ~/.config/kicad/7.0/pcbnew.json or discard it. (pcbnew_do.kiauto.file_util - file_util.py:211) (kibot - kiplot.py:124)
ERROR:DRC returned 5 (kibot.gs - gs.py:815)
```

I got it after doing:

```
touch ~/.config/kicad/7.0/pcbnew.json.pre_script
```

To artificially simulate a bad crash, and then run KiBot asking to do the DRC, from the root of the repo I used:

```
src/kibot -c tests/yaml_samples/drc.kibot.yaml -b tests/board_samples/kicad_7/light_control.kicad_pcb -d pp
```

KiAuto detected the back-ups, refused to run and reported the error.

In which cases you get no error?

set-soft, 2024-04-02:
Hi @FrankvVeelen !
Any news on this?

FrankvVeelen, 2024-04-02:
Hi @set-soft,

Thanks for taking a look. I've tried to reproduce the error as well, but I'm not able to either. 
Seems something weird happened. 
I'll close this for now and report back if I run into this again.

closed 2024-04-02 by FrankvVeelen