INTI-CMNB/KiBot #594: [FEATURE] Handle hierachical sheets better

oliv3r, 2024-03-24:
KiBot currently takes no effort to detect (or distinguish) between a h(ierarchical) schematic and a kicad project.

This shows when running kibot (without specifying a schematic file) in a project that has a hierarchical schematic.

KiBot could check first, if no schematic is requested in the cli, and multiple schematics are present if there is a `.kicad_pro` file. If it exist, use that filename for the schematic.

I think KiBot might first scan for `.kicad_pro` files and use schematic/layout based on the project file as its first 'guess' anyway ...

Eeschema does the following logic when opening a the root schematic. It checks if there is a `.kicad_pro` file matching the name, and if so, extract/creates the schematic tree of this.

There might be a problem where non-kicad tools are used to generate stuff, as they will not know the hierarchical nature of the schematic, but I don't know if this will be an issue at all.

set-soft, 2024-03-24:
And what makes you think KiBot isn't checking for this?

oliv3r, 2024-03-24:
Because warning 044 comes up saying as things might not be right.

It complaints about multiple schematics being present. Kibot should be smart to figure that the other schematics are hierarchical and automatically used/included from the root/project.

E.g. generating the PDF with etc and DRC should just work, as after consuming all files, no 'other' schematics are there.

Basically kibot will always warn on hierarchical sheets, even they kind of belong together.

set-soft, 2024-03-25:
> Because warning 044 comes up saying as things might not be right.

Which is true indeed.

> It complaints about multiple schematics being present. Kibot should be smart to figure that the other schematics are hierarchical and automatically used/included from the root/project. E.g. [...]

The only way to know if *all* the schematics belongs to a single hierarchy is by loading all of them, a real overkill. But even in this case: how can KiBot know you don't want to use a partial hierarchy? 

The only way is to provide the name of the schematic you want to use as top-level. 

If you don´t like it you are free to filter W044.

closed 2024-03-25 by set-soft

oliv3r, 2024-03-25:
>  is by loading all of them, a real overkill.

I don't know if it is overkill ... but ignoring 44 is certainly the lazy/easy way.

I would have suggested to move the warning to check of `.kicad_pro` files, if there's only one, it's quite clear what is wanted/intended. Loading all of them isn't too horrible then.

How would a partially hierarchical kibot run look like? Unless each sheet is completly isolated,you'd always end up with missing/broken connections no?

set-soft, 2024-03-25:
> How would a partially hierarchical kibot run look like? Unless each sheet is completly isolated,you'd always end up with missing/broken connections no?

Not necessarily, KiCad supports recycling the same schematic in more than one project. So each schematic has its own references. And you can have alternative top-levels. I know this isn't the most common use, but if KiBot is guessing it should be informed. Currently we don't have "warning levels". So you can just ignore the warnings you are already aware of.