INTI-CMNB/KiBot #598: [BUG] KiCad8 hierarchical sheet

Kedarius, 2024-04-01:
**Describe the bug**
I am testing the KiCad 8 support and everything seems to work fine for projects with 1 sheet. For projects with hierarchical sheets KiBot fails with:

```
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.5', 'console_scripts', 'kibot')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 546, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 599, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 589, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 493, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 214, in run
    self.options.run(target)
  File "/usr/lib/python3/dist-packages/kibot/out_ibom.py", line 192, in run
    GS.sch.save_netlist(f, self._comps)
  File "/usr/lib/python3/dist-packages/kibot/kicad/v5_sch.py", line 2066, in save_netlist
    self.save_netlist_components(root, comps, excluded, fitted, no_field)
  File "/usr/lib/python3/dist-packages/kibot/kicad/v5_sch.py", line 1979, in save_netlist_components
    prop.set('value', os.path.basename(c.parent_sheet.fname))
                                       ^^^^^^^^^^^^^^
AttributeError: 'SchematicComponentV6' object has no attribute 'parent_sheet'.
```

**Environment (please complete the following information):**
Where are you running KiBot:
I am using docker, I tried both latest and dev images: 
ghcr.io/inti-cmnb/kicad8_auto_full:dev
ghcr.io/inti-cmnb/kicad8_auto_full:latest

set-soft, 2024-04-01:
Hi @Kedarius !
Please post a simple example showing this issue. As you might guess regression tests uses hierarchical sheets and they work.

Kedarius, 2024-04-01:
Thank for the reply. I looked into it deeper and it seems that the issue is related to the ibom output.
I attached a simple project that fails with the ibom output. The kibot config is included. I also included run.sh but that uses absolute paths, so you would need to change it.
[20240401-kibot-sheets-issue-ibom.tar.gz](https://github.com/INTI-CMNB/KiBot/files/14826240/20240401-kibot-sheets-issue-ibom.tar.gz)
The output is:
```
# docker run --volume /data/electronics/temp/kibot:/kicad-project: --volume /data/electronics/temp/kibot/out:/output:  --volume /data/electronics/tools/kibot/:/kibot:  ghcr.io/inti-cmnb/kicad8_auto_full:dev kibot -c /kicad-project/kibot-development.yaml  --out-dir /output/ --board  "/kicad-project/panel-stick.kicad_pcb" --schematic  "/kicad-project/panel-stick.kicad_sch" ibom
- Updating BoM in XML format
- 'Generates an interactive web page useful to identify the position of the components in the PCB.' (ibom) [ibom]
WARNING:(W076) Unknown sheet property `Field2` () (kibot - v6_sch.py:1689)
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/__main__.py", line 546, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 599, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 589, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 493, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.11/dist-packages/kibot/out_base.py", line 214, in run
    self.options.run(target)
  File "/usr/local/lib/python3.11/dist-packages/kibot/out_ibom.py", line 192, in run
    GS.sch.save_netlist(f, self._comps)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kicad/v5_sch.py", line 2066, in save_netlist
    self.save_netlist_components(root, comps, excluded, fitted, no_field)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kicad/v5_sch.py", line 1979, in save_netlist_components
    prop.set('value', os.path.basename(c.parent_sheet.fname))
                                       ^^^^^^^^^^^^^^
AttributeError: 'SchematicComponentV6' object has no attribute 'parent_sheet'

```

The project has been created in KiCad7 and I've been using it with Kibot for some time. Now I did open it and resave under KiCad8 and tried the new Kibot.

Let me know if I could provide more info...

set-soft, 2024-04-01:
Ok, I found the problem.
For some reason your logos (F1, F2, G1 and G2) aren't filtered, I didn't investigate it because this is most probably a consequence of the filters you are using.
At the same time they are only in the PCB, and correctly marked this way. But this means they don't have a parent sheet or sheet path. So KiBot failed to put them in the netlist file.
Also note that as you are applying filters to the iBoM output it must create a filtered netlist, so you don't really need to update the netlist using a preflight, this is internally created anyways.

Kedarius, 2024-04-01:
I see. The only strange thing is that it worked fine with v7. Probably because of the changes in exclusion fields....
The config I provided is very old and I keep updating it and modifying it as needed - so there is probably a lot of "garbage".

set-soft, 2024-04-01:
If you find some change in the filters just report it.
Now the *dev* images should fix the issue.

closed 2024-04-01 by set-soft

Kedarius, 2024-04-01:
I confirm that it works fine, thanks!

commit f6071289 (2024-04-01): [Netlist][Fixed] Generation problems

- With components on the PCB but not in schematic.
- I.e. logos reaching iBoM output

Fixes #598