INTI-CMNB/KiBot #604: [BUG] KiCad 8: Crash with custom component reference name in KiBot 1.6.5

rainer38, 2024-04-07:
**Describe the bug**
When i add a footprint in pcbnew like a mounting hole the reference is named "REF**". It seems that some parser in KiBot only accepts known reference prefixes like U, R, C, ... but not names which has no real component prefix.
When i rename the reference to something like "mh" it also crashes but when i name it U1 it works without problems.

This worked with custom reference names until v1.6.3.

**To Reproduce**
Just create a pcb and add a mounting hole like "MountingHole_3.2mm_M3_Pad_Via" with reference "REF**" and try to create a POS file.

**Expected behavior**
The reference name should be irrelevant.

**Crashlog**
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.5', 'console_scripts', 'kibot')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 546, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 599, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 589, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 493, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 214, in run
    self.options.run(target)
  File "/usr/lib/python3/dist-packages/kibot/out_position.py", line 252, in run
    super().run(fname)
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 1048, in run
    get_board_comps_data(comps)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 344, in get_board_comps_data
    c = create_component_from_footprint(m, ref)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 317, in create_component_from_footprint
    c.split_ref()
  File "/usr/lib/python3/dist-packages/kibot/kicad/v5_sch.py", line 1079, in split_ref
    raise SchError('Malformed component reference `{}`'.format(self.ref))
kibot.kicad.error.SchError: Malformed component reference `REF**`

**Environment (please complete the following information):**
I'm running KiBot 1.6.5  in Debian Linux on command line.

Application: KiCad x86_64 on x86_64

Version: 8.0.1-dirty, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/8.5.0 GnuTLS/3.7.9 zlib/1.2.13 brotli/1.0.9 zstd/1.5.4 libidn2/2.3.3 libpsl/0.21.2 (+libidn2/2.3.3) libssh2/1.10.0 nghttp2/1.52.0 librtmp/2.3 OpenLDAP/2.5.13

Platform: Debian GNU/Linux 12 (bookworm), 64 bit, Little endian, wxGTK, X11, KDE, x11

Build Info:
	Date: Apr  7 2024 09:07:41
	wxWidgets: 3.2.2 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.74.0
	OCC: 7.6.3
	Curl: 8.5.0
	ngspice: 42
	Compiler: GCC 12.2.0 with C++ ABI 1017

set-soft, 2024-04-07:
Hi @rainer38 !

Ok, the bug is that the outputs run wasn't catching the SchError signals. Now fixed.

But I guess you pretend to KiBot accept your _REF**_, if this is the case then the answer is no. You must annotate your project first, and references must be in the **PREFIX**_NUMBER_ format, as the IPC normative explain. Things like "A hole" isn't a valid reference. Repeated references aren't a good idea, just imagine you have 5 components with "REF**" and KiBot reports an error with one of them, how can you know which one?

closed 2024-04-07 by set-soft

set-soft, 2024-04-08:
Hi @rainer38 !

The above patch makes KiBot more tolerant, it won't stop, but the component will be disregarded, and a warning message will be issued.

commit a120a85f (2024-04-07): [Fixed] Sch errors are now caught during output runs.

Closes #604