INTI-CMNB/KiBot #606: [BUG] [HOW] pcbnew not found

Unichord, 2024-04-08:
I installed KiCad 8.0.1 on WSL running in Win10, can't run kibot anymore after I updated kibot to 1.6.5

kibot-check fails and returns this error:
```
Traceback (most recent call last):
  File "/usr/bin/kibot-check", line 1696, in <module>
    import pcbnew
ModuleNotFoundError: No module named 'pcbnew'
```

This is the output of `apt upgrade`
```
Setting up kibot (1.6.5-1) ...
ERROR:Failed to import pcbnew Python module. Is KiCad installed? Do you need to add it to PYTHONPATH? (kibot.gs - gs.py:814)
ERROR:Try running the installation checker: kibot-check (kibot.gs - gs.py:814)
dpkg: error processing package kibot (--configure):
 installed kibot package post-installation script subprocess returned error exit status 16
Errors were encountered while processing:
 kibot
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

It's most probably an environment issue, but I have no clue on how to fix it. If anything else is needed, please let me know.

set-soft, 2024-04-08:
How did you install KiCad 8?

On a WSL shell try:

```shell
$ python3
>>> import pcbnew
```

If it fails then KiCad is wrongly installed. Try to find where is *pcbnew.py*, like this:

```
find / -name pcbnew.py
```

Also try running:

```
which kicad-cli
```

Unichord, 2024-04-08:
Hello Salvador, I installed KiCad like this:
```
sudo add-apt-repository ppa:kicad/kicad-8.0-releases
sudo apt install kicad
```

The output of the python shell is:
```
python3
Python 3.8.10 (default, Nov 22 2023, 10:22:35)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pcbnew
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pcbnew'
```

The output of `find` is this:
```
/home/andrea/.local/lib/python3.8/site-packages/pcbnewTransition/pcbnew.py
/root/.local/lib/python3.8/site-packages/pcbnewTransition/pcbnew.py
/usr/lib/python3.8/site-packages/pcbnew.py
/usr/lib/python3/dist-packages/pcbnewTransition/pcbnew.py
/usr/local/lib/python3.8/dist-packages/pcbnewTransition/pcbnew.py
```

And `which kicad-cli` returns this:
```
which kicad-cli
/usr/bin/kicad-cli
```

set-soft, 2024-04-08:
> Hello Salvador, I installed KiCad like this:
> 
> ```
> sudo add-apt-repository ppa:kicad/kicad-8.0-releases
> sudo apt install kicad
> ```

Ok, this package contains *pcbnew.py* and a link to the binary.

> The output of the python shell is:
> 
> ```
> python3
> Python 3.8.10 (default, Nov 22 2023, 10:22:35)

Your Python is 3.8, a little bit old. Which Ubuntu do you have installed? (Should be 20.04)

> [GCC 9.4.0] on linux
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import pcbnew
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ModuleNotFoundError: No module named 'pcbnew'
> ```

Python can't load the library, so the lib is wrongly installed.

> The output of `find` is this:
>
> ```
> /home/andrea/.local/lib/python3.8/site-packages/pcbnewTransition/pcbnew.py
> /root/.local/lib/python3.8/site-packages/pcbnewTransition/pcbnew.py [...]

Here the relevant is: */usr/lib/python3.8/site-packages/pcbnew.py*, which is right for Python 3.8.

Note: You have a real mess with KiKit and friends, you have 4 copies of pcbnewTransition, this is a potential problem for. You should have just one: */usr/lib/python3/dist-packages/pcbnewTransition/pcbnew.py*

> And `which kicad-cli` returns this:
> 
> ```
> which kicad-cli
> /usr/bin/kicad-cli
> ```

So this is in the right place. Try:

```
$ ls -la /usr/lib/python3.8/site-packages /usr/bin
```

Also try reinstalling KiCad, if you didn't clean the apt cache you should be able to run:

```
sudo dpkg -i /var/cache/apt/archives/kicad_8.0.1-0~ubuntu20.04.1_amd64.deb
```

Unichord, 2024-04-08:
> Your Python is 3.8, a little bit old. Which Ubuntu do you have installed? (Should be 20.04)

I will update it when I can, I have Ubuntu 20.04

> Note: You have a real mess with KiKit and friends, you have 4 copies of pcbnewTransition, this is a potential problem for. You should have just one: _/usr/lib/python3/dist-packages/pcbnewTransition/pcbnew.py_

Ok, should I just delete them? I really don't know what to do with this. I believe kikit was installed only by kibot, but I don't really remember, it was a while ago.

> 
> > And `which kicad-cli` returns this:
> > ```
> > which kicad-cli
> > /usr/bin/kicad-cli
> > ```
> 
> So this is in the right place. Try:
> 
> ```
> $ ls -la /usr/lib/python3.8/site-packages /usr/bin
> ```
```
drwxr-xr-x  2 root root   4096 Mar 15 08:11 .
drwxr-xr-x 31 root root  20480 Mar  7 14:36 ..
lrwxrwxrwx  1 root root     27 Mar 14 16:11 _pcbnew.so -> ../../../bin/_pcbnew.kiface
-rw-r--r--  1 root root 797969 Mar 14 16:11 pcbnew.py
```
I don't know if it's useful, but I also searched for this:
```
ls -la /usr/lib/python3.8/site-packages /usr/bin | grep pcbnew
-rwxr-xr-x  1 root   root     35862688 Mar 14 16:11 _pcbnew.kiface
-rwxr-xr-x  1 root   root      2375128 Mar 14 16:11 pcbnew
-rwxr-xr-x  1 root   root        80761 Mar 20 11:43 pcbnew_do
lrwxrwxrwx  1 root root     27 Mar 14 16:11 _pcbnew.so -> ../../../bin/_pcbnew.kiface
-rw-r--r--  1 root root 797969 Mar 14 16:11 pcbnew.py
```
And this:
```
ls -la /usr/lib/python3.8/site-packages /usr/bin | grep python
-rwxr-xr-x  1 root   root      4461376 Mar 14 16:11 _kipython.kiface
lrwxrwxrwx  1 root   root           23 Nov 22 11:22 pdb3.8 -> ../lib/python3.8/pdb.py
lrwxrwxrwx  1 root   root           31 Mar 13  2020 py3versions -> ../share/python3/py3versions.py
lrwxrwxrwx  1 root   root            9 Mar 13  2020 python3 -> python3.8
lrwxrwxrwx  1 root   root           16 Mar 13  2020 python3-config -> python3.8-config
-rwxr-xr-x  1 root   root      5465880 Nov 22 11:22 python3.8
lrwxrwxrwx  1 root   root           33 Nov 22 11:22 python3.8-config -> x86_64-linux-gnu-python3.8-config
lrwxrwxrwx  1 root   root           33 Mar 13  2020 x86_64-linux-gnu-python3-config -> x86_64-linux-gnu-python3.8-config
-rwxr-xr-x  1 root   root         3241 Nov 22 11:22 x86_64-linux-gnu-python3.8-config
```

> Also try reinstalling KiCad, if you didn't clean the apt cache you should be able to run:
> 
> ```
> sudo dpkg -i /var/cache/apt/archives/kicad_8.0.1-0~ubuntu20.04.1_amd64.deb
> ```
No luck, cache was clean apparently. I will try reinstalling it from scratch.

set-soft, 2024-04-08:
> Ok, should I just delete them? I really don't know what to do with this. I believe kikit was installed only by kibot, but I don't really remember, it was a while ago.

It looks like you have various installations:
- /home/andrea/.local/lib/python3.8/site-packages/pcbnewTransition/pcbnew.py   This looks like pip installed as user *andrea*.  Using *pip uninstall* could help
- /root/.local/lib/python3.8/site-packages/pcbnewTransition/pcbnew.py This looks like pip installed as user, but using *root*
- /usr/local/lib/python3.8/dist-packages/pcbnewTransition/pcbnew.py And this is strange like a system wide installation, but using local prefix. Not sure how to cleanly remove.

You should take a look at what is installed in:
- /home/andrea/.local/lib/python3.8/site-packages/
- /root/.local/lib/python3.8/site-packages/
- /usr/local/lib/python3.8/dist-packages/

> ```
> drwxr-xr-x  2 root root   4096 Mar 15 08:11 .
> drwxr-xr-x 31 root root  20480 Mar  7 14:36 ..
> lrwxrwxrwx  1 root root     27 Mar 14 16:11 _pcbnew.so -> ../../../bin/_pcbnew.kiface
> -rw-r--r--  1 [...]

The sizes matches with the 8.0.1 package for 20.04

> > Also try reinstalling KiCad, if you didn't clean the apt cache you should be able to run:
> > ```
> > sudo dpkg -i /var/cache/apt/archives/kicad_8.0.1-0~ubuntu20.04.1_amd64.deb
> > ```
> 
> No luck, cache was clean apparently. I will try reinstalling it from scratch.

Do you have an environment variable named *PYTHONPATH*, try:

```
echo ${PYTHONPATH}
```

Also try this:

```
$ python3
>>> import sys
>>> print(sys.path)
>>> exit()
```

Looking at the values in my system I suspect the KiCad package is wrong. Try this:

```
$ PYTHONPATH=/usr/lib/python3.8/site-packages/  /usr/bin/kibot-check
```

Or just:

```
$ export PYTHONPATH=/usr/lib/python3.8/site-packages/
```

And then install KiBot.

Unichord, 2024-04-08:
This is what happens when I add PYTHONPATH to the environment variable as you suggested:
```
kibot-check
KiBot installation checker

Core:
Linux: 5.15.146.1 (Linux CLE-ANDREA 5.15.146.1-microsoft-standard-WSL2 #1 SMP Thu Jan 11 04:09:03 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux)
Python: 3.8.10 (default, Nov 22 2023, 10:22:35)  [GCC 9.4.0]
KiCad: 8.0.1-8.0.1-0~ubuntu20.04.1
Kibot: 1.6.5

Modules:
Colorama: 0.4.3
LXML: 4.5.0
Lark: 0.7.8
PyYAML: 5.3.1
QRCodeGen: Ok
Requests: 2.22.0
XLSXWriter: 1.1.2
Xvfbwrapper: Ok
markdown2: 2.4.10
numpy: 1.24.4

Tools:
Bash: 5.0.17 (GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu))
Blender: Not available
Ghostscript: 9.50.0 (9.50)
Git: 2.25.1 (git version 2.25.1)
ImageMagick: 6.9.10.23 (Version: ImageMagick 6.9.10-23 Q16 x86_64 20190101 https://imagemagick.org)
Interactive HTML BoM: 2.9.0 (v2.9.0)
KiBoM: 1.9.1 (KiBOM Version: 1.9.1)
KiCad Automation tools (kiauto): 2.3.0 (pcbnew_do 2.3.0 - Copyright 2018-2024, INTI/Productize SPRL - License: Apache)
KiCad PCB/SCH Diff (kidiff): 2.4.3 (kicad-diff.py 2.4.3 - Copyright 2020-2022, INTI/Salvador E. Tropea - License:)
KiCost: 1.1.18 (KiCost v1.1.18)
Failed to run /usr/bin/kikit, error 1
KiKit: *UNKNOWN*
OpenSCAD: 2019.5.0 (OpenSCAD version 2019.05)
Pandoc: 2.5.0 (pandoc 2.5)
RAR: 5.50.0 (RAR 5.50   Copyright (c) 1993-2017 Alexander Roshal   11 Aug 2017)
RSVG tools: 2.48.9 (rsvg-convert version 2.48.9)
Xvfb: Ok (xvfb-run)

* Blender not installed or too old
  Install the `blender` package, i.e.: `sudo apt-get install blender`
  - Mandatory for `blender_export`

* KiCad PCB/SCH Diff not installed or too old
  Visit: https://github.com/INTI-CMNB/KiDiff
  Download it from: https://github.com/INTI-CMNB/KiDiff/releases
  This tool might be automatically downloaded by KiBot.
  - Mandatory for: `diff`, `kiri`

* KiKit not installed or too old
  Visit: https://github.com/INTI-CMNB/KiKit
  Download it from: https://github.com/INTI-CMNB/KiKit/releases
  Official 1.3.0 release does not work, use my fork if 1.3.0 is the latest
  You can also try the official 1.4.0 release
  This tool might be automatically downloaded by KiBot.
  - Mandatory for: `panelize`, `stencil_3d`, `stencil_for_jig`
  - Optional to separate multiboard projects for general use
```

So it seems fine (except kikit), but if I try to update and upgrade this happens:
```
sudo apt upgrade -y
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up kibot (1.6.5-1) ...
ERROR:Failed to import pcbnew Python module. Is KiCad installed? Do you need to add it to PYTHONPATH? (kibot.gs - gs.py:814)
ERROR:Try running the installation checker: kibot-check (kibot.gs - gs.py:814)
dpkg: error processing package kibot (--configure):
 installed kibot package post-installation script subprocess returned error exit status 16
Errors were encountered while processing:
 kibot
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Regarding kikit, this happens when I try to uninstall it (also with sudo):
```
pip uninstall kikit
Found existing installation: KiKit 1.5.1.post1
Not uninstalling kikit at /usr/lib/python3/dist-packages, outside environment /usr
Can't uninstall 'KiKit'. No files were found to uninstall.
```

set-soft, 2024-04-08:
> This is what happens when I add PYTHONPATH to the environment variable as you suggested:
[snip]
> So it seems fine (except kikit),

Ok, so the KiCad package is wrong. 

> but if I try to update and upgrade this happens:
[snip]

This is because you defined the variable for the user and then you switched to root. Try this:

```
$ sudo su
# export PYTHONPATH=/usr/lib/python3.8/site-packages/
#  apt upgrade -y
```

Look in google how to make the environment variable available to all users. 

 
> Regarding kikit, this happens when I try to uninstall it (also with sudo):
> 
> ```
> pip uninstall kikit
> Found existing installation: KiKit 1.5.1.post1
> Not uninstalling kikit at /usr/lib/python3/dist-packages, outside environment /usr
> Can't uninstall 'KiKit'. No files were found to uninstall.
> ```

Try the --user option.

set-soft, 2024-04-08:
BTW: Try running `kikit --version` manually

Unichord, 2024-04-09:
> Try the --user option.

The following happens also as root:
```
andrea@CLE-ANDREA:/mnt/c/Users/Andrea$ pip uninstall --user kikit

Usage:
  pip uninstall [options] <package> ...
  pip uninstall [options] -r <requirements file> ...

no such option: --user
```

> This is because you defined the variable for the user and then you switched to root. Try this:
> 
> ```
> $ sudo su
> # export PYTHONPATH=/usr/lib/python3.8/site-packages/
> #  apt upgrade -y
> ```

This works, apparently. At least I don't have an error anymore when I apt upgrade. Thank you.

> BTW: Try running `kikit --version` manually

This is what happens, also as root:
```
andrea@CLE-ANDREA:/mnt/c/Users/Andrea$ kikit --version
Traceback (most recent call last):
  File "/usr/bin/kikit", line 33, in <module>
    sys.exit(load_entry_point('KiKit==1.5.1.post1', 'console_scripts', 'kikit')())
  File "/usr/bin/kikit", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/kikit/ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "/usr/lib/python3/dist-packages/kikit/plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "/usr/lib/python3/dist-packages/kikit/panelize.py", line 4, in <module>
    from pcbnewTransition import pcbnew, isV6
  File "/home/andrea/.local/lib/python3.8/site-packages/pcbnewTransition/__init__.py", line 23, in <module>
    from .transition import *
  File "/home/andrea/.local/lib/python3.8/site-packages/pcbnewTransition/transition.py", line 110, in <module>
    pcbnew.PCB_SHAPE = pcbnew.DRAWSEGMENT
AttributeError: module 'pcbnewTransition.pcbnew' has no attribute 'DRAWSEGMENT'
```

By the way, thank you for your support.

set-soft, 2024-04-09:
> `File "/home/andrea/.local/lib/python3.8/site-packages/pcbnewTransition/__init__.py", line 23, in <module>`

You are using the wrong pcbnewTransition, this is one installed with pip and is obsolete.
Try:

```
$ rm -r /home/andrea/.local/lib/python3.8/site-packages/pcbnewTransition*
```

Then run KiKit again. Remove any copy that isn't from */usr/lib/python3*

Unichord, 2024-04-09:
> You are using the wrong pcbnewTransition, this is one installed with pip and is obsolete. Try:
>
> ```
> $ rm -r /home/andrea/.local/lib/python3.8/site-packages/pcbnewTransition*
> ```
>
> Then run KiKit [...]

Done. The only one remaining is:
`/usr/lib/python3/dist-packages/pcbnewTransition`

Still, I get errors when I run `kikit --version`:

```
Traceback (most recent call last):
  File "/usr/bin/kikit", line 33, in <module>
    sys.exit(load_entry_point('KiKit==1.5.1.post1', 'console_scripts', 'kikit')())
  File "/usr/bin/kikit", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/kikit/ui.py", line 2, in <module>
    from kikit import (panelize_ui, export_ui, present_ui, stencil_ui,
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui.py", line 7, in <module>
    from kikit.panelize_ui_sections import *
  File "/usr/lib/python3/dist-packages/kikit/panelize_ui_sections.py", line 4, in <module>
    from kikit import plugin
  File "/usr/lib/python3/dist-packages/kikit/plugin.py", line 10, in <module>
    from kikit.panelize import Panel
  File "/usr/lib/python3/dist-packages/kikit/panelize.py", line 4, in <module>
    from pcbnewTransition import pcbnew, isV6
  File "/usr/lib/python3/dist-packages/pcbnewTransition/__init__.py", line 23, in <module>
    from .transition import *
  File "/usr/lib/python3/dist-packages/pcbnewTransition/transition.py", line 9, in <module>
    import pcbnew
ModuleNotFoundError: No module named 'pcbnew'
```

set-soft, 2024-04-09:
> ModuleNotFoundError: No module named 'pcbnew'

This is the same error you had with KiBot, are you defining the environment variable?

Another solution is to make something like this:

```
$ sudo ln -s /usr/lib/python3.8/site-packages/pcbnew.py /usr/lib/python3.8/dist-packages/pcbnew.py
$ sudo ln -s /usr/lib/python3.8/site-packages/_pcbnew.so /usr/lib/python3.8/dist-packages/_pcbnew.so
```

Unichord, 2024-04-09:
> > ModuleNotFoundError: No module named 'pcbnew'
> 
> This is the same error you had with KiBot, are you defining the environment variable?

I guess so, I created a `env_kibot.sh` file inside /etc/profile.d which is like this:
```
#!/bin/bash
export PYTHONPATH=/usr/lib/python3.8/site-packages/
```
and I have checked it to be present in env.

> Another solution is to make something like this:
> 
> ```
> $ sudo ln -s /usr/lib/python3.8/site-packages/pcbnew.py /usr/lib/python3.8/dist-packages/pcbnew.py
> $ sudo ln -s /usr/lib/python3.8/site-packages/_pcbnew.so /usr/lib/python3.8/dist-packages/_pcbnew.so
> ```

Result:
`ln: failed to create symbolic link '/usr/lib/python3.8/dist-packages/pcbnew.py': No such file or directory`

`ln: failed to create symbolic link '/usr/lib/python3.8/dist-packages/_pcbnew.so': No such file or directory`

set-soft, 2024-04-09:
> I guess so, I created a `env_kibot.sh` file inside /etc/profile.d which is like this:
>
> ```
> #!/bin/bash
> export PYTHONPATH=/usr/lib/python3.8/site-packages/
> ```
>
> and I have checked it to be present in [...]

Double check it using the Python interpreter, i.e. 

```
$ python3
>>> import sys
>>> sys.path
```

> Result: `ln: failed to create symbolic link '/usr/lib/python3.8/dist-packages/pcbnew.py': No such file or directory`
>
> `ln: failed to create symbolic link [...]

Just create the missing dir:

```
$ sudo mkdir /usr/lib/python3.8/dist-packages/
```

If you take a look at the Debian package for Ubuntu 22.04 you'll see the files are in a similar dir

Unichord, 2024-04-09:
Interpreter test (also posted above):
```
python3
Python 3.8.10 (default, Nov 22 2023, 10:22:35)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path
['', '/usr/lib/python3.8/site-packages', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/andrea/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.8/dist-packages']
```

dist-packages test:
```
ls -la /usr/lib/python3.8/dist-packages/
total 28
drwxr-xr-x  2 root root  4096 Apr  9 14:18 .
drwxr-xr-x 32 root root 20480 Apr  9 14:18 ..
lrwxrwxrwx  1 root root    43 Apr  9 14:18 _pcbnew.so -> /usr/lib/python3.8/site-packages/_pcbnew.so
lrwxrwxrwx  1 root root    42 Apr  9 14:18 pcbnew.py -> /usr/lib/python3.8/site-packages/pcbnew.py
```

set-soft, 2024-04-09:
Hi @Unichord !

With the above results I don't see why KiKit could fail. In particular when all the other scripts work (i.e. KiBot and iBoM).

What do you get from:

```
dpkg -l "python3-pcbnew*" kikit
```

Unichord, 2024-04-09:
This is the output (also as root):
```
dpkg -l "python3-pcbnew*" kikit
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                     Version      Architecture Description
+++-========================-============-============-============================================
ii  kikit                    1.5.1-1      all          KiKit - Automation for KiCAD
ii  python3-pcbnewtransition 0.4.1-1      all          KiCad 5/6/7/8 pcbnew API compatibility layer
```

Just a thought: would docker somehow help solve all these problems? I'm by all means not an expert, but from what I read it helps dependency-related issues.

set-soft, 2024-04-09:
The versions looks right

> would docker somehow help solve all these problems? I'm by all means not an expert, but from what I read it helps dependency-related issues.

Yes, with docker images you have a container where all the needed dependencies are met. I use them all the time. I.e. I currently have KiCad 7.0.11 installed, so any test using KiCad 8, 6 or 5 is done using docker images.

set-soft, 2024-04-11:
Hi @Unichord !

I see tree paths here:

1. You might want to try docker images, as you suggest in the last message
2. You could try to upgrade your Ubuntu, to a version with more recent Python
3. We can try to modify the KiKit code to get more information about the problem. This is more an issue in KiKit, which likes to mess with pcbnew.

Please let me know so I can decide to keep this issue open or not

Unichord, 2024-04-12:
> Hi @Unichord !
>
> I see tree paths here:
>
> 1. You might want to try docker images, as you suggest in the last message
> 2. You could try to upgrade your Ubuntu, to a version with more recent Python
> 3. We [...]

1. I will look into Docker for sure, is there any tutorial on how to install kibot using Docker? Any "full package" so that I can install that and be sure that everything will work? Maybe I don't understand fully how Docker works...

2. I'm upgrading Ubuntu, so I'll keep you posted on how that works out.

3. Possibly, but that seems like a long and tedious work... With many possible downsides. I'll try and keep this as a last resort.

Thank you for your continued support.

set-soft, 2024-04-12:
> 1. I will look into Docker for sure, is there any tutorial on how to install kibot using Docker? Any "full package" so that I can install that and be sure that everything will work? Maybe I don't [...]

Hmmm ... I don't know about docker tools for Windows, I use the command line tools for linux, but I guess there are simple stuff to install.

After this you'll need to get the docker image for KiBot, the names are explained in the docs.

Then comes: how to use them. I see various users just use a bash shell for Windows and use it from command line, but I don't know if something simpler is available.

> 2. I'm upgrading Ubuntu, so I'll keep you posted on how that works out.
> 3. Possibly, but that seems like a long and tedious work... With many possible downsides. I'll try and keep this as a last resort.

Ok.

Unichord, 2024-04-18:
When upgrading Ubuntu version kibot was uninstalled, so since I have to reinstall everything, I'll try and run kibot with Docker.

I guess I managed to install it (ghcr.io/inti-cmnb/kicad8_auto) in WSL, but I don't understand how to "run" it, or to have it always running so that I can use it from a shell. I tried `docker run ghcr.io/inti-cmnb/kicad8_auto` and it seems to be working, but I cannot use commands like kibot-check or whatever.

set-soft, 2024-04-18:
> When upgrading Ubuntu version kibot was uninstalled, so since I have to reinstall everything, I'll try and run kibot with Docker.

The *kibot* package pulls most dependencies. Don´t know why upgrading Ubuntu uninstalled KiBot, this is not normal.

> I guess I managed to install it (ghcr.io/inti-cmnb/kicad8_auto) in WSL, but I don't understand how to "run" it, or to have it always running so that I can use it from a shell. I tried `docker run [...]

I recommend the full images, like *ghcr.io/inti-cmnb/kicad8_auto:latest*.  Take a look at the scripts in the repo, like [this](https://github.com/INTI-CMNB/kicad_auto_test/blob/master/ki8/run_shell_same_user.sh)

The *--rm* means *remove the instance after finishing*, no background run.
The *-it* means interactive, so you can type stuff and see their output.
The *-v* (aka *--volume*) is used to mount stuff so you see them inside the docker image.
The */bin/bash* at the end is to run a shell, so you can do things.

Note that a lot of users runs the docker images on Windows directly. The above script is for Linux, so the syntax is ok for WSL.

set-soft, 2024-04-18:
> If I understand correctly, I should run this script every time I need to run kibot, am I right?

Not sure I understand. Once you run the script you get a shell inside the docker container, you can run KiBot many times in this shell. But yes, you must start the container at least once.

> >Note that a lot of users runs the docker images on Windows directly.
>
> This would be even better, I installed Docker Desktop for Windows and it seems like it's working too, but I don't know how to properly run it like the script you wrote, and it's documented quite poorly it seems, I have to dig deeper into it.

Ok, the trick here is how to mount the volumes so you can access your project in an easy way. Also how to keep your KiCad preferences synced between the KiCad for Windows and the KiCad inside the docker image.

CRImier, 2024-04-27:
for the reference, I run kicad 8 ppa on 20.04 too and kikit didn't work after update, here's [how I fixed my install](https://crimier.github.io/posts/kicad8-ubuntu-kikit/), partially with help of this post.

Unichord, 2024-05-02:
> Not sure I understand. Once you run the script you get a shell inside the docker container, you can run KiBot many times in this shell. But yes, you must start the container at least once.

Maybe I'm running kibot in an unconventional way? I'll explain my process (or what I want to achieve) to clarify what I mean: once I complete a PCB (which happens quite often, I'm talking about multiple times per week) I run a `.ps1` script located inside the project folder, which runs kibot for the project in which the script itself is located.
This saves a lot of time, instead of opening WSL, navigate to the corresponding folder, launch kibot with the correct parameters.

I'm trying to automate this but due to lack of time lately I didn't yet. But a 1-click script is in my to-do list. Starting docker is a bit slow, though, so I will consider running it in background.

> Ok, the trick here is how to mount the volumes so you can access your project in an easy way. Also how to keep your KiCad preferences synced between the KiCad for Windows and the KiCad inside the [...]

What do you mean? Like relative paths and so on?

set-soft, 2024-05-03:
21a063d4414f198e7012119ff26e0d175cb65f48 implements it in the report. The full template is an example.

Let me know if you can try it and get useful results

closed 2024-05-03 by set-soft

set-soft, 2024-05-03:
> Maybe I'm running kibot in an unconventional way? I'll explain my process (or what I want to achieve) to clarify what I mean: once I complete a PCB (which happens quite often, I'm talking about [...]

The problem comes when you need to iterate. Lets say you don't like the angle of a view, or the order of the BoM columns, or you just realize some data is missing in the components. So you start running KiBot over and over until you get the exact results. Once you have  it done you can run KiBot after each revision of the project.

If you start a docker container for each iteration you'll waste time. But if you open it in an interactive way you can iterate *inside* the container, running a script, or just using the bash history (last command you ran).

> What do you mean? Like relative paths and so on?

Yes, but in general it depends on the layout of your projects, this is something quite personal. 

If you setup KiCad using absolute paths, and even worst, using drive letters, things might get complicated. You might need to have two different configurations: one for Linux and another for Windows, and then make sure both are synced.