INTI-CMNB/KiBot #608: [BUG] KICAD 8 - zoom option has no effect (2D/3D)

mdeweerd, 2024-04-13:
I upgraded a project from KICAD7 to KICAD8.0.1 and used ghcr.io/inti-cmnb/kicad8_auto_full:dev (fresh 2024/04/13).

Changes to the 'zoom' option in render_3d have no effect (the images are regenerated, but the board always has the same size in the image).

Changing the zoom setting in the target below has no effect.
```yaml
  - name: render_3d_top_custom
    type: render_3d
    extends: render_3d_top
    dir: +PCB3D
    disable_run_by_default: render_3d_top
    options:
      zoom: -1
      width: 3840
      # solder_mask: '#1A1A1A'
      move_y: 0
      height: 5120
```

The base configuration it extends from an included configuration file is:
```
  - name: render_3d_top
    type: render_3d
    dir: +PCB3D
    options:
      no_tht: false
      no_smd: false
      no_virtual: true
      background1: '#66667F'
      background2: '#CCCCE5'
      board: '#332B16'
      copper: '#B29C00'
      silk: '#E5E5E5'
      solder_mask: '#143324'
      solder_paste: '#808080'
      move_x: 0
      move_y: 0
      ray_tracing: false
      wait_ray_tracing: 5
      view: top
      zoom: 5
      width: 1200
      height: 720
      orthographic: false
```

set-soft, 2024-04-13:
I can't reproduce it. I tried the following:

```yaml
# Example KiBot config file
kibot:
  version: 1

outputs:
  - name: render_3d_top
    type: render_3d
    dir: +PCB3D
    options:
      no_tht: false
      no_smd: false
      no_virtual: true
      background1: '#66667F'
      background2: '#CCCCE5'
      board: '#332B16'
      copper: '#B29C00'
      silk: '#E5E5E5'
      solder_mask: '#143324'
      solder_paste: '#808080'
      move_x: 0
      move_y: 0
      ray_tracing: false
      wait_ray_tracing: 5
      view: top
      zoom: 5
      width: 1200
      height: 720
      orthographic: false

  - name: render_3d_top_custom
    type: render_3d
    extends: render_3d_top
    dir: +PCB3D
    disable_run_by_default: render_3d_top
    output_id: _2
    options:
      zoom: 2
      width: 3840
      # solder_mask: '#1A1A1A'
      move_y: 0
      height: 5120

  - name: render_3d_top_custom_8
    type: render_3d
    extends: render_3d_top
    dir: +PCB3D
    disable_run_by_default: render_3d_top
    output_id: _8
    options:
      zoom: 8
      width: 3840
      # solder_mask: '#1A1A1A'
      move_y: 0
      height: 5120
```

And got:

[image]

And

[image]

Did you notice what the docs added for KiCad 8? **Note that KiCad 8 starts with a zoom to fit, so you might not even need it**.

The zoom works, is just that the values used for KiCad 7 and 8 are different. This is regrettable, but is how KiCad is developed 🤷🏻‍♂️

If you need more reliable renders use the Blender output, is far better than KiCad rendering, and the KiCad developers told me that this is the way to go for PCB rendering.

set-soft, 2024-04-13:
The previous images are using 1200x720, here are using 3840x5120

[image]

[image]

mdeweerd, 2024-04-13:
Thank you for checking this out.

It may be related to selected layers in the GUI.

I have been doing work on the design re-enabled some layers/objects and I now get the views I expected and the 'zoom' setting is impacting the views - I did not try changing the zoom level yet under this configuration to see if I can change it.  (I've tried quite a few zooms before reporting this).

So I'll have to revert to one of the previous commits, validate that the issue exists and check what is impacting the zoom functionnality.

I confirm that now 'zoom: 0' is a good fit - in KiCAD 7 it was 2.

set-soft, 2024-04-13:
Ok, I'll keep it open while you find more info about it.

set-soft, 2024-05-07:
Hi @mdeweerd !

Any news? Should we close it?

set-soft, 2024-06-25:
Ok, closing it because we don't have more info. Feel free to open a new issue if you found a reproducible problem

closed 2024-06-25 by set-soft