INTI-CMNB/KiBot #609: BOM generation breaks

linuxwanted, 2024-04-15:
Fill as much as you can, remove the rest

**Describe the bug**
HTML and XLSX bom breaks with KiCAD 8 and KiBot 1.6.5 (Ubuntu 22.04/installed via pip) on new designs (some old updated designs KiCAD6->7->8 seems to work) Any Idea how I can debug further, no Idea about Lark ...

**To Reproduce**
Command line options passed to KiBot, configuration files (.kibot.yaml), PCB/Schematic example to reproduce it (Ex: a link to a repo where the files are)
kibot -v
bom.kibot.yaml:

kibot:
  version: 1

global:
  filters:
  - number: 1007
  - number: 1015
  - number: 58

outputs:

- name: basic_ibom
  comment: Interactive HTML BoM
  type: ibom
  dir: bom
  options:
    dark_mode: true
    highlight_pin1: true

# special logos at:
# /media/syncthing/kicad/kibot/IBRW_Logo.png
# /media/syncthing/kicad/kibot/IBRW_Logo2.png

- name: generic_bom_html
  comment: Generic Bill of Materials in HTML format
  type: bom
  dir: bom
  options:
    format: HTML
    group_fields: &id001
    - part
    - part lib
    - value
    - footprint
    - footprint lib
    - voltage
    - tolerance
    - current
    - power
    - 'mfg#'
    count_smd_tht: true
    #component_aliases: [ ['r'], ['l'], ['c'], ['sw'], ['zener'], ['d'] ]
    conf:
      exclude_any: &id002
        - column: References
          regex: '^TP[0-9]*'
        - column: References
          regex: '^FID'
        - column: Part
          regex: 'mount.*hole'
        - column: Part
          regex: 'solder.*bridge'
        - column: Part
          regex: 'test.*point'
        - column: Footprint
          regex: 'test.*point'
        - column: Footprint
          regex: 'mount.*hole'
        - column: Footprint
          regex: 'fiducial'
[... 118 lines cut ...]
  File "/home/ralf/.local/lib/python3.10/site-packages/lark/load_grammar.py", line 491, in <listcomp>
    terminals = [TerminalDef(name, transformer.transform( term_tree ), priority)
  File "/home/ralf/.local/lib/python3.10/site-packages/lark/visitors.py", line 138, in transform
    tree = t.transform(tree)
  File "/home/ralf/.local/lib/python3.10/site-packages/lark/visitors.py", line 84, in transform
    return self._transform_tree(tree)
  File "/home/ralf/.local/lib/python3.10/site-packages/lark/visitors.py", line 80, in _transform_tree
    children = list(self._transform_children(tree.children))
  File "/home/ralf/.local/lib/python3.10/site-packages/lark/visitors.py", line 71, in _transform_children
    yield self._transform_tree(c)
  File "/home/ralf/.local/lib/python3.10/site-packages/lark/visitors.py", line 81, in _transform_tree
    return self._call_userfunc(tree, children)
  File "/home/ralf/.local/lib/python3.10/site-packages/lark/visitors.py", line 47, in _call_userfunc
    return f(children)
  File "/home/ralf/.local/lib/python3.10/site-packages/lark/load_grammar.py", line 418, in expansion
    raise GrammarError("Lark doesn't support joining terminals with conflicting flags!"+str(items))
lark.exceptions.GrammarError: Lark doesn't support joining terminals with conflicting flags!['(?i:(?:(?i:(?:(?i:CAPACITOR)|(?i:CAPA)))|(?i:(?i:C)(?i:(?:(?i:AP))?))))', '(?:(?:[ \t\x0c\r\n])+|[,;])']

**Environment (please complete the following information):**
Where are you running KiBot:
- Using a desktop system? Which OS, distro and version?
KiCAD 8 and KiBot 1.6.5 (Ubuntu 22.04 kibot installed via pip3)

**Additional context**
Add any other context about the problem here.

set-soft, 2024-04-15:
This seems to be some issue with the Lark version you have installed. Try running `kibot-check` to find which version is in use.

Note that the conflict is with some Lark installed at user level (/home/ralf/.local/lib/python3.10/site-packages/lark/load_grammar.py), not an official package. The problem will most probably go away if you uninstall this local copy. The version shipped with Ubuntu 22.04 should work.

set-soft, 2024-04-15:
> * Using a desktop system? Which OS, distro and version?
>   KiCAD 8 and KiBot 1.6.5 (Ubuntu 22.04 kibot installed via pip3)

BTW: Why are you using pip in on a Debian based system? You have Debian packages [here](https://github.com/set-soft/debian). The pip tool is designed to help Windows users and promote experimental chaos on Linux systems, it only woks when using virtual envs, but if you want to waste disk space just use docker images, they work far better.

set-soft, 2024-05-07:
Ok, I tried with the last available version of Lark (1.1.9) and all works perfectly.

So I guess you have a very old Lark installed, just upgrade it.

closed 2024-05-07 by set-soft