INTI-CMNB/KiBot #613: [HOW] Filter after grouping?

mdeweerd, 2024-04-21:
**What do you want to achieve?**

For components that change values/manf# depending on variant, I would like to reuse fields from other components and only filter after those are "reused".

More specifically, there is a RES 100k in the schematic that changes to a RES 10k for a certain variant.   There are already 100k and 10k components of the same type else where which have a "lcsc#" part number.

I use the following to generate the LCSC bom:

``` 
filters:
  - name: only_lcsc_parts
    comment: Only parts with LCSC code
    type: generic
    include_only:
      - column: LCSC#
        regex: ^C(\d+|_TO_SELECT)
  - name: JLCPCB_bom
    comment: BoM for JLCPCB
    type: bom
    dir: +JLCPCB
    options:
      exclude_filter: only_jlc_parts
      ref_separator: ','
      group_fields: [Value, manf#]
      [...]
```

When I do not specify an lcsc# code while expecting to reuse a lcsc# code from another component, it does not appear in the BOM - probably because the filter is applied before.

When I specify the lcsc# code according to the variants, the result is as expected.  Example of a working example:
[image]

Is there a method to apply the filter after the grouping/reuse of the manufacturer code?

set-soft, 2024-04-23:
Hi @mdeweerd !

A small example always help, do you have one?

I think your problem here is the broad filter you are using. How can KiBot know you don't want to exclude the component in question?

I don't see which role plays here a variant, this problem should be similar if no variants are applied. You have a component without *LCSC#* id and you apply a filter to remove such a component ...

I think the solution here is to narrow the filter. You could add another field to mark components you want to include even when *LCSC#* is empty and add a second regex in the *include_only* section to match them. In this way they will be kept, even when *LCSC#* is empty, then the lazy grouping will do the work.

mdeweerd, 2024-04-23:
Test case: I'll see to set one up

> How can KiBot know you don't want to exclude the component in question?

Maybe the exclude_filter should run after resolving the variants.
In my case the variants set this field when appropriate.  So if that is resolved before applying the exclude_filter then it should be ok.

It seems more logical to run the exclude_filter after applying the variant rule.

FYI, I resolved the particular issue in two steps which are different:
1. Set the 'lcsc#' field in the base configuration.
2. Stop filtering on the 'lcsc#' field as the JLCPCB process has evolved and I prefer knowing which components are required in the design but not mountable by JLCPCB.

So in summary, I think the "lesson to learn" from this is to apply the exclude_filter after resolving the variant configuration.

set-soft, 2024-04-23:
> Maybe the exclude_filter should run after resolving the variants.

Such a change in behavior can have really strange impact.

> It seems more logical to run the exclude_filter after applying the variant rule.

Without an example I can just guess you are confusing *variants* and *filters*, the "variant rule" has nothing to do with the filter that renames fields. 

The fields are renamed by the default *pre_transform* filter of the KiCost variant.  This runs before the *exclude_filter*, as its name says. But you could remove it, or add any filter before it.

I think the lesson here is: avoid relying on the lazy grouping.

closed 2024-04-23 by set-soft