INTI-CMNB/KiBot #614: [FEATURE] Add option to disable drawings rendering in render_3d

mdeweerd, 2024-04-21:
With kibot 1.6.6., KiCAD8 the User.Drawings are rendered, which gives unexpected black lines:

[image]

The layer was disable in my GUI.

Maybe there is a way to disable this - IHMO this can be inactive by default and potentially be controlled with an option such as `show_drawings` similar to `show_silkscreen`.

set-soft, 2024-04-22:
Note that this is huge change in KiCad behavior.

KiCad 7:
- Only one option for comments and drawings
- Only one option for eco 1 and 2
- Comments, drawings, eco 1 and eco 2 **NOT** displayed when using realistic colors

KiCad 8:
- Separated options for comments and drawings
- Separated  options for eco 1 and 2
- Comments, drawings, eco 1 and eco 2 displayed when using realistic colors
- Default is to show them if no option is found in the JSON file

mdeweerd, 2024-04-22:
Thank you for the feedback - that does imply a few changes.

I'ld be find by adding what is needed to "the JSON file".  However, I only find notifications.json in the root of the local kicad directory on windows, and packages.json, repository.json in the pcm subdirectories.

I've gone through the preferences for the 3D settings but did not find anything related.

But because I am using docker I guess I need to bind a json file when running in docker.

set-soft, 2024-04-22:
I'm adding the relevant options to KiAuto, but I found KiCad 8 did other changes related to colors

set-soft, 2024-04-23:
The above patch adds more control about what KiCad 8 draws. Needs KiAuto 2.3.1 and should avoid user drawings (added by default by KiCad when the JSON config doesn't explicitly disable them!)

But note that I'm deprecating this output in favor of the Blender output. The KiCad developers said they don't like the idea of having the 3D render in `kicad-cli` and the behavior of the GUI changes all the time. The 3D render isn't intended to be non-interactive.

To show you just one detail, here is the same test case using different KiCad versions:

[image]
[image]
[image]
[image]

Or using the "realistic" mode:

[image]
[image]
[image]
[image]

closed 2024-04-23 by set-soft

mdeweerd, 2024-04-23:
I understand that he 3D render is changing accross versions.
Even if not interactive, the 3D snapshot gives a better idea about the depth of components vs. the 2D image (=orthogonal 3D) which is also created using the same method so I guess it would suffer from the same issues.

The 2D image is useful to compare with the real board.  I hope that this is not deprecated.

set-soft, 2024-04-23:
Try using an *orthographic* camera in Blender to get a pseudo-2D image.

The deprecation means that I won't give priority to this output, and won't worry much if it has errors or limitations.

If something is missing in the Blender Export output we should add it, but we can't currently rely on the KiCad renderer, and this is the opinion of the KiCad developers.