INTI-CMNB/KiBot #615: [FEATURE] kibot_config for ouptut `type: kicost`

mdeweerd, 2024-04-27:
Is/can the 'kicost_config' config option (be made) available on the `type: kicost` output?

(`type: bom` with kicost set to true does not produce a result when there are multiple components for a single reference, without `kicost_config` there is no cache).

mdeweerd, 2024-05-02:
I managed to setup the subparts filter, thanks for the hint.

(Now I need to work out why  'uk.farnell.com' is consulted despite having set the following in the referenced kicost_config.yaml)
```yaml
APIs:
  TME:
    country: FR
    language: EN
  Mouser:
    country: FR
  Element14:
    farnell_country: FR
```
)

set-soft, 2024-05-03:
Look in the log for the message "# Getting part data from Element14", it should contain the country used.

mdeweerd, 2024-05-03:
@set-soft - thanks for following up.

I am thinking about how to best cope with/report the issue I seem to have:
- The process became a lot slower since I added the splitparts (I can drink several coffees while waiting for one generation - before not even one coffee).
- The behavior is different when I include the 'bom:' target vs. when it is defined in the root yaml.
- I get 'Server error `Not Acceptable` (406)' fpr TME when the file is not included, and no error when running kicost.  When I looked a while back this was because parts were not split, but now they are split.
- I add kicost_api_disable on tem, mouser, etc, but they are still being queried.

To validat the the slowness comes from adding subparts_splitter, I need to test again after removing the subparts_splitter filter shown below:

```yaml
filters:
  - name: kicost_rename
    comment: Variant rename KiCost style
    type: var_rename_kicost
    # Empty will default to :
    separator: ''
    variant_to_value: true
  - name: subparts_splitter
    type: subparts
    # We want to also split the `Description` field
    split_fields: ['Description','comment']
    split_fields_expand: true
    separators: ';'

# And all the variants use:
    pre_transform: [kicost_rename,subparts_splitter]
```

The issues with distributors are puzzling also because the configuration seems to be loaded, some excerpts when using '-vvvv' for high verbosity.

It looks like I am overlooking something small, but what I checked seems to be confirmed ok.

````plaintext
# Confirms the configuration is loaded.
^[[36m^[[1mDEBUG:Loading config from: /root/workdir/LIBS/kicad-redacted-lib/kibot/kicost_config.yaml (kibot.kicost - config.py:163)^[[0m^M
^[[33m^[[1mWARNING:(WC018) Unknown option `country` for API `Mouser` (kibot.kicost - config.py:120)^[[0m^M 
^[[36m^[[1mDEBUG:Loaded API options (kibot.kicost - config.py:145)^[[0m^M
^[[36m^[[1mDEBUG:- TME: (kibot.kicost - config.py:147)^[[0m^M
^[[36m^[[1mDEBUG:  - country: FR (kibot.kicost - config.py:151)^[[0m^M
^[[36m^[[1mDEBUG:  - language: EN (kibot.kicost - config.py:151)^[[0m^M
^[[36m^[[1mDEBUG:- Mouser: (kibot.kicost - config.py:147)^[[0m^M 
^[[36m^[[1mDEBUG:  - country: FR (kibot.kicost - config.py:151)^[[0m^M
^[[36m^[[1mDEBUG:- Element14: (kibot.kicost - config.py:147)^[[0m^M
^[[36m^[[1mDEBUG:  - farnell_country: FR (kibot.kicost - config.py:151)^[[0m^M

# Confirms FR later
^[[36m^[[1mDEBUG:API options with defaults (kibot.kicost - config.py:145)^[[0m^M
[...]
^[[36m^[[1mDEBUG:- Element14: (kibot.kicost - config.py:147)^[[0m^M
^[[36m^[[1mDEBUG:  - farnell_country: FR (kibot.kicost - config.py:151)^[[0m^M
^[[36m^[[1mDEBUG:  - key: mXXXXXXXXXXXXXXXXXXXXXXs (kibot.kicost - config.py:151)^[[0m^M
^[[36m^[[1mDEBUG:  - cache_ttl: 7 (kibot.kicost - config.py:151)^[[0m^M
^[[36m^[[1mDEBUG:  - cache_path: /root/kicost_cache/Element14 (kibot.kicost - config.py:151)^[[0m^M

# Uses uk
^[[36m^[[1mDEBUG:Considering: Element14 ['farnell', 'newark'] (kibot.kicost.dist - distributor.py:159)^[[0m^M
^[[37m - Element14 [api] (https://partner.element14.com/)^[[0m^M
^[[36m^[[1mDEBUG:# Getting part data from Element14 (farnell uk)... (kibot.kicost.dist - api_element14.py:265)^[[0m^M

# Currently TME is blocking

DEBUG:Searching P/N: 1728284 61900811121 691412120008B 1727078 2211S-03G SSW-112-01-T-S from Phoenix Wurth Elektronik Multicomp Samtec (kibot.kicost.dist - api_tme.py:326)
DEBUG:Looking in cache /root/kicost_cache/TME/search_1728284 61900811121 691412120008B 1727078 2211S-03G SSW-112-01-T-S_FR_en_USD.dat (kibot.kicost.dist - distributor.py:92)
DEBUG:TME json request endpoint /Products/Search params {'Country': 'FR', 'Language': 'en', 'SearchPlain': '1728284-61900811121-691412120008B-1727078-2211S-03G-SSW-112-01-T-S', 'SearchOrder': 'PRICE_FIRST_QUANTITY', 'SearchOrderType': 'ASC'} (kibot.kicost.dist - api_tme.py:132)
Progress:  41% 9/22 [00:00<00:00, 32.03part/s] 
ERROR:Trace stack: (kibot.gs - gs.py:802)
  File "/usr/lib/python3/dist-packages/kibot/bom/xlsx_writer.py", line 710, in create_kicost_sheet
    return _create_kicost_sheet(workbook, groups, image_data, fmt_title, fmt_info, fmt_subtitle, fmt_head, fmt_cols, cfg)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 26 lines cut ...]
DEBUG:Query params: {'callInfo.apiKey': 'mXXXXXXXXXXXXXXXXXXXXXXs',
 'callInfo.responseDataFormat': 'JSON',
 'resultsSettings.responseGroup': 'large',
 'storeInfo.id': 'uk.farnell.com',
 'term': 'manuPartNum:MB05S'} (kibot.kicost.dist - api_element14.py:191)
Progress:  24% 13/54 [00:08<00:26,  1.54part/s]
ERROR:Trace stack: (kibot.gs - gs.py:822)
  File "/usr/local/lib/python3.11/dist-packages/kibot/bom/xlsx_writer.py", line 731, in create_kicost_sheet
    return _create_kicost_sheet(workbook, groups, image_data, fmt_title, fmt_info, fmt_subtitle, fmt_head, fmt_cols, cfg)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/bom/xlsx_writer.py", line 686, in _create_kicost_sheet
    query_part_info(parts, dist_list)
  File "/usr/lib/python3/dist-packages/kicost/kicost.py", line 76, in query_part_info
    get_dist_parts_info(parts, dist_list, currency)
  File "/usr/lib/python3/dist-packages/kicost/distributors/__init__.py", line 50, in get_dist_parts_info
    distributor_class.get_dist_parts_info(parts, dist_list, currency)
[... cut ...] ackages/kicost/distributors/distributor.py", line 166, in get_dist_parts_info
    solved = api.query_part_info(parts, list(remaining), currency)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/distributors/api_element14.py", line 373, in query_part_info
    raise KiCostError(msg, ERR_SCRAPE)
ERROR:KiCost error: `Server error `Forbidden` (403)` (5) (kibot.gs - gs.py:836)
```

When running the kicost output I do not get these issues.

closed 2024-05-07 by set-soft

mdeweerd, 2024-05-07:
I'll open a new issue when the time comes - I tested with the update and it's referring to the french farnell site now, but still running in an issue;

The slowness occurs for every target, not only kicost related targets - I think the components for the variants are computed again for each output and I'ld say it progresses at about 3 components per second.
I'll try to throw a python line profiler at it.

At the time I made the tests dev was at 1.6.6 and a relase at 1.6.5 .

This is a partial output for the bom output with kicost, failing on the second component (Progress:   5% 2/39 [00:01<00:19,  1.93part/s]
) with a 'forbidden' result.

```
DEBUG:* Data found: (kibot.kicost.dist - api_element14.py:300)
DEBUG:{'attributes': [{'attributeLabel': 'tariffCode', 'attributeValue': '85322900'},
                {'attributeLabel': 'Longueur du produit',
                 'attributeUnit': 'mm',
                 'attributeValue': '6.5'},
                {'attributeLabel': 'rohsCompliant', 'attributeValue': 'YES'},
                {'attributeLabel': 'Humidité',
                 'attributeValue': 'GRADE II (Condition Test A)'},
                {'attributeLabel': 'hazardous', 'attributeValue': 'false'},
                {'attributeLabel': 'rohsPhthalatesCompliant',
                 'attributeValue': 'YES'},
                {'attributeLabel': 'Température de fonctionnement max..',
                 'attributeUnit': '°C',
                 'attributeValue': '125'},
                {'attributeLabel': 'Montage du condensateur',
                 'attributeValue': 'Traversant'},
                {'attributeLabel': 'Gamme de produit',
                 'attributeValue': 'B32529'},
                {'attributeLabel': 'Largeur du produit',
                 'attributeUnit': 'mm',
                 'attributeValue': '7.2'},
                {'attributeLabel': 'usEccn', 'attributeValue': 'EAR99'},
                {'attributeLabel': 'Hauteur du produit',
                 'attributeUnit': 'mm',
                 'attributeValue': '6.5'},
                {'attributeLabel': 'Tolérance de capacité',
                 'attributeValue': '0'},
                {'attributeLabel': 'Bornes de condensateur',
                 'attributeValue': 'À sorties radiales'},
                {'attributeLabel': "Température\xa0d'utilisation min",
                 'attributeUnit': '°C',
                 'attributeValue': '-55'},
                {'attributeLabel': 'euEccn', 'attributeValue': 'NLR'},
                {'attributeLabel': 'Type de diélectrique',
                 'attributeValue': 'PET Métallisé empilé'},
                {'attributeLabel': 'Capacité',
                 'attributeUnit': 'pF',
                 'attributeValue': '2200'},
                {'attributeLabel': 'Pas',
                 'attributeUnit': 'mm',
                 'attributeValue': '5'},
                {'attributeLabel': 'Tension (DC)',
                 'attributeUnit': 'V',
                 'attributeValue': '63'},
                {'attributeLabel': 'productTraceability',
                 'attributeValue': 'No'},
                {'attributeLabel': 'Tension (AC)',
                 'attributeUnit': 'V',
                 'attributeValue': '40'},
                {'attributeLabel': 'Boîtier de condensateur',
                 'attributeValue': 'Coffret Radial, 2 broches'},
                {'attributeLabel': 'SVHC', 'attributeValue': 'To Be Advised'}],
 'brandId': '1000611',
 'brandName': 'EPCOS',
 'comingSoon': False,
 'commodityClassCode': '017007001',
 'countryOfOrigin': 'BR',
 'datasheets': [{'description': 'Technical Data Sheet (972.49KB) EN',
                 'type': 'T',
[... 141 lines cut ...]
DEBUG:Query params: {'callInfo.apiKey': 'mXXXXXXXXXXXXXXXXXXXXXXs',
 'callInfo.responseDataFormat': 'JSON',
 'resultsSettings.responseGroup': 'large',
 'storeInfo.id': 'fr.farnell.com',
 'term': 'manuPartNum:C0603C103J5RAC'} (kibot.kicost.dist - api_element14.py:191)
Progress:   5% 2/39 [00:01<00:19,  1.93part/s]
ERROR:Trace stack: (kibot.gs - gs.py:822)
  File "/usr/local/lib/python3.11/dist-packages/kibot/bom/xlsx_writer.py", line 731, in create_kicost_sheet
    return _create_kicost_sheet(workbook, groups, image_data, fmt_title, fmt_info, fmt_subtitle, fmt_head, fmt_cols, cfg)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/bom/xlsx_writer.py", line 686, in _create_kicost_sheet
    query_part_info(parts, dist_list)
  File "/usr/lib/python3/dist-packages/kicost/kicost.py", line 76, in query_part_info
    get_dist_parts_info(parts, dist_list, currency)
  File "/usr/lib/python3/dist-packages/kicost/distributors/__init__.py", line 50, in get_dist_parts_info
    distributor_class.get_dist_parts_info(parts, dist_list, currency)
  File "/usr/lib/python3/dist-packages/kicost/distributors/distributor.py", line 166, in get_dist_parts_info
    solved = api.query_part_info(parts, list(remaining), currency)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kicost/distributors/api_element14.py", line 375, in query_part_info
    raise KiCostError(msg, ERR_SCRAPE)
ERROR:KiCost error: `Server error `Forbidden` (403)` (5) (kibot.gs - gs.py:836)
```

It's also strange that now the kicost output is failing as well (it seems to consult uk.farnell.com) implying that no kicost relevant output is working anymore (with the current parameter set).

set-soft, 2024-05-07:
> The slowness occurs for every target, not only kicost related targets - I think the components for the variants are computed again for each output and I'ld say it progresses at about 3 components per second.
> I'll try to throw a python line profiler at it.

I never saw such a slow processing. But for boards that takes some measurable time to load (many seconds or even minutes) the best way to deal with variants is to first create them, and in another call to KiBot generate the outputs for the variant avoiding variants and filters. Some work is still needed to avoid loading the PCB/Schematic more than one, even when no variants/filters are used, but things are much faster anyways.

> It's also strange that now the kicost output is failing as well (it seems to consult uk.farnell.com) implying that no kicost relevant output is working anymore (with the current parameter set).

I suspect this is related to some data rate limitation from the server. Perhaps we are asking too fast. What happens if you try again? This component succeeds and the next fails?

commit 878e07d8 (2024-05-07): [KiCost][Added] Option to specify a configuration file

Closes #615
commit 465b6da8 (2024-09-18): [KiCost][Added] Option to specify a configuration file

Closes #615