INTI-CMNB/KiBot #616: [FEATURE] Solder paste usage in report

N0cynym, 2024-04-28:
I would like to ask for a feature which could print a output like below to the kibot report:

```
Metal Amount: 			87.75 %
Stencil Thickness: 		0.12 mm
Solder Paste Gravity: 	4.15 g/cm3
------------------------------------
Total Number of Pads: 	262
Pad Area - Front: 		152.89 mm2
Pad Area - Back: 		356.70 mm2
------------------------------------
Solder Paste - Front: 	0.76 g
Solder Paste - Back: 	1.78 g
Solder Paste - Total: 	2.54 g
------------------------------------
```

I found a GitHub project called [kicad-solderpaste-usage](https://github.com/mfussi/kicad-solderpaste-usage) which generates this output.

The calculation is based on [this blog post](https://www.indium.com/blog/calculating-solder-paste-usage.php#ixzz246lOB9HY).

I think this could be usefull for people who need to do material analytics/reports of their PCBs like for example in case of adding their PCB to [IMDS](https://en.wikipedia.org/wiki/International_Material_Data_System).

set-soft, 2024-05-03:
Hi @N0cynym !

Hmmm ... I tried the script and found it quite faulty.

1. Only supports KiCad 5
2. Only supports rectangular and circular pads (oval and rounded rectangles are approximated, others are ignored)
3. The SMD detection is failing, at least in the released JAR (the code looks ok). And isn't a good idea, it should check the paste layers.
4. The current parameters related to the solder paste aren't computed (margin and margin ratio)

I'm doing an implementation inside KiBot that solves 1, 2 and 3. But I think 4 can be a problem.

N0cynym, 2024-05-03:
Thanks for having a look into this issue.

At least I think a layer area calculation -> material volume calculation would be very useful.
With this information the calculation of the layer material wight could be done later based on the material information you haven't in the board or schematics.

If it's possible to add layer area/volume calculation to the reports than we'll have a good benefit to do the rest.

set-soft, 2024-05-03:
Try the report output from the *dev* branch, the full report template will include solder paste estimation.

The 21a063d4414f198e7012119ff26e0d175cb65f48 patch added it.

closed 2024-05-03 by set-soft

N0cynym, 2024-05-03:
@set-soft thanks for implementing this feature! 🥳 

I can confirm it works! 👍 

[image]

N0cynym, 2024-05-03:
@set-soft I realized that variants seems to be ignored 🤔 

Means pad count with solder paste is  every time the same number!

So if I make a variant without all components (excludes all components) it'll still add paste information to the table but have no data in the paste gerbers!!!

set-soft, 2024-05-03:
If you take a look at the [report](https://kibot.readthedocs.io/en/latest/configuration/outputs/report.html) options you'll notice no variant options are available.

So far we never needed variants for the report. Will see if I can add them.

N0cynym, 2024-05-03:
I just realiced this. The reason why I was mention this is because I make one document using pdfunite out of my prints and exports (including report)...

Would be awsome if report export could care about variants 👍

set-soft, 2024-05-06:
The above patch adds variants

N0cynym, 2024-05-07:
I can confirm that this works  ☺️👍