INTI-CMNB/KiBot #618: [BUG] Variant value overwrite causes conflict with blender export

N0cynym, 2024-05-02:
**Describe the bug**

Blender export fails if you add in the schematic variant replacement values for components like: `variant_name:Value -> 123`

Kibot log:

```
WARNING:(W135) Value field mismatch for `X1` (SCH: `123` PCB: `~`) (kibot - kiplot.py:358)
- 'Exports the PCB in blender format' (blender_3d_export_bottom) [blender_export]
ERROR:Unable to generate `/my/layout/export/path/project_name_variant_name_pcb3d/pads/__X1_125_0` from 'Pads information and board bounds for PCB3D 2.1' (_PCB2Blender_tools_2_1) [pcb2blender_tools] (kibot.gs - gs.py:836)
```

**To Reproduce**

Create schematic with components where you replace Value.

Export blender layer where this component is placed.

```
- "kibot `#option for verbose = -v`
             -g variant=variant_name
             -s run_drc,run_erc blender_3d_export"
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD  `GitLab` workflow
- Which tag are you using? -> [1.7.0-1_k8.0.1_d_sid_b3.5.1](https://github.com/inti-cmnb/kicad_auto_test/pkgs/container/kicad8_auto_full/207446220?tag=1.7.0-1_k8.0.1_d_sid_b3.5.1)

set-soft, 2024-05-02:
Can you post a small example?

N0cynym, 2024-05-03:
> Can you post a small example?

See this files:

[kicad-ci-tests-618.zip](https://github.com/INTI-CMNB/KiBot/files/15196301/kicad-ci-tests-618.zip)

Kibot config:

```yaml
  - name: 'blender_3d_export_top'
    comment: "Exports the PCB in blender format"
    type: blender_export
    disable_run_by_default: _PCB2Blender_2_1
    options:
      camera:
        name: MyCamera
        pos_x: 0
        pos_y: 0
        pos_z: 0.15
      light:
        name: MyLight
        pos_x: -0.15
        pos_y: 0.1
        pos_z: 0.15
      render_options:
        transparent_background: true
        samples: 1
        resolution_x: 3840
        resolution_y: 2160
        view: top
      outputs:
        - type: blender
        - type: render

  - name: 'blender_3d_export_bottom'
    comment: "Exports the PCB in blender format"
    type: blender_export
    disable_run_by_default: _PCB2Blender_2_1
    options:
      pcb3d: _PCB2Blender_2_1
      camera:
        name: MyCamera
        pos_x: -0.0575
        pos_y: 0
        pos_z: 0.15
      light:
        name: MyLight
        pos_x: -0.15
        pos_y: 0.1
        pos_z: 0.15
      render_options:
        transparent_background: true
        samples: 1
        resolution_x: 3840
        resolution_y: 2160
        view: bottom
      outputs:
        - type: blender
        - type: render
```

Kibot call:

```
- "kibot -v
             -g variant=variant_1
             -s all blender_3d_export_top
                    blender_3d_export_bottom"
```

N0cynym, 2024-05-03:
This is the error message at least I get with this setup:

```
ERROR:Unable to generate `/path/of/kibot/export/files/kicad-ci-tests_variant_1_pcb3d/pads/__R1_3_0` from 'Pads information and board bounds for PCB3D 2.1' (_PCB2Blender_tools_2_1) [pcb2blender_tools] (kibot.gs - gs.py:836)
```

N0cynym, 2024-05-03:
Here's my kibot configuration:

- [kicad-ci-tests-618.kiplot.zip](https://github.com/INTI-CMNB/KiBot/files/15200420/kicad-ci-tests-618.kiplot.zip)

set-soft, 2024-05-03:
Ok, now it works even when using the *pcb3d: _PCB2Blender_2_7*

commit 48ceff49 (2024-05-03): [PCB2Blender_Tools][Fixed] transform filters might make it fail

Fixes #618
closed 2024-05-03 by set-soft