INTI-CMNB/KiBot #621: [BUG] ERROR:Unknown symbol attribute

nopeppermint, 2024-05-22:
**Describe the bug**
I would like to generate an assembly plan with DNP components marked with a cross.

This works with the project https://github.com/INTI-CMNB/kibot_variants_arduprog the following settings:

kibot -c test.kibot.yaml (executed in a gitlab runner)

test.kibot.yaml:
```
kibot:
    version: 1
    
outputs:     
  - name: print_pdf
    comment: "PDF for the PCB"
    type: pcb_print
    dir: Generated
    options:
      force_edge_cuts: true
      keep_temporal_files: false
      dnf_filter: _kibom_dnf_Config
      scaling: 2.0
      pages:
        - layers: [ F.Silkscreen, F.Fab ]
          sheet: 'Assembly Top'
        - layers: [ B.Silkscreen, B.Fab ]
          sheet: 'Assembly Bottom'
          mirror: true     
```

R1 is not assembled:

[image]

[t1-assembly (4).pdf](https://github.com/INTI-CMNB/KiBot/files/15403030/t1-assembly.4.pdf)

The Problem is, now that I changed to my own proprietary project which I can not share I got an error while parsing the schematic.
(Even though I am generating an assembly plan..)

```
ERROR:Unknown symbol attribute `[Symbol('text_box'), 'Bias & Int. Supply', [Symbol('at'), -11.43, 29.21, 0], [Symbol('size'), 8.255, -6.35], [Symbol('stroke'), [Symbol('width'), 0], [Symbol('type'), Symbol('default')]], [Symbol('fill'), [Symbol('type'), Symbol('none')]], [Symbol('effects'), [Symbol('font'), [Symbol('size'), 1.27, 1.27]]]]` (kibot.gs - gs.py:834)
```

I isolated the error source down to a single schematic symbol:
[image]

[test_schematic.zip](https://github.com/INTI-CMNB/KiBot/files/15403155/test_schematic.zip)

**Expected behavior**
- not caring about schematic when generating pcb pdf exports
- or supporting this symbol attribute

**Environment **
Where are you running KiBot:
- In a CI/CD workflow GitLab
- Which tag are you using? ghcr.io/inti-cmnb/kicad8_auto_full:latest

nopeppermint, 2024-05-22:
it might be the "&" symbol which cause problems

nopeppermint, 2024-05-22:
like this

[image]

it's working without error

[working.zip](https://github.com/INTI-CMNB/KiBot/files/15403379/working.zip)

so the "bug" seems to be the not supported "text boxes" inside schematic symbols.

nopeppermint, 2024-05-22:
btw. is there an option to change the color of the crosses to red?

[image]
[image]

set-soft, 2024-05-22:
> btw. is there an option to change the color of the crosses to red (instead of the layer color)?

Nope, they are part of the layer text. 
Adding such an irregular thing is complex (more than one color for things in the same layer), but feel free to add feature request about it.

set-soft, 2024-05-22:
> so the "bug" seems to be the not supported "text boxes" inside schematic symbols.

Patch ba0df187f7ede6504a820bd108b241c4145e082e adds support for this.

closed 2024-05-22 by set-soft

nopeppermint, 2024-05-22:
thank you very much @set-soft

commit ba0df187 (2024-05-22): [Schematic][Added] Support for text boxes inside symbols

Closes #621