INTI-CMNB/KiBot #623: [BUG]

nopeppermint, 2024-05-22:
**Describe the bug**

I tried to generate an assembly plan with the exact same setup as in issue https://github.com/INTI-CMNB/KiBot/issues/621 but this time with my custom kicad template.

I got this error: "Unknown bitmap attribute `[Symbol('data'),"
My custom template contains two embedded bitmaps, I imagine they fail:

```
	(bitmap
		(name "")
		(pos 13.749 20.732)
		(scale 0.5)
		(data "iVBORw0KGgoAAAANSUhEUgAAAgAAAABNCAMAAADn/yxTAAAAIGNIUk0AAHomAACAhAAA+gAAAIDo"
```

```
ERROR:In section 'print_pdf' (pcb_print): Error reading `TEMPLATE_NAME.kicad_wks` (Unknown bitmap attribute `[Symbol('data'), 'iVBORw0KGgoAAAANSUhEUgAAAgAAAABNCAMAAADn/yxTAAAAIGNIUk0AAHomAACAhAAA+gAAAIDo', 'AAB1MAAA6mAAADqYAAAXcJy6UTwAAABmUExURf///+Tk5evr6/n5+cjJyrq7vK2tr6anqJGSlJiZ', 'm9bW18HCw8/Q0PHy98bK27e90ouUt32HrlFfk0NRihcpbyY2eDREgW56pZqiwNTY5J+gofLy8t3d', '3uPl7amvyWBsnLS0tv///9Yi6nwAAAABdFJOUwBA5thmAAAAAWJLR0QAiAUdSAAAAAd0SU1FB+gC', 'DA4YFJSKvrkAAAdUSURBVHja7V1pY6IwEBU5lEvxalXU8v9/5Yq2TQIJuWYgdPM+ut0ZZublJYQc', 'i4UIgTqWC48/gzCI4mS11kWaJFEWaPsKoijKE1U8/7gINJ2Um21V7fZDOBx5eP5+Ev+f05GPD9F/', '+Kyq6ry5SJ42iFdpUoTgmctfmZPbDfKrdukZJEWtVpc6yxNTJ2mS3+SxPHF/fH0cHcPh83EXPnD8', 'DvA6JKk2mVsneSbOXBhZVv/bh7yJ1oW+xPTcZBIOlOeDfbmQSPAouY+c/1JcFFsIkLmVQGGKFKL8', 'r9oMjwnqGMZNGg1RoDrZFwoPp4qXGBJbjJu5uJ+52p5ZFKKBykRwbtKbyMnF2db/g8NlKDMpN3Ng', 'TXSdFh3bSzjbL/AZ3GoYKNFEfi5ON/9vPLpP3VCB9YdSoXnPz0PDiAB0/YWVAa7/s0PjdQOzqH+f', 'AXSFg1EzF4KM/ljko9Sfy7RyHvU/HrfqBEDIXMLXHjDwXgYwHPXHG19TF1YVp1KZALlxfsT4baMB', 'Rv3X1379byiOut3lZuq6quNTlQA4JVpyHAMi6xEAoadZM1L2wn7qsmrgrkgA1MzhsIsjARmSI3ba', 'YUYCcDzu1AiAlbm3F6DZBVlhFgv4ccwb7Djwc+qi6uBDjQBIGr1uXtbBXwF/0HkRWGL5YWdN5vIK', '8MZFhQC1cWpkCBF7gF4fADgF2AE9ITirHuC4K1UIUKBlLmPqkip9YpOB6q/Y8TmJ7xpBgPihpWa7', 't0NHQE6W5oaNbwUJ6hCgAc4c0fyYcZsvQEDMs1P1LO/sQcYuK5gHb1GxJTrc7U0qGxcSgGS00HEn', 'BJlUuDLWgRb2EL4yczTUEABEaWiDMA++WJSdV8idvUkN4yIC1IiZC+mGCRQo6bEa+mfSNSSGhrtI', 'uemywOY0pNDoxkUECMAzd6X8wFsXWCR9dmRqWZiwm72xxbTyz8bDEoBkDqiTpjwVCAQQSDPpsmGG', 'AHRXBkGpaeW/hZwAMEMAtjEWoFl8gU+AhBscVBjWmFj+2QyxOcLMXIOgzBMQwF68UOV/xxrfC4yP', 'R4AblTlUAtAPTIZsMENZNgw73DHl/95ZnlaJ/nA8AgRTEICvC1BhWGHbUegN1BPqGRcRgDQdxTX3', 'WpnzBOgpdGllzdy4iACYmbuKpm1sMC8CKCs0uvEpCLCmvP6fBHBE/lt4AhiHYU4AV+S/hSeAcRim', 'BHBH/lt4AhiHYUgAVPl/sMY/5MY9AYzDMCJAiSn/ZWdl2peC8akJgACXCdDdPniGeixj454AxmEY', 'EKCr0LKjG0Yw7glgHIY2AUwUGt24J4BxGLoEcE/+W3gCGIehSQAH5b+FJ4BxGFoEcFL+W3gCGIeh', 'Q4Ax5f+kZdwTwDgMDQKcMeX/zHLroGd8agJclU+dk8BhApSdswM+QeXf0vjUBPgPZgIv7LGBp4fi', '/xvHuCeAcRiKBLBTaHTjngDGYSgRwGn5b+EJYByGCgHclv8WngDGYSgQwHH5b+EJYByGlACjyv/O', '0LgngHEYMgJsMOUfbE+RJ4BxGBICTL3rUxGeAMZhDBJg+l2fipiGAH9+Y4gDuz4VMc3GkL9OgJnI', 'fws5AaAuZxptaxj/V6gdboWcALOR/xYiAly5v9qAT4AGyHrIJwB8GPLt4fOR/xbj7Q6mmw7cHusf', 'BHwCTHBAxJjyv7c3LicAzFE47AER9HgABoLToOhzaWAg2dfqxqZ/DYgIAL9/lz5chzqDDKhvJtYb', '/s9QRx2Rs4d5LcOhXZ+KkJ8RFBvbFnkq4M9vpOrCEJaEAXSuY8hP1zdc2vWpCBEBiKRCiTTjh7iF', 'GQRQh4QxDZMaGsBIDXUmbe/f3Nr1qQj5OYFA74HUxR0hcxlJDnEQbSIoNNVgYboy4qgnKfOTfzag', 'jqaRn2H6AGoZ4ALrHhdOXbCuC+inpaPQLuNLhQB4C3eb1vxY9wVg3Hv0BjsGvDt/aSSFswoB8A7a', 'fw37xroxBO/CCObcue2sboy4qxAALXPvE/uwbozo9cw4Nx91e4AZ6b/ylTFYvWeD2sX03ipxrz76', '7QGmLqoOzmoEQM4cjsJwXir9tXFdlGoEwM4cyvCM88o6zsWRl6mrqg52dnr0iyOJE4Q+hjvhjzHe', '7E8qVPaVGQcHR66OxbiamD9jgXAFcsNxM5NO4NRZPT54ezh85laYlRHNWI1zfXw5j5kAd66Pb+1D', 'qnMq/qwUwr5yxPyj52fBgN7adPpu9X5cIezV603PQwA20kwGP/ZEcDOPqXhlgfOzAZy9CdT9kCkv', 'pgI7cxlI64xlC35qKLGJh26e6OwFcQ28BUTU0gz+5zIwmY5FTbQu7DiQNpnKdSB1Ya82q0Lm6eFu', 'P/DF/4L4O+d/FcUWFvZDgWs0/Dl+mUVxosuDNEny6Kbxnb8uYvNQVnGm5Opy3rv3YWC/2wrXj3wz', 'YDUUXZ1ZZa7QKNIyUIL58o46uEVRSzdFxM+/vmn6Kzeb6gnQK4C5oErMx/MpthvJ2qE6T5JGYWWW', 'buYaYeb+AUnjjD4cjZ9CAAAAJXRFWHRkYXRlOmNyZWF0ZQAyMDI0LTAyLTA3VDE0OjQwOjIzKzAw', 'OjAwR9Lu9AAAACV0RVh0ZGF0ZTptb2RpZnkAMjAyNC0wMi0xMlQxNDoxNzoyMCswMDowMGkT7bsA', 'AAAodEVYdGRhdGU6dGltZXN0YW1wADIwMjQtMDItMTJUMTQ6MjQ6MTArMDA6MDDCXBaFAAAAGXRF', 'WHRTb2Z0d2FyZQBBZG9iZSBJbWFnZVJlYWR5ccllPAAAAABJRU5ErkJggg==']`) (kibot.gs - gs.py:836)

```

**To Reproduce**
use attached kicad template for your pcb file, it should fail with the same error as above.
[failing_template.zip](https://github.com/INTI-CMNB/KiBot/files/15404500/failing_template.zip)

**Expected behavior**
- pcb templates with embedded pictures should not throw an error

**Environment (please complete the following information):**
Where are you running KiBot: => see https://github.com/INTI-CMNB/KiBot/issues/621

set-soft, 2024-05-22:
> * pcb templates with embedded pictures should not throw an error

I agree, but is impossible to support undocumented features, take a look at the [docs](https://dev-docs.kicad.org/en/file-formats/sexpr-worksheet/). The documented symbol is *pngdata*, looks like KiCad 8 is using a new symbol called *data*.

So this is not a bug, we need to support an undocumented feature. Will see it after lunch.

nopeppermint, 2024-05-22:
I see, it's more a kicad documentation bug in a first step..

nopeppermint, 2024-05-22:
pngdata data btw. does not work anymore either, I just tried to replace "data" with "pngdata" in the template itself :-(

[image]

nopeppermint, 2024-05-22:
bug report for the documentation: https://gitlab.com/kicad/services/kicad-dev-docs/-/issues/34

set-soft, 2024-05-22:
Ok, the above patch should solve the problem.
But one issue remains: PNGs with "pHYS" chunks aren't correctly scaled. So your "joysteer" logo will look too small. 
I'll try to fix it tomorrow.

closed 2024-05-22 by set-soft

set-soft, 2024-05-23:
And this patch should solve the size of the PNG.
In this case it specifies 72 PPI instead of the default 300 PPI assumed when no PPI info is available.

commit 1f75300e (2024-05-22): [Worksheet][Added] Support for KiCad 8 bitmaps

Closes #623