INTI-CMNB/KiBot #627: [HOW] variable expansion problem when using NET_NAME

MarkusSBS, 2024-06-06:
**What do you want to achieve?**
I want to expand the value of a testpoint to the netname. something simulate is done in https://github.com/INTI-CMNB/KiBot/issues/511 and there should be a workaround but it does not work for me.

`  - name: tp_pos
    comment: "TestPoint Position and netname"
    type: position
    dir: PCB_Docu
    output_id: 'TestPoints'
    options:
      output: "${SBS_SAP}_%p-%v_Rev%r-%i_${SBS_Doc_status}_%D-${git_hash}.%x"
      format: ASCII
      units: millimeters
      only_smd: false
      use_aux_axis_as_origin: true
      include_virtual: true
      separate_files_for_front_and_back: false
      pre_transform:
        - "expand_var"
      dnf_filter:
        - "only_tp"`

`  # Select only testpoints
  - name: "only_tp"
    type: generic
    comment: "Select only testpoints"
    include_only:
      - column: References
        regex: '^[t,P][p,P]\d+' 
  
  - name: "expand_var"
    type: expand_text_vars
    comment: "Expand all text variable to their value"
    include_kicad_env: true
    include_os_env: false`

`kibot -d Docu -vvv -s run_drc,run_erc tp_pos`

I can not find any problem in the log 

```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (`GitLab`)
- Using a docker image? (ghcr.io/inti-cmnb/kicad7_auto_full:dev)

**Additional context**
Add any other context about your setup/project.

set-soft, 2024-06-06:
Hallo @MarkusSBS !
Do you have a simple test case? (small schematic, PCB and KiBot config)

set-soft, 2024-06-06:
Hi @MarkusSBS !
The problem is that you must synchronize your PCB with the schematic.
Just press the icon to do it:

[image]

This will say that values are updated.

In the schematic you should see the value as **${NET_NAME(1)}**, but in the PCB it should be the name of the net.

closed 2024-06-06 by set-soft