INTI-CMNB/KiBot #628: [BUG] AttributeError: 'SchematicComponentV6' object has no attribute 'x'

mdeweerd, 2024-06-08:
Field 'x' is not found.

```plaintext
Traceback (most recent call last):^M
  File "/usr/bin/kibot", line 33, in <module>^M
    sys.exit(load_entry_point('kibot==1.7.0', 'console_scripts', 'kibot')())^M
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^M
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 549, in main^M
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,^M
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 608, in generate_outputs^M
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)^M
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 598, in _generate_outputs^M
    run_output(out, dont_stop)^M
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 502, in run_output^M
    out.run(get_output_dir(out.dir, out))^M
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 214, in run^M
    self.options.run(target)^M
  File "/usr/lib/python3/dist-packages/kibot/out_any_sch_print.py", line 46, in run^M
    super().run(name)^M
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 1068, in run^M
    comps = self.variant.filter(comps)^M
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^M
  File "/usr/lib/python3/dist-packages/kibot/var_kicost.py", line 72, in filter^M
    comps = super().filter(comps)^M
            ^^^^^^^^^^^^^^^^^^^^^^M
  File "/usr/lib/python3/dist-packages/kibot/var_base.py", line 404, in filter^M
    comps = apply_pre_transform(comps, self.pre_transform)^M
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^M
  File "/usr/lib/python3/dist-packages/kibot/fil_base.py", line 176, in apply_pre_transform^M
    ret = filter.filter(c)^M
          ^^^^^^^^^^^^^^^^^M
  File "/usr/lib/python3/dist-packages/kibot/fil_base.py", line 130, in filter^M
    ret = f.filter(c)^M
          ^^^^^^^^^^^^M
  File "/usr/lib/python3/dist-packages/kibot/fil_subparts.py", line 256, in filter^M
    return self.do_split(comp, max_num_subparts, split_fields)^M
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^M
  File "/usr/lib/python3/dist-packages/kibot/fil_subparts.py", line 201, in do_split^M
    new_comp.set_field(field+'_qty', qty)^M
  File "/usr/lib/python3/dist-packages/kibot/kicad/v5_sch.py", line 961, in set_field^M
    f.x = self.x^M
          ^^^^^^^M
AttributeError: 'SchematicComponentV6' object has no attribute 'x'^M
```

Using Docker  on windows ghcr.io/inti-cmnb/kicad8_auto_full:latest and ghcr.io/inti-cmnb/kicad8_auto_full:dev.
Both give the same result - I am checking for more details.

mdeweerd, 2024-06-08:
I was able to work around this by modifying the code in v5_sch.py to:

```
        try:
          f.x = self.x
          f.y = self.y
        except AttributeError as e:
          print(f"**** ERROR *** no x %r" % (e,))
          print(vars(self))
```

Following that, it appears that there were reference mismatches between the PCB and the schematic.

set-soft, 2024-06-10:
I need a test case for this. It might be related to some component from the PCB, not found in the schematic, and some particular filter. 

Both SchematicFieldV6 and SchematicField constructors defines the x, y attributes. I can add a workaround similar to what you did, but knowing the source of the issue is much better.

set-soft, 2024-06-10:
Never mind, is the component that lacks x,y; which is the case for components from the PCB (where x,y is for the footprint).

closed 2024-06-10 by set-soft