INTI-CMNB/KiBot #629: [BUG] In Draw Satckup preflight thickness dimension is on wrong layer

RemyChatel, 2024-06-13:
## Describe the bug

When using the pre-flight `draw_stackup: true` to draw the stackup on User.Comments, I have noticed that the total thickness dimension was drawn on User.Drawings instead of User.Comments as requested.

It means that when trying to print only User.Comments in a PDF to get a PDF report of the stackup, it is missing that dimension.

Maybe this is the intended default behaviour to keep all dimension objects on User.Drawings, but in that case would it be possible to give the user the option to change this default behaviour?

## To Reproduce

Kibot 1.7.0 (using docker image ghcr.io/inti-cmnb/kicad7_auto:1.7.0) in Windows 11 WSL2-Ubuntu22, using the following command:

```
docker run --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY \
    -v $(pwd):/home/$USER/workdir \
    --user $(id -u):$(id -g) \
    --env NO_AT_BRIDGE=1 \
    --workdir="/home/$USER" \
    --volume="/etc/group:/etc/group:ro" \
    --volume="/home/$USER/.config/kicad:/home/$USER/.config/kicad:rw" \
    --volume="/home/$USER/.cache/kicad:/home/$USER/.cache/kicad:rw" \
    --volume="/home/$USER/.cache/kibot:/home/$USER/.cache/kibot:rw" \
    --volume="/home/$USER/.cache/mesa_shader_cache:/home/$USER/.cache/mesa_shader_cache:rw" \
    --volume="/home/$USER/.local/share/kicad:/home/$USER/.local/share/kicad:rw" \
    --volume="/etc/passwd:/etc/passwd:ro" \
    --volume="/etc/shadow:/etc/shadow:ro" \
    ghcr.io/inti-cmnb/kicad7_auto:1.7.0 /bin/bash -c "cd workdir; kibot"
```

`kibot.config.yaml` :

```
kibot:
  version: 1
preflight:
  draw_stackup: true
outputs:
- name: basic_pdf
  comment: Individual layers in PDF format
  type: pdf
  layers:
  - layer: User.Comments
```

- Create an empty new KiCad project
- Add rectangle to Edge.Cuts (to pass DRC mostly)
- Add two rectangles on User.Comments
- Group the two rectangles of User.Comments and give the group the name `kibot_stackup`
- Run `kibot` using the above configuration

The result is that most of the stackup information is correctly on layer User.Comments, but the dimension for total thickness is on User.Drawings

[image]

## Expected behavior

When drawing the `kibot_stackup` group on a layer, all elements of the stackup should be drawn on the same layer

[image]

## Proposed fix

Maybe it would be sufficient to add `dim.SetLayer(layer)` in the `draw_thickness` function of `pre_draw_stackup.py` around line:

https://github.com/INTI-CMNB/KiBot/blob/9a4e6716a6e95de816b7eaa7f2e8a039b090b145/kibot/pre_draw_stackup.py#L276

This fix works on my local (in WSL) install of kibot, I couldn't test inside the docker as I am not sure how to modify it or build it with that modification.

Thank you very much for your time!

set-soft, 2024-06-13:
Hi @RemyChatel !

You are right, the above patch solves the problem. It will be available in the *dev* images as soon as the regression tests are finished.

Thanks!

closed 2024-06-13 by set-soft

RemyChatel, 2024-06-13:
Hi @set-soft 

Thank you for the blazing fast support and this amazing tool !

Have a nice day

commit ce92162b (2024-06-13): [Draw Stackup][Fixed] Dimension always drawn on User.Drawings layer

Fixed #629