INTI-CMNB/KiBot #632: [BUG] AttributeError at /kibot/out_base.py", line 398

jorg-stor, 2024-06-20:
Hi! 
I'm getting the following attribute error when having "remove_solder_paste_for_dnp" set to false in Kibot (V1.7.0) config.
It seems like kibot/out_base.py", line 398 should be looking for global_remove_solder_mask_for_dnp attribute instead of remove_solder_mask_for_dnp.

Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.7.0', 'console_scripts', 'kibot')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 549, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 608, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 598, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 502, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 214, in run
    self.options.run(target)
  File "/usr/lib/python3/dist-packages/kibot/out_any_drill.py", line 165, in run
    self.filter_pcb_components()
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 827, in filter_pcb_components
    self.remove_paste_and_glue(GS.board, self.comps_hash)
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 398, in remove_paste_and_glue
    GS.remove_solder_mask_for_dnp):
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'GS' has no attribute 'remove_solder_mask_for_dnp'. Did you mean: 'global_remove_solder_mask_for_dnp'?

Thank you very much for your time!

set-soft, 2024-06-22:
Hi @jorg-stor !

You are right, the correct name is `global_remove_solder_mask_for_dnp`, as suggested by the interpreter.
I guess you have `remove_solder_paste_for_dnp` and `remove_adhesive_for_dnp` set to *false* and the interpreter needed to evaluate the last element, which seems to be uncommon because the patch was introduced almost a year ago: abeea7ec03154a29a6eba37cb13eae0cf00baaff

Patch 459a4e0b6d78c3e795e4bcd911b4c21953070295 should solve it. Can you try using the *dev* branch? (currently v1.7.1)

closed 2024-06-22 by set-soft

jorg-stor, 2024-06-22:
Hi @set-soft, 
Thanks for solving this! I just tested and the CI/CD runner now passed without problems using the dev branch.

Yeah I like having those configs set to false, so the stencils and PCBs are prepared for assembling DNP components also. Then if I decide to populate some of the components later on, the EMS do not need to have new stencils made.

I've been running Kibot v1.6.3 with Kicad7 for the last year now or so, but decided to switch to Kicad8 now and update the Kibot version also. But indeed it cannot be many that set those dnp configs to false if it's passed almost a year and it has not been spotted/reported. 

Thanks for your great work with the Kibot project!

commit 459a4e0b (2024-06-22): [Variants][Fixed] remove_solder_mask_for_dnp wrongly defined

Problems when remove_solder_paste_for_dnp and remove_adhesive_for_dnp
are both disabled

Fixes #632