INTI-CMNB/KiBot #633: [BUG] Unknown problem with update_xml in KiCad 8

andersruneson, 2024-06-24:
**Describe the bug**
We're running KiCad 7 and trying to migrate to KiCad 8, but fail to run KiBot with a KiCad 8 project.

**Log**
Log output from running kibot: https://pastebin.ubuntu.com/p/8m4pQ7JkZg/

For me it looks like update_xml is failing. I will try without that.

Let me know if more info is needed.

**Expected behavior**
Build should be fine I guess.

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow in GitLab
- Using a docker image kicad8_auto:dev

**Additional context**
Add any other context about the problem here.

andersruneson, 2024-06-24:
When deactivating update_xml altogether the build passes without errors.

set-soft, 2024-06-24:
This looks like a bug in KiCad, but you should just use KiCad to check the PCB/schematic parity. 
You should also migrate to the new *drc* preflight.

andersruneson, 2024-06-24:
Migrated a second project with the same issue:

```
ERROR:True (kibot.pre_update_xml - pre_update_xml.py:221)
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/__main__.py", line 568, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 614, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 5[94](https://git.elysics.se/Parts/EA/EA00070/-/jobs/206931#L94), in _generate_outputs
    preflight_checks(skip_pre, targets)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 451, in preflight_checks
    BasePreFlight.run_enabled(targets)
  File "/usr/local/lib/python3.11/dist-packages/kibot/pre_base.py", line 149, in run_enabled
    v.run()
  File "/usr/local/lib/python3.11/dist-packages/kibot/pre_update_xml.py", line 223, in run
    self.check_pcb_parity()
  File "/usr/local/lib/python3.11/dist-packages/kibot/pre_update_xml.py", line 199, in check_pcb_parity
    self.check_nets(net_nodes, errors, excluded)
  File "/usr/local/lib/python3.11/dist-packages/kibot/pre_update_xml.py", line 143, in check_nets
    for pad in con.GetNetItems(n, pcbnew.PCB_PAD_T)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 10[96](https://git.elysics.se/Parts/EA/EA00070/-/jobs/206931#L96)4, in GetNetItems
    return _pcbnew.CONNECTIVITY_DATA_GetNetItems(self, aNetCode, aTypes)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: in method 'CONNECTIVITY_DATA_GetNetItems', argument 3 of type 'std::initializer_list< KICAD_T > const &'
```

set-soft, 2024-06-24:
The problem is in KiCad, the argument is valid for KiCad 7, but not for KiCad 8

set-soft, 2024-06-24:
Even when the prototype didn't change

set-soft, 2024-06-24:
The above patch should stop and inform what to do.

set-soft, 2024-06-24:
> Aha, strange, we want to have the check_pcb_parity to make sure we do not release PCBs which have a diff between schematic and PCB. 

This is why you must migrate to the `drc` preflight, you are getting warnings about it when using `run_drc` (deprecated for KiCad 8)

closed 2024-06-24 by set-soft

andersruneson, 2024-06-24:
Right, thanks.