INTI-CMNB/KiBot #634: [BUG] ibom give error 2 running latest KiBot

andersruneson, 2024-06-24:
**Describe the bug**
I've migrated a project to KiBot 8 but fail on ibom. The same happens when running kicad8_auto:dev on a KiCad 7 project that worked before. I'm not sure when this started but I've been running 1.6.5 without issues. 

This is the output from KiBot:
```
- 'Generic Bill of Materials in XLSX format' (prototype_bom_xlsx) [bom]
- 'Generic Bill of Materials in XML format' (single_per_row_bom_xml) [bom]
- 'Components position for Pick & Place' (basic_position_ASCII) [position]
- 'Components position for Pick & Place' (basic_position_CSV) [position]
- 'Top Assembly Drawing' (basic_top_pcb_print_pdf) [pcb_print]
- 'Bottom Assembly Drawing' (basic_bottom_pcb_print_pdf) [pcb_print]
- 'Top Conformal Coating' (basic_top_conformal_coating_pdf) [pcb_print]
- 'Bottom Conformal Coating' (basic_bottom_conformal_coating_pdf) [pcb_print]
- 'IPC-D-356 netlist for testing' (ipc_netlist) [netlist]
- 'Interactive HTML BoM' (basic_ibom) [ibom]
ERROR:Failed to create BoM, error 2 (kibot.gs - gs.py:836)
ERROR:P (kibot.gs - gs.py:846)
ERROR:p (kibot.gs - gs.py:846)
```

This is my config:
```
- name: basic_ibom
  comment: Interactive HTML BoM
  type: ibom
  dir: '@EA@-@EArevLong@ PCBA'
  run_by_default: true
  options:
    output: '@EA@-@EArevLong@ Interactive Assembly Drawing.%x'
    dnf_filter: 
    - _kibom_dnf_Config
    - elysics_extra_mech
    - _mechanical
    extra_data_file: '@EA@-@EArevLong@ BOM.xml'
    include_tracks: true
    include_nets: true
    highlight_pin1: true
    show_fields: 'MPN,Case,Value'
    group_fields: 'MPN'
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow in GitLab
- Using a docker image kicad8_auto:dev

set-soft, 2024-06-24:
Hi @andersruneson !
I need something I can reproduce. I don't know what `elysics_extra_mech` is, or the content from `extra_data_file`
Did you try using debug information?
The message from ibom looks quite cryptic

andersruneson, 2024-06-24:
Hi

Which debug information? Running the verbose (-v) output from kibot? 

elysics_extra_mech is a filter:
```
filters:
- name: elysics_extra_mech
  type: generic
  comment: 'Exclude Elysics extra mechanical'
  exclude_any:
    - column: No BOM
      regex: '.*(Yes|yes|1).*'
```

extra_data_file is the xml from this output:
```yaml
- name: single_per_row_bom_xml
  comment: Generic Bill of Materials in XML format
  type: bom
  run_by_default: true
  options:
    output: '@EA@-@EArevLong@ BOM.%x'
    format: XML
    group_fields: []
    group_connectors: false
    columns:
    - field: 'Part'
      name: 'ReferenceNumber'
    - field: 'References'
      name: 'Designator'
    - field: 'MPN'
      name: 'MPN'
    - field: 'Manufacturer'
      name: 'Manufacturer'
    - field: 'Footprint'
      name: 'Footprint'
    - field: 'MPN'
      name: 'MPN'
    - field: 'Description'
      name: 'Description'
    - field: 'Temperature'
      name: 'Temperature'
    - field: 'Value'
      name: 'Value'
    - field: 'Digi-Key'
      name: 'Digi-Key'
```

andersruneson, 2024-06-24:
I ran with --verbose, seems to be something relevant:

```
DEBUG:Output destination: /builds/Parts/EA/EA00070/source/out/EA00070-draft PCBA (kibot - kiplot.py:458)
DEBUG:Filters reset (kibot - fil_base.py:194)
DEBUG:Applying filter `_kibom_dnf_Config,elysics_extra_mech,_mechanical` to fitted (kibot - fil_base.py:205)
DEBUG:Starting tool check for Interactive HTML BoM (kibot - dep_downloader.py:841)
DEBUG:- Returning `/usr/bin/generate_interactive_bom.py` (kibot - dep_downloader.py:848)
DEBUG:Doing Interactive BoM (kibot - out_ibom.py:166)
DEBUG:Creating variant netlist `/tmp/tmp-kibot-ibom-r0138sxh/EA00070.xml` (kibot - out_ibom.py:189)
DEBUG:Removing paste, mask and/or glue (kibot - out_base.py:404)
DEBUG:Storing modified PCB to `/tmp/tmp-kibot-ibom-r0138sxh/EA00070.kicad_pcb` (kibot - out_base.py:959)
DEBUG:Copying project `/builds/Parts/EA/EA00070/source/EA00070.kicad_pro` to `/tmp/tmp-kibot-ibom-r0138sxh/EA00070.kicad_pro` (kibot.gs - gs.py:527)
DEBUG:Copying project local settings `/builds/Parts/EA/EA00070/source/EA00070.kicad_prl` to `/tmp/tmp-kibot-ibom-r0138sxh/EA00070.kicad_prl` (kibot.gs - gs.py:535)
DEBUG:Copying /builds/Parts/EA/EA00070/source/templates/Default.kicad_wks -> /tmp/tmp-kibot-ibom-r0138sxh/schematic.kicad_wks (kibot - config.py:640)
DEBUG:Replacing page layout templates/Default.kicad_wks -> schematic.kicad_wks (kibot - config.py:643)
DEBUG:Saving modified project to /tmp/tmp-kibot-ibom-r0138sxh/EA00070.kicad_pro (kibot - config.py:659)
DEBUG:Restoring paste, mask and/or glue (kibot - out_base.py:491)
DEBUG:Running: ['/usr/bin/generate_interactive_bom.py', '/tmp/tmp-kibot-ibom-r0138sxh/EA00070.kicad_pcb', '--dest-dir', '/builds/Parts/EA/EA00070/source/out/EA00070-draft PCBA', '--no-browser', '--highlight-pin1', 'all', '--checkboxes', 'Sourced,Placed', '--bom-view', 'left-right', '--layer-view', 'FB', '--name-format', 'ibom', '--include-tracks', '--include-nets', '--sort-order', 'C,R,L,D,U,Y,X,F,SW,A,~,HS,CNN,J,P,NT,MH', '--extra-data-file', '/tmp/tmp-kibot-ibom-r0138sxh/EA00070.xml', '--show-fields', 'MPN,Case,Value', '--group-fields', 'MPN', '--blacklist', 'D15,G1,G2,J1,J3,MH1,MH10,MH2,MH3,MH4,MH5,MH6,MH7,MH8,MH9,NT1,NT2,NT3,NT4,NT5,NT6,NT7,NT8,NT9,Q5,R45,R46,R48,R49,TP1,TP10,TP2,TP3,TP4,TP5,TP6,TP7,TP8,TP9', '--old-fadhes', "[<pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17f00> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a148a0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17b10> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a140f0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14990> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14db0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17fc0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a157d0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a170c0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14360> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a16460> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17c90> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14c00> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17300> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a16070> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15530> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15950> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14930> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a154d0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14030> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15c80> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17a50> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14c60> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17420> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15f50> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15200> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14c90> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a14390> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17f60> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15380> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15cb0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15ce0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15a70> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17390> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15e00> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a16f70> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a159b0> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *'  [... cut ...] xy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a15e30> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17a20> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a17000> >, <pcbnew.PCB_SHAPE; proxy of <Swig Object of type 'PCB_SHAPE *' at 0x7f4ea5a164c0> >]
 (kibot.gs - gs.py:842)
ERROR:P (kibot.gs - gs.py:846)
ERROR:p (kibot.gs - gs.py:846)
DEBUG:Removing temporal variant dir `/tmp/tmp-kibot-ibom-r0138sxh` (kibot - out_ibom.py:238)
```

andersruneson, 2024-06-25:
> '--old-fadhes', "[<pcbnew.PCB_SHAPE; proxy ...

where does that come from???

andersruneson, 2024-06-25:
I tested out kicad8_auto:latest instead of kicad8_auto:dev and that worked fine. So I'm not sure you want to keep this issue?
Thanks for all you work with KiBot! We did send you a small donation yesterday.

set-soft, 2024-06-25:
Hi @andersruneson !

> > '--old-fadhes', "[<pcbnew.PCB_SHAPE; proxy ...
> 
> where does that come from???

An internal variable that should be named *_old_fadhes* instead of *old_adhes*. I optimized the way KiBot looks for object members, but the new mechanism exposed it. Was only in the *dev* code.

> I tested out kicad8_auto:latest instead of kicad8_auto:dev and that worked fine. 

Correct, the issue didn't affect the stable release.

> So I'm not sure you want to keep this issue?

We must clean these bugs before the next release ;-)
I'm doing some important changes, the short term benefit is better documentation. In particular about what is the default for things that aren't specified. I'm also relaxing some requirements, allowing to left things without definition and getting some simple behavior in this case. The long term benefit will be to allow using a GUI to create simple configurations.

> Thanks for all you work with KiBot! We did send you a small donation yesterday.

Thank you very much!

set-soft, 2024-06-25:
Ok, now should work

closed 2024-06-25 by set-soft