INTI-CMNB/KiBot #636: [HOW] to corretly use hide_excluded in pcb_print

mroetsc, 2024-06-28:
**What do you want to achieve?**
I need to have different variants of a project, where a part is rotated. My current idea to have one variant for each rotation of the part and have the parts lay on top of each other. The part, that is not placed in the variant, should not be shown on pcb_print. The separation of the variants currently works correctly, which is proven by the crossed out part in the generated PDF. However, because both parts lay on top of each other, it is hard to see which one actually is crossed out. Is there a better way to achieve this goal with KiBot?

The problem now is, that including `hide_excluded = true` does not actually exclude the part from the view, it still shows up as crossed out, rather than not at all.

I am calling KiBot with `kibot -c CONFIG.yaml -d OUT_DIR -b BOARD.kicad_pcb -e SCHEMATIC.kicad_sch -g variant=TEST`

**Do you have some PCB/Schematic to use as example?**
I can provide an example project if it turns out to be necessary

**Relevant part of kibot_config.yaml:**
```yaml
kibot:
  version: 1

global:
  units: 'millimeters'

variants:
  - name: 'TEST'
    type: kibom
    file_id: _TEST
    variant: TEST

outputs:
  - name: pcb_print
    comment: 'PDF of PCB'
    type: pcb_print
    dir: PDFs
    options:
      title: "+ %V"
      scaling: 2.0
      hide_excluded: True
      colored_pads: True
      color_theme: Plot
      pages:
        - layers: [ F.Cu,F.CrtYd,Dwgs.User,Cmts.User,Eco1.User,Eco2.User,Edge.Cuts,Margin,F.Fab ]
          sheet: 'Front'
```

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow
- Using a customized docker image based on kicad8_auto:latest, that adds some python packages for another custom script as well as jinja and dasel (This is because the kibot_config.yaml is dynamically generated for each project we have).

set-soft, 2024-06-28:
Hi @mroetsc !

Your approach looks too complex and problematic. I'll suggest using:

- A rotation filter to change the footprint rotation using a field called `JLCRotOffset` (you can customize it). You'll need to define one rotation for each variant.
- A variant rename filter to select which field is used.

The transformation will run first renaming and then rotating. 
You'll need to be patient about the values used in the `JLCRotOffset` because the various options and tool compatibility makes it hard to choose them.

mroetsc, 2024-07-03:
Hello @set-soft, thanks for the suggestion.
I looked into it, however, this will not work as well for our use case as the approach I described, as far as I can tell.

Here are some more information about how we run KiBot:
We have many projects on a self-hosted gitea instance. For this reason, we need a universal way of running KiBot across all of these projects, without changing much. The current approach is to have a master repo, that contains jinja templates for the kibot_config.yaml file. The gitea actions runner (fully compatible with GitHub workflow syntax) then checks out all relevant files, reads in some project data (name and the like) and generates the final kibot_config.yaml. Some configurations can be made via a toml file, that is read as an input for the jinja template. With this approach, we can cover different variant names in different projects or other project specific needs by simply configuring this from a toml file.

One of these project specific needs would be to hide excluded parts in a variant (`hide_excluded`) to achieve the goal described in the first post. 
I am aware that this is slightly more complex under the hood, but would clear things up for the end user, who would simply have to change a value in the toml file.

But `hide_excluded` does not produce the expected result, meaning parts excluded by a variant are still shown (crossed out). Is there something else to consider with `hide_excluded`?

set-soft, 2024-07-03:
Hi @mroetsc !

Your workflow doesn't explain why you can't just use variants for the rotation.

Anyways, I think you are confusing *dnf_filter* and *exclude_filter*. The *hide_excluded* option applies to stuff filtered by the *exclude_filter*, i.e. test points, mechanical holes, etc.

The above patch adds more debug info (I suggest using `-L logfile`) so you can know which components are actually hidden.

It also includes the following example:

```yaml
kibot:
  version: 1

globals:
  hide_excluded: true

filters:
  - name: no_R1
    type: generic
    exclude_refs: [R1]

variants:
  - name: 't1_v1'
    comment: 'Test 1 Variant V1'
    type: kibom
    file_id: '_(V1)'
    variant: V1
    exclude_filter: no_R1

outputs:
  - name: 'print_front'
    comment: "Experiment"
    type: pcb_print
    options:
      format: 'PNG'
      scaling: 2
      pages:
        - layers: [F.Fab, Edge.Cuts]
```

You can see it working using:
`src/kibot -L log -c tests/yaml_samples/pcb_print_hide_excluded.kibot.yaml -b tests/board_samples/kicad_8/kibom-variante.kicad_pcb -d pp -g variant=t1_v1`

mroetsc, 2024-07-03:
Hi @set-soft,
thank you for the clarification, I was missing this information:
> The hide_excluded option applies to stuff filtered by the exclude_filter, i.e. test points, mechanical holes, etc.

This clears things up and explains why it wasn't working, thank you for the quick help again.

closed 2024-07-03 by mroetsc