INTI-CMNB/KiBot #638: [FEATURE] Add Testpoint report output

CraigBurden, 2024-07-03:
There is a great tool by @snhobbs, that generates a testpoint report, cataloging all testpoints and a few useful bit of information about each.

https://github.com/snhobbs/kicad-testpointsn

Test points are typically marked as "exclude_from_bom" and so do not appear on a BOM, as they typically should not.
But when designing a test fixture for a board this information is needed.  

Adding the test report generator to the available outputs so that it can be generated along with all other outputs when a PCB release is made

CraigBurden, 2024-07-03:
@snhobbs Good catch and its an absolute pleasure!
I love the plug-in!

set-soft, 2024-07-03:
Hi @CraigBurden and @snhobbs !

The idea is interesting, currently you can ask KiBot to generate a BoM with component coordinates, so I think most of the information is already available. 

I found that the link to *JigsApp* at the beginning of the *kicad-testpoints* is wrong, should be https://www.thejigsapp.com/ Missing http:// and a typo, is ok in the links section.

CraigBurden, 2024-07-03:
@set-soft Thanks for the feedback, I haven't tried that within KiBot yet, but my experience of all the other BOM tools out there like KiBOM and the integrated BOM tools in KiCAD is that they do not do what we need.

The reason is that test points typically have the "exclude_from_bom" and/or dnp attributes set. This means that they are removed from the BOM outputs. The former exclusion is impossible to bypass in those tools.

snhobbs, 2024-07-04:
Oh that's interesting, can it do pads as well as footprints or does it need to be an actual standalone footprint?

CraigBurden, 2024-07-05:
@set-soft, @snhobbs asks a good question. Often we want to land spring pins on pads of footprints with more than one pad. Your suggestion assumes the footpoint centre is the same as the pad centre and that there is only a single pad? 

For example:
[image]
This footprint has 8 pads, none of which are on the centre of the footprint itself

set-soft, 2024-07-05:
Hi @CraigBurden !

> Oh that's interesting, can it do pads as well as footprints or does it need to be an actual standalone footprint?

As this is part of the BoM currently only full components. But I guess we could define a field to ask for individual pins. Perhaps this could be supported by *kicad-testpoint*. What do you think about some name like *_TestPoint_Pin#* where # is an arbitrary number defined by the user (and optional). The content of this field should be a valid pin of the component. In this way the user could ask to define as much individual pins as desired. 

Using fields makes this "self contained", you don't need external files. Then the tools could pick these fields and generate pseudo references like **J2_1** for a real reference named **J2** with a field *_TestPoint_Pin1*. If the user doesn't define the value for *_TestPoint_Pin1* we could assume this is for pin 1.

set-soft, 2024-07-05:
> @set-soft, @snhobbs asks a good question. Often we want to land spring pins on pads of footprints with more than one pad. Your suggestion assumes the footpoint centre is the same as the pad centre and that there is only a single pad?

Yes, this is what a testpoint is ;-)

> To be clear, the way that @snhobbs's plug-in decides on what pads to add to the report has nothing to do with the name/value/reference of the footprint, its driven by fabrication attributes on pads. So I can add this attribute to any pad on any footprint. Hence I can land pins on the THT pins below a RJ45 connector if I want to as well as all the dedicated test points

This looks like the job for a transformation filter.

set-soft, 2024-07-05:
An important note: the attribute used for "nail of beds" seems to be quite orthogonal in KiCad. None of the library testpoints has this attribute.

snhobbs, 2024-07-06:
Yeah, I've found using only testpoint parts to be too restricting. They're still the first line of attack for sure but the idea is to implement the same feature other EDA suite have (see altium's https://www.altium.com/documentation/altium-designer/adding-testpoints-pcb). Doing this with the fabrication option is a bit cheeky but I can't find a good reason against using it. The command line tool can use either the fabrication setting of the pad or a config file with the ref des & pad number.

CraigBurden, 2024-07-11:
> Yeah, I've found using only testpoint parts to be too restricting. They're still the first line of attack for sure but the idea is to implement the same feature other EDA suite have (see altium's [...]

Agreed, the only alternative that I have explored in the past is to select the centroid of all mask openings on a top or bottom layer. That works okay when you have a side that only has openings that you want to land on, but when there you have assembly and test lands mixed it gets very messy.

@set-soft The fabrication attribute is not set by default on the build-in libraries, but since I started using @snhobbs's library I have just set the attributes in my personal library. The complexity is that you often what to land test pins on points that aren't "officially" test points. For example I regularly use convex tipped test pins to connect to the THT pins under a connector similar to what I showed above. This is very common. So the solution of how to designate what co-ordinates are added to the report needs to be flexible to allow that. 
I feel like despite the fact that the build-in libraries don't set the attribute by default, this is still the best option that is available to us. The "correct" solution in my mind is an interactive selection tool within the KiCAD UI, but that would likely never happen

set-soft, 2024-07-11:
> I feel like despite the fact that the build-in libraries don't set the attribute by default, this is still the best option that is available to us.

I agree

set-soft, 2024-07-16:
Hi @snhobbs !

> The command line tool can use either the fabrication setting of the pad or a config file with the ref des & pad number.

How about the side for THT components? You only know the footprint side ...

snhobbs, 2024-07-16:
The tool pulls out the pad locations as well as the pad side, pad type (THT vs SMT), and the component side. This tool doesn't mess with the sides at all but the tools we use to generate jigs by default chooses the opposite side of the THT part as the probing side. I could imagine a few times where there would be some ambiguity there but it would be very rare.

set-soft, 2024-07-17:
Ok, the above patch adds a suitable filter, running `src/kibot -c tests/yaml_samples/test_points_2.kibot.yaml -b tests/board_samples/kicad_8/test_points.kicad_pcb` I'm getting:

[image]

Which is similar to what @snhobbs CLI tool is generating:

[image]

snhobbs, 2024-07-17:
Nice!

MarkusSBS, 2024-09-10:
it would be nice to have a report with testpoint coverage:

================= Summary ======================
Total TestPoint : 282
Total Nets : 406
Total connected Nets : 372
Unconnected Nets : 34
Connected Nets with TestPoint : 270
Connected Nets without TestPoint : 102
TestPoint coverage : 72.6%

====== Connected Nets without TestPoint ========
Net-(C8-Pad2)
Net-(C9-Pad2)
Net-(C17-Pad2)
Net-(C18-Pad2)
Net-(C72-Pad1)
Net-(C72-Pad2)
Net-(C75-Pad2)

========= 1 TestPoint connected Nets ===========
+3.3STBY : TP179
+3.3V : TP175
+3.3VADC : TP100
+12P : TP142
+12V : TP176

===== More than 1 TestPoint connected Nets =====
/AFE/C0 : TP157, TP23, TP24, TP25
BAT+ : TP115, TP116, TP117
CORE+ : TP101, TP102, TP103
GND : TP125, TP126, TP127, TP128, TP309, TP59

=============== Unconnected Nets ===============
/environment_sensor/ENV_Sensor_WAKEUP
Net-(CON13-Pad1)
unconnected-(CON2-Pad3)

this was generated with a kicad6 script but i can not find it anymore

MarkusSBS, 2024-09-10:
https://github.com/Peboli/kicad_netlist_tp_parser

set-soft, 2024-09-10:
> https://github.com/Peboli/kicad_netlist_tp_parser

I tried it and I have too much doubts.

* This is an schematic tool, so you must rely on testpoints using some particular name. So you can't use any pad as a testpoint. This is inconsistent with the behavior discussed in this thread
* It reports unconnected nets for pins that you don't want to connect and explicitly have a "don't connect" element.

I think the whole unconnected thing is irrelevant in a PCB, the really important unconnected thing is reported by the DRC. Unconnected pins in the schematic are irrelevant for the PCB.

closed 2024-09-17 by set-soft

MarkusSBS, 2024-09-24:
I want to change the report layout but i am strugeling with the pandoc syntax. can you may give a hint how to solve it.

# Testpoints
- Total pads: ${total_pads}
- Total testpoints: ${testpoint_pads}
- Total Nets: ${total_nets}
- With testpoint: ${nets_with_testpoint}
- TestPoint coverage: ${%5.2f,testpoint_coverage} %

| Nets without testpoint           | Nets with one testpoint                   | Nets with more than one testpoint            |
|----------------------------------|-------------------------------------------|----------------------------------------------|
| #nets_without_testpoint:- ${net} | #nets_with_one_testpoint:- ${net}: ${pad} | #nets_with_many_testpoints:- ${net}: ${pads} |

[image]

MarkusSBS, 2024-09-24:
ah ok the idea was to save space with 3 rows