INTI-CMNB/KiBot #640: [HOW]

chof747, 2024-07-05:
**What do I want to achieve?**
In my boards I have a logo placed with the reference designator G*** which does not have a component related to it.
When I run kibot using the JLCPCB standard import
I get the following error message:

````console
Using SCH file: mcu-bat-power.kicad_sch
- 'Gerbers compatible with JLCPCB' (_JLCPCB_gerbers) [gerber]
- 'Drill files compatible with JLCPCB' (_JLCPCB_drill) [excellon]
- 'Pick and place file, JLCPCB style' (_JLCPCB_position) [position]
Traceback (most recent call last):
  File "/usr/bin/kibot", line 33, in <module>
    sys.exit(load_entry_point('kibot==1.6.5', 'console_scripts', 'kibot')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 546, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 599, in generate_outputs
    _generate_outputs(outputs, targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 589, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 493, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 214, in run
    self.options.run(target)
  File "/usr/lib/python3/dist-packages/kibot/out_position.py", line 252, in run
    super().run(fname)
  File "/usr/lib/python3/dist-packages/kibot/out_base.py", line 1048, in run
    get_board_comps_data(comps)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 344, in get_board_comps_data
    c = create_component_from_footprint(m, ref)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 317, in create_component_from_footprint
    c.split_ref()
  File "/usr/lib/python3/dist-packages/kibot/kicad/v5_sch.py", line 1079, in split_ref
    raise SchError('Malformed component reference `{}`'.format(self.ref))
kibot.kicad.error.SchError: Malformed component reference `G***`
````

How can I tell the JLCPCB outputs imported or better globally that the reference 'G***' should be excluded from all operations?

When I rename that reference to something without * it works fine but I am not sure if this causes problems elsewhere as I have seen many other projects referencing the logos on the board in that way.

**My configuraton file:**
```yaml
# This is a working example.
# For a more complete reference use `--example`
kibot:
  version: 1

global:
  filters:
  - number: 1007
  - number: 1015
  - number: 58

import:
- file: JLCPCB
  definitions:
    _KIBOT_MANF_DIR: Manufacturers/JLCPCB
    _KIBOT_POS_PRE_TRANSFORM: '[''_kicost_rename'', ''_rot_footprint'']'
    _KIBOT_BOM_ENABLED: 'false'
 
outputs:
  - name: basic_boardview
    comment: Board View export
    type: boardview
    dir: Assembly

...
definitions:
  _KIBOT_MANF_DIR_COMP: Manufacturers
  _KIBOT_GERBER_LAYERS: "- layer: 'F.Cu'\n        suffix: 'F_Cu'\n        description:\
    \ 'Front copper'\n      - layer: 'B.Cu'\n        suffix: 'B_Cu'\n        description:\
    \ 'Bottom copper'\n      - layer: 'F.Paste'\n        suffix: 'F_Paste'\n     \
    \   description: 'Front solder paste'\n      - layer: 'B.Silkscreen'\n       \
    \ suffix: 'B_Silkscreen'\n        description: 'Bottom silkscreen (artwork)'\n\
    \      - layer: 'F.Silkscreen'\n        suffix: 'F_Silkscreen'\n        description:\
    \ 'Front silkscreen (artwork)'\n      - layer: 'B.Mask'\n        suffix: 'B_Mask'\n\
    \        description: 'Bottom soldermask (negative)'\n      - layer: 'F.Mask'\n\
    \        suffix: 'F_Mask'\n        description: 'Front soldermask (negative)'\n\
    \      - layer: 'Edge.Cuts'\n        suffix: 'Edge_Cuts'\n        description:\
    \ 'Board shape'"
```

**Environment (please complete the following information):**
I run kibot in a docker image: 

docker run --rm -it --workdir="/home/$USER" -v $(pwd):/home/$USER/workdir ghcr.io/inti-cmnb/kicad8_auto:1.6.5

set-soft, 2024-07-05:
Hi @chof747 !

This was fixed in 1.7.0, already released. Currently you get a warning, see #604

closed 2024-07-05 by set-soft

chof747, 2024-07-05:
Thank you, switching to latest docker container worked :-)