INTI-CMNB/KiBot #641: PCB draw doesnt populate component [BUG]

rizwan-isa-3E, 2024-07-05:
I am using Kibot to automaticaly output an image to show components. However, it doesnt do it. Here is my settings in the .yaml file:

  - name: board_top_filled
    comment: "Top layer view with components"
    type: pcbdraw
    dir: PCB
    output_id: '-filled'
    options:
      format: jpg
      dpi: 300
      show_components: all
here is the output from the terminal window:

'Gerbers' (gerbers) [gerber]

'Drill files' (drill) [excellon]

'Pick & Place' (position) [position]

'PDF for the PCB' (print_pdf) [pcb_print]

'Top layer view' (board_top) [pcbdraw]

WARNING:(W103) (component) Component _footprints:7 has no footprint. (kibot - out_pcbdraw.py:43)

WARNING:(W103) (component) Component _footprints:B has no footprint. (kibot - out_pcbdraw.py:43)

'Top layer view with components' (board_top_filled) [pcbdraw]

'Bottom layer view' (board_bottom) [pcbdraw]

'Bottom layer view with components' (board_bottom_filled) [pcbdraw]

'STEP 3D model' (3D) [step]

set-soft, 2024-07-08:
Hi @rizwan-isa-3E !

What components are you using? With this information I can only guess the problem is that none of them is in the PcbDraw library.

I used your config with the PcbDraw example (tests/data/ArduinoLearningKitStarter.kicad_pcb) and got:

[image]

Can you post the PCB?

rizwan-isa-3E, 2024-07-10:
Hi @set-soft!

These are custom 3rd party components in my footprints.pretty library. How do I get these into PcbDraw library?

B32778G4107K000.wrl

RJ45_Molex_0855135013_Vertical.wrl

set-soft, 2024-07-10:
Hi @rizwan-isa-3E !

You have to provide a suitable drawing, please check the PcbDraw site:  https://github.com/yaqwsx/PcbDraw

PcbDraw doesn't have too much of components, the author idea is that people contribute the missing ones.

closed 2024-07-10 by set-soft