INTI-CMNB/KiBot #643: [FEATURE] Temperature range check

andersruneson, 2024-07-11:
**Is your feature request related to a problem? Please describe.**
When designing a board according to specification there is normally a product requirement that the board should work in a certain temperature range, like -40 to 85 deg Celsius, the all the components needs to fulfill this range and it would be convenient to have a check for that when running CI on my board.

**Describe the solution you'd like**
An output could be run created that makes a report which lists all components, their range and a separate list with the components that are out of range according to the defined configured range.

**Describe alternatives you've considered**
Could also just be a preflight like erc/drc which exits on out of range errors.

I was thinking that I could do this as a separate script to run in my CI-chain, but it would be much better to have the config settings for the range in the kibot-config. It would also be more work in totalt to make a separate script and then integrate into KiBot instead of just doing this directly in KiBot.

**Additional context**
Some example configuration below. There need to be some kind of setting (like below exit_error_on_out_of_range) so that KiBot would set an error just like a DRC error.

There is also need to configure which fields to look for temperature data, I don't use the built in KiCad libraries so I don't know how temperature data is normally stored in the components, but I assume one would use use one property with a range, or two properties with max and min data.

```
global:
  pcb_material: 'FR4'
  solder_mask_color: 'green'
  silk_screen_color: 'white'
  pcb_finish: 'ENIG'
  copper_thickness: 35
  temperature_range_max: 85
  temperature_range_min: 30

outputs:
- name: temperature
  comment: Simple temperature report
  type: temperature_report
  options:
    temp_max_field: "Temperature Max"
    temp_min_field: "Temperature Min"
    temp_range_field: "Temperature"
    temp_range_regex: "(-?\d+).+?(-?\d+).*"
    exit_error_on_out_of_range: true
```

Thanks for your feedback on this idea.

set-soft, 2024-07-11:
Hi @andersruneson !

> I don't use the built in KiCad libraries so I don't know how temperature data is normally stored in the components, but I assume one would use use one property with a range, or two properties with max and min data.

I don't think KiCad handles temperatures at all. In particular: KiCad libs are generic, you have a 74LS00 component, but this is for any manufacturer and grade (commercial, industrial, etc.)

How do you get the temperature in your components?

KiBot can get temperature ranges from component distributors, and can even push these values to fields. But configuring it for CI/CD is complex.

andersruneson, 2024-07-11:
My company has a component database and use the new dblib-connector since in kicad 7, all components have manufacturer part number in the database, so we don't use "any 10kohm resistor", we use "a specific 10k resistor". When adding a new component in the database we collect the data from Digi-Key, like voltage, tolerance, and temperature range. 
This is the way many companies work with electronic design, when we place components in the schematic editor we already know exactly which component it is.

[image]

set-soft, 2024-07-12:
Hi @andersruneson !
I agree that the correct way to do things is by using specific components, identified by some code, not generic ones.
Ok, the image clarifies the way you make it, and the regular expression you used.
KiBot can get this information from digikey and put it in the schematic, but I see how you solve it.
Thanks

set-soft, 2024-07-12:
Ok, the above patch is quite raw, but is starting to work, I'm running:

 ```
src/kibot -c tests/yaml_samples/check_fields_temp_1.kibot.yaml -b tests/board_samples/kicad_8/temp_range.kicad_sch -d pp
```

From the root of the dev repo, and getting:

```
WARNING:(W162) R2 field `Temp` doesn't match `(-?\d+).+?(?:-?\d+).*` (no_match) (kibot - pre_check_fields.py:94)
R3 field `Temp` fails 20.0 >= 85
WARNING:(W162) R4 field `Temp` fails -5.0 <= -10 (kibot - pre_check_fields.py:94)
R5 field `Temp` fails 60.0 >= 85
Found 2 unique warning/s (2 total)
```

andersruneson, 2024-07-12:
That was quick! Flexibel solution 👍
Many thanks, will give it a try when Im back at my computer 🙏

set-soft, 2024-07-13:
Yes, I tried to make it flexible.
In the past we needed to check RoHS and temperature range for an electromedical device.
The code is WIP, I need to check some details, but is starting to work

closed 2024-07-15 by set-soft

Egoruch, 2025-02-01:
Hi! I tried to check example and it works, so now it's easy to modify, because I'm using two fields:

_Maximum Operational Temperature
Minimum Operational Temperature_

[image]

Since each board can have their own operational temperature requirements and modifying .yaml files is not good I'm thinking about adding two global (text) variables in the KiCad project itself, which is very logical:

[image]

But since ${VARIABLE} is string and we need number to compare, here is the question:
Is it possible to convert string to the number to use it it like this:
`numeric_reference: ${VARIABLE} `

[image]

set-soft, 2025-02-04:
Hi @Egoruch !

The way you tried won't work because you can't expand KiCad variables at any place.
The above patch extends the preprocessor functionality (See https://kibot.readthedocs.io/en/latest/configuration/substitution.html#).
Now you can specify `--defs-from-project` and use the KiCad vars for it.

Egoruch, 2025-02-05:
Thanks for patch, but since I use stable version _1.8.2_ I can't test it right now.

The another cool thing related to the checking field of components is check components status.
e.g. I have field **_Lifecycle_** for each component with following content (from Altium):

- Planned
- Draft
- Prototype
- Production
- Obsolete

When doing board release all components should be **Prototype** (verified by engineer) or **Production** (confirmed on real board).
I tried to use the same mechanism (_check_fields_) for it, but it doesn't work since it's string (not number).

So, is it possible to compare not just numbers but also text (string) using _check_fields_ mechanism?

set-soft, 2025-02-07:
> Thanks for patch, but since I use stable version _1.8.2_ I can't test it right now.

Just switch to *dev* images. They have less bugs than 1.8.2.

> The another cool thing related to the checking field of components is check components status. e.g. I have field **_Lifecycle_** for each component with following content (from Altium):
>
> * Planned
> * [...]

You can check using regular expressions.
https://kibot.readthedocs.io/en/latest/configuration/preflights/FieldCheck.html
If you don't use `numeric_condition` you can do it.

set-soft, 2025-02-07:
> > KiBot can get this information from digikey and put it in the schematic, but I see how you solve it.
> 
> [@set-soft](https://github.com/set-soft) can you point me to an example of this.

You must use the following filter:
https://kibot.readthedocs.io/en/latest/configuration/filters/spec_to_field.html

An example is https://github.com/INTI-CMNB/KiBot/blob/dev/tests/yaml_samples/spec_to_field_1.kibot.yaml

Note that you must edit it.
In the global section you'll see how to specify custom names for the fields to be filled. In this example I test Spanish names.
The filter fills the RoHS, Tolerance, Power, Voltage and Current fields using data from Mouser and Digikey.
The `bom_internal` output must be functional, here we provide a custom KiCost config (tests/data/kicost_dk_mou_cache.yaml) you don't need it if you are already getting information for costs.
The modified schematic is created by `create_sch` output. It will contain the data filled. You should double check this schematic, and use a repo in case you need to rollback the KiBot changes.

Egoruch, 2025-02-07:
> Just switch to _dev_ images. They have less bugs than 1.8.2.
>
>
> You can check using regular expressions. https://kibot.readthedocs.io/en/latest/configuration/preflights/FieldCheck.html If you don't [...]

Thanks, I tested in in that way and it works:

kibot:
  version: 1
preflight:
  check_fields:
    - field: Lifecycle
      regex: '(Production|Prototype)'
      severity: error
      severity_missing: error
      severity_no_match: error

Production component (**passed**):
[image]
Draft components (**not passed**):
[image]