INTI-CMNB/KiBot #644: [BUG] auto-crop in render3d is cropping within the bounds of the PCB

Wol, 2024-07-12:
As part of the auto_crop function within the render3d component, it seems to be calling `-trim +repage` twice. For images I'm seeing, it's first cropping the background to the PCB outline, but then the second `-trim` then crops the PCB background color down to the first object within the PCB itself.

The commit at https://github.com/INTI-CMNB/KiBot/commit/81ce2004d58b7812b50191a824974b51f1907bb6#diff-c346d8dc8224d45775c7a231f73c7ef006d36c3c6ab0196d49bb83a7f253280bR288 seems to not provide much information about why `-trim` is called twice!

The right hand side of the original photo:
[image]

After the first `-trim`. This is what I want!
[image]

After the second trim: Now it's trimmed too far.
[image]

I'm running KiBot as part of KiCi, within a docker container on gitlab-ci.

Configuration is:

`auto_crop: false
      clip_silk_on_via_annulus: true
      download: true
      height: @KICI_RENDER3D_HEIGHT@
      highlight: []
      move_x: @KICI_RENDER3D_BOTTOM_MOVE_X@
      move_y: @KICI_RENDER3D_BOTTOM_MOVE_Y@
      no_smd: false
      no_tht: false
      no_virtual: false
      orthographic: true
      ray_tracing: false
      force_stackup_colors: true
      realistic: true
      show_adhesive: false
      show_board_body: true
      show_comments: false
      show_components: all
      show_eco: false
      show_silkscreen: true
      show_soldermask: true
      show_solderpaste: true
      show_zones: true
      subtract_mask_from_silk: true
      transparent_background: @KICI_RENDER3D_BG_TRANSPARENT@
      transparent_background_fuzz: 15
      variant: ''
      view: 'bottom'
      wait_render: -600
      width: @KICI_RENDER3D_WIDTH@
      zoom: @KICI_RENDER3D_BOTTOM_ZOOM@
`

My current workaround is to just not crop images using KiBot, and instead having to manually run the convert step instead.

set-soft, 2024-07-15:
Hi @Wol !

Thanks for reporting. Next time please include an example and more information about the used options, the template you show has a lot of variables.

The double trim pass was needed at the time the option was implemented, now it isn't, so now we have an option to enable the double pass.

closed 2024-07-15 by set-soft

Wol, 2024-07-15:
Perfect! Thank you very much!

set-soft, 2024-07-15:
Please try using a *dev* docker image, it will solve the problem