INTI-CMNB/KiBot #651: [BUG] run_by_default: false not working for outputs

soldermate, 2024-08-06:
**Describe the bug**
run_by_default: false doesn't seem to work. Outputs are still generated when no specific output is passed to kibot.

**To Reproduce**
```
  options: 
    run_by_default: false
```
**Expected behavior**
When the option is set to false, the output should not be generated unless that output or group is passed to kibot.
It would be good if preflights still run when no outputs are passed in and all outputs have run_by_default: false

**Environment**
ghcr.io/inti-cmnb/kicad8_auto_full:dev

set-soft, 2024-08-06:
Please provide a minimal example showing your issue.

> It would be good if preflights still run when no outputs are passed in and all outputs have run_by_default: false

I suggest creating a small example where you think the behavior is different

soldermate, 2024-08-08:
Sorry @set-soft, I was using run_by_default: false indented under options, when it should not be indented.

output.kibot.yaml
```
kibot:
  version: 1

preflight:
  erc: 
    dir: test-reports
    dont_stop: true
  drc: 
    dir: test-reports
    dont_stop: true

outputs:
- name: 'schematic_pdf'
  comment: "Circuit schematic"
  type: pdf_sch_print
  dir: doc
  options: 
    run_by_default: false
```

`kibot -c output.kibot.yaml -e schematic.kicad_sch -b pcb.kicad_pcb`

The schematic output is still produced and prints the following warning:

WARNING:(W015) Unknown option run_by_default (kibot - optionable.py:149)
Circuit schematic' (schematic_pdf) [pdf_sch_print]

closed 2024-08-08 by soldermate