INTI-CMNB/KiBot #652: [BUG]

soldermate, 2024-08-19:
**Describe the bug**
Text variables not propagating to the PCB panel when the panelize output runs.  
The text variables are correctly substituted in the project, but the same variables when used in PCB layers are not substituted.

**To Reproduce**
command
`VERSION="foobar"`
`kibot -v -c board.kibot.yaml -e sch.kicad_sch -b pcb.kicad_pcb -E GIT_VERSION=$VERSION`

board.kibot.yaml
```
kibot:
  version: 1

global:
  out_dir: 'generated'
  output: '%f-@GIT_VERSION@_%i%I%v.%x'
  set_text_variables_before_output: true

preflight:
  set_text_variables:
    - name: 'GIT_VERSION'
      text: '@GIT_VERSION@'

outputs:
  - name: project_panel
    comment: Kicad panel project
    type: panelize
    dir: panel
    options:
      configs: panel_config.json
```

panel_config.json

```
{
    "layout": {
        "type": "grid",
        "rows": 1,
        "cols": 3,
        "hspace": "4.6mm",
        "vspace": "0mm",
        "rotation": "0deg",
        "alternation": "none",
        "renamenet": "Board_{n}-{orig}",
        "renameref": "{orig}"
    },
    "source": {
        "type": "auto",
        "tolerance": "10mm"
    },
    "tabs": {
        "type": "annotation"
    },
    "cuts": {
        "type": "mousebites",
        "drill": "0.8mm",
        "spacing": "1.3mm",
        "offset": "0.25mm",
        "prolong": "0.5mm"
    },
    "framing": {
        "type": "railstb",
        "mintotalheight": "120mm"
    },
    "fiducials": {
        "type": "3fid",
        "hoffset": "10mm",
        "voffset": "5mm",
        "coppersize": "1mm",
        "opening": "2mm"
    },
    "tooling": {
        "type": "4hole",
        "size": "3.9mm",
        "hoffset": "5mm",
        "voffset": "5mm"
    },
    "text": {
        "type": "simple",
        "text": "origin panel to board (x,y): 75mm, 19mm\nboard step (x,y): 95mm, 0.0mm",
        "layer": "F.SilkS",
        "anchor": "bl",
        "voffset": "-5mm",
        "hoffset": "15mm",
        "hjustify": "left"
    },
    "text2": {
        "type": "simple",
        "text": "origin panel to board (x,y): 75mm, 19mm\nboard step (x,y): 95mm, 0.0mm",
        "layer": "B.SilkS",
        "anchor": "br",
        "voffset": "-5mm",
        "hoffset": "-15mm",
        "hjustify": "left"
    },
    "post": {
        "type": "auto",
        "millradius": "1mm",
        "origin": "bl",
        "dimensions": "true"
    },
    "copperfill": {
        "type": "hatched",
        "clearance": "1mm"
    },
    "page": {
        "type": "A3"
    },
    "debug": {
        "trace": "true"
    }
}
```

**Expected behavior**
Text variables in PCB layers should be substituted in the generated panel project.

**Screenshots**
Correct substitution in project settings view after generating the panel
[image]

Substition does not occur in PCB layer
[image]

**Environment**
image: ghcr.io/inti-cmnb/kicad8_auto_full:dev

**Additional context**
Add any other context about the problem here.

set-soft, 2024-08-19:
Hi @soldermate !
Please post a small example where we can see the issue.
I can't reproduce it.

set-soft, 2024-08-19:
This is not a bug in KiBot, is a KiCad "feature".
I'm trying to make our workaround stronger, but I'm not sure about undesirable side effects.
See the discussion about this KiCad behavior here: https://gitlab.com/kicad/code/kicad/-/issues/14360

set-soft, 2024-08-19:
Ok @soldermate, try the *dev* images now, they should work

closed 2024-08-19 by set-soft

soldermate, 2024-08-20:
Thank you for the quick turn around @set-soft. Appreciate this is not really a bug, the work around is very helpful. I don't agree with the functionality.

commit 82f4e2e9 (2024-08-19): [Changed] Stronger workaround for KiCad PCB text cache

Fixes issues with external tools when not using variants
Fixes #652