INTI-CMNB/KiBot #653: [HOW] DRC exlusiones stoped working

MarkusSBS, 2024-08-22:
DRC stoped working for me and my jobs are failing.

Jobs for repositorys worked fine 2 monthe ago and without any changes fail if i rerun the job.

I have testet it with kicad 7 and i migrated a projekt  to kicad 8 and have the same result.

is there a migration step i missed? I have no errors and some exlusiones in my local kicad DRC scan but all exlusions are reportet as errors.

MarkusSBS, 2024-08-22:
here my kibot test projekt. I just rerun the pipeline: 
[image]

set-soft, 2024-08-22:
> DRC stoped working for me and my jobs are failing.
> 
> Jobs for repositorys worked fine 2 monthe ago and without any changes fail if i rerun the job.

Is hard to know the reason with no information about versions involved.

> I have testet it with kicad 7 and i migrated a projekt to kicad 8 and have the same result.
> 
> is there a migration step i missed? 

When migrating to KiCad 8 you should also migrate from *run_drc* to *drc*, a warning explains it. The *drc* preflight is cleaner and uses a fast mechanism.

> I have no errors and some exlusiones in my local kicad DRC scan but all exlusions are reportet as errors.

DRC exclusions are stored in the project file (*kicad_pro*), they looks like this:

```
      "drc_exclusions": [
        "clearance|176385001|79614999|7e023245-2c2b-4e2b-bfb9-5d35176e88f2|666713b0-70f4-42df-8761-f65bc212d03b",
        "invalid_outline|165100000|80645000|e6d8b5bb-6f41-4222-9a89-b8a6280d0752|00000000-0000-0000-0000-000000000000",
        "silk_over_copper|153440000|84620000|3f5fe6b7-98fc-4d3e-9567-f9f7202d1455|f4f99e3d-7269-4f6a-a759-16ad2a258779"
```

They are verified in the regression tests, so I need an example where they fail to work.

MarkusSBS, 2024-08-22:
drc_exclusions are included in the projekt file.

docker image ist image:
  name:  ghcr.io/inti-cmnb/kicad7_auto_full:dev

I will send you a zip of the test repo

set-soft, 2024-08-23:
Hallo @MarkusSBS !

This was related to a known KiCad bug that we had a workaround, but wasn't applied at all the needed places. In your particular case the *set_text_variables* triggered the issue.

Now should be fixed.

closed 2024-08-23 by set-soft

MarkusSBS, 2025-07-02:
normaly i will upgrade to a newer version but for old projekt there is sometimes no kicad involved on user side in generating a new dataset. for example a different variant so it would by nice to have a working kicad 8 version without changing and fixing the yaml config for the latest version. Work in progress projekts will migrate to the newer version with no problem

MarkusSBS, 2025-07-02:
i will test it 

these rounding issue sounds realistic but still sucks a little bit on the kicad side by breaking older projekt if there is an update.