INTI-CMNB/KiBot #655: [BUG] kibot-<hex>.kicad_dru temporary file not deleted

memen45, 2024-08-27:
**Describe the bug**
During execution, several temporary files are created for kibot, e.g. `kibot_4e6917n2.kicad_pro`, `kibot_4e6917n2.kicad_pcb`, `kibot_4e6917n2.kicad_prl`, `kibot_4e6917n2.kicad_dru`. At the end, all those temporary files are removed, except for the last one: `kibot_4e6917n2.kicad_dru`. 

**To Reproduce**
Add a file `PROJECT_NAME.kicad_dru` in the source folder, with a simple content like 
```
(version 1)
```
Run the KiBot command. Observe that temporary files are created and all but `.kicad_dru` are removed.

**Expected behavior**
After completion of the kibot CLI command, the temporary `.kicad_dru` file is removed.

**Environment (please complete the following information):**
KiBot CLI, KiCad 8.0.4, Ubuntu 24.04 LTS

**Additional context**
Ran into this after using the `panelize` output. It creates a `.kicad_dru` file. After running the kibot command on the panel, the temporary file is created and never removed.

set-soft, 2024-08-27:
Hi @memen45 !
We need a simple case where it happens.
KiBot doesn't create temporal files for fun, they are created when needed.
The code should remove DRU files, as PRL files, so I need an example that fails to do it.

memen45, 2024-08-27:
Happens during `step` output only. 
Tested with a `test.kibot.yaml` with the following contents:
```
kibot:
  version: 1

outputs:
  - name: step
    comment: "Renders - 3D model (STEP)"
    type: step
    options:
      metric_units: true
      origin: 85,66.2
```

`kibot -b my_kicad_file.kicad_pcb -c .kibot/test.kibot.yaml --out-dir export/test`

Do you need a complete zip with a filled directory or is this specific enough?

Cimos, 2024-08-27:
I have seen the same behaviour but hadn't gotten around to submitting a ticket.

set-soft, 2024-08-28:
Hi @memen45 and @Cimos !
The above patch should fix the problem. Is available in the *dev* branch and docker images

commit cd92d88e (2024-08-28): [3D Outputs][Fixed] Temporal .kicad_dru file not removed

Fixes #655
closed 2024-08-28 by set-soft