INTI-CMNB/KiBot #656: [BUG] Output type bom on panelized board missing column values

memen45, 2024-08-28:
**Describe the bug**
Using the output type `bom` with additional field LCSC on a panelized board results in unset values for LCSC in the final bom.csv.

**To Reproduce**
The board is panelized with the `"renameref": "{orig}_{n}"` option for kikit, so all footprints have their board number appended to their original ref (without this option, the BOM only contains the few components from the original single schematic, which might be another bug?).

The `test.kibot.yaml` contains the following:
```yaml
kibot:
  version: 1

outputs:
  - name: bom
    comment: "BOM"
    type: bom
    dir: PCB
    options:
      output: "bom.%x"
      ref_separator: ","
      columns:
        - field: References
          name: Designator
        - Footprint
        - field: "Quantity Per PCB"
          name: Quantity
        - Value
        - field: "LCSC"
          name: "LCSC Part #"
      csv:
        hide_pcb_info: true
        hide_stats_info: true
        quote_all: true
```

`kibot -b export/panel/project-panel.kicad_pcb -c .kibot/test.kibot.yaml --out-dir export` gives the following output
```
Using SCH file: project.sch
- 'BOM' (bom) [bom]
WARNING:(W009) KiCad config without environment.vars section (kibot - config.py:359)
WARNING:(W013) `C1_15` component in board, but not in schematic (kibot - kiplot.py:345)
<...>
```
A warning for each of the components in the pcb is given and omitted here. The resulting `bom.csv` is a union of the components from the schematic (with their original designators/refs, e.g. C1, R1, D1) and the components from the panelized PCB (with their renamed refs like C1_1, R1_1, etc...):
```csv
"Designator","Footprint","Quantity","Value","LCSC Part #"
"R1","R_0402_1005Metric","1","200R","C25087"
"R1_6,R1_26,R1_15,R1_19,<etc...>","R_0402_1005Metric","54","200R",""
```
- Components from the schematic have the LCSC value correctly set in the `bom.csv`
- Components from the PCB have their LCSC value not set in the `bom.csv`
- Opening the `project-panel.kicad_pcb` in kicad and checking the footprint properties shows LCSC property is correctly set

**Expected behavior**
- Components from the PCB have their LCSC value correctly set in the `bom.csv`
```csv
"Designator","Footprint","Quantity","Value","LCSC Part #"
"R1","R_0402_1005Metric","1","200R","C25087"
"R1_6,R1_26,R1_15,R1_19,<etc...>","R_0402_1005Metric","54","200R","C25087"
```

**Environment (please complete the following information):**
Kicad 8.0.4 on Ubuntu 24.04

set-soft, 2024-08-29:
Hi @memen45 !
I changed the subject because this is not a bug, the way KiKit changes the references is really particular and far from the standard PREFIX*NUMBER*. 

> A warning for each of the components in the pcb is given and omitted here. 

Which is telling you that something is wrong.

> The resulting bom.csv is a union of the components from the schematic (with their original designators/refs, e.g. C1, R1, D1) and the components from the panelized PCB (with their renamed refs like C1_1, R1_1, etc...):

Which is the expected behavior.

I assume that what you want is a BoM to send to JLCPB so they can assembly your board, right?
Because this is not the real BoM, is a BoM forced so somebody can pretend the panel is just one board.
The real BoM doesn't have to mess with the panel, it just declares a number of boards that is a multiple of the boards in the panel.

> Components from the PCB have their LCSC value not set in the bom.csv

Do they have the LCSC field in the PCB file?

Please create a very simple example, just a couple of components and a panel with 2 or 4 boards, so we can explore the situation.

memen45, 2024-08-29:
I filed it as a bug, as I would expect that the additional fields (in the example `LCSC`, but also variant value fields like `VARIANT1:value` are not taken into account) that are correctly set in the `.kicad_pcb` would also be set in the BOM. This is not the case. I isolated this particular issue, but there seem to be more highly related issues when trying slightly different settings.

If we change to feature request, then maybe it is useful to also discuss the other options and their issues.

## Panelization
There are two ways to set the refs for the panel components. 
- keep the original refs, so you get n copies with ref R1, C1, etc.
- enable rename refs option, so you get unique refs like R1_1 to C1_n.
In both cases, all component fields are copied correctly to the `project-panel.kicad_pcb` (LCSC field is set, VARIANT1:value is set).

In the end, we need two outputs, the BOM.csv and position.csv. Depending on the above panelization, you get different incomplete/incorrect files. I tried to read through the corresponding Python code (out_bom and out_position files), but I am not that familiar with the kicad API, so could not pinpoint to a loc.

|           | Bom | Position |
| -------- | -------- | --- |
| R1_n | missing fields, variant value not honored |  |
| R1 | quantity=1 (no components from pcb) | Rotation same for all components with same ref, footprint x,y offset |

I will try to make a minimal project showcasing these issues.

memen45, 2024-08-29:
Attached the minimal example kicad project with the previously described behaviour. There are three steps needed to obtain the exports, which are all described in the included `readme.md`. Hope this helps and let me know if you have any questions.

[kibottest.zip](https://github.com/user-attachments/files/16797837/kibottest.zip)

memen45, 2024-08-29:
Apparently kicad archive project does not include everything from the folder. Attached the complete folder including readme and config files.

[kibottest.zip](https://github.com/user-attachments/files/16803015/kibottest.zip)

set-soft, 2024-08-30:
Hi @memen45 !
I found the issue, the above patch should solve it.

Now a couple of things:
1. I 'll suggest using a filter to exclude the schematic parts, like this:

```yaml
filters:
  - name: only_jlc_parts
    comment: "Only parts with JLC code"
    type: generic
    include_only:
      - column: "LCSC"
        regex: '^C\d+'

  - name: only_panel_parts
    comment: 'Only parts from a panel created using "renameref": "{orig}_{n}"'
    type: generic
    include_only:
      - column: "Reference"
        regex: '^\w+\d+_\d+'
```
and then:
```yaml
      exclude_filter: ['only_jlc_parts', 'only_panel_parts']
```

2. I don't fully understand why do you really need to use renameref. This damages the references. Is that some issue with JLCPCB? Why you can't just say you need 6x9 times C1? Isn't this supported by JLCPCB? After all you'll be specifying 6x9 positions for C1's

closed 2024-08-30 by set-soft

memen45, 2024-08-30:
Awesome! Does it also fix the `VARIANT1:value` field when run with a variant?

With regards to your second point: I tried that as well, but then the rotation for all the components is the same in the `position.csv` file. It seems like it is using the same C1 rotation for all C1's in the position file, even though during the panelization half the boards are rotated 180 degrees.

set-soft, 2024-08-30:
> Awesome! Does it also fix the `VARIANT1:value` field when run with a variant?

Should fix it.

> With regards to your second point: I tried that as well, but then the rotation for all the components is the same in the `position.csv` file. It seems like it is using the same C1 rotation for all [...]

The above patch adds support for repeated references to the *position* output. Works for a reduced version of your example (2x2), which I added as a test.

commit 0db48148 (2024-08-30): [BoM][Fixed] User fields for PCB only not filled

They were collected and available, just not displayed in the BoM

Fixes #656