INTI-CMNB/KiBot #657: [BUG]PCB print fail if WKS sheet is in Kicad 8 format with a PNG in it

MarkusSBS, 2024-09-10:
Fill as much as you can, remove the rest

**Describe the bug**
If i try to run the CI pipeline with a WKS template sheet migrated with kicad 8 it fails with Unknown bitmap attribute. if i save the WKS with the new logo in an old Kicad 6 it works 

On the workstation boath WKS render ok in Kicad.

**To Reproduce**
run Kibot with a kicad 8 wks 

Old format:

  (bitmap (name "") (pos 145.194 14.5872) (scale 1)
  (pngdata
    (data "89 50 4E 47 0D 0A 1A 0A 00 00 00 0D 49 48 44 52 00 00 01 81 00 00 01 3A 08 06 00 00 00 F0 54 EE ")
    (data "92 00 00 00 04 73 42 49 54 08 08 08 08 7C 08 64 88 00 00 20 00 49 44 41 54 78 9C EC BD 77 94 1D ")
    (data "C7 7D E7 FB F9 F5 0D 73 27 E7 C1 20 A7 41 06 08 10 00 91 08 E6 20 92 A2 44 59 C9 71 2D 4B F6 DA ")

New format:

	(bitmap
		(name "")
		(pos 148.289 20.2222)
		(scale 0.188)
		(data "iVBORw0KGgoAAAANSUhEUgAABH4AAAI8CAYAAACH9WxEAAAAAXNSR0IArs4c6QAAAARnQU1BAACx"
			"jwv8YQUAAAAJcEhZcwAADsQAAA7EAZUrDhsAAE0MSURBVHhe7d0J3Hz9Pd//hMgtCYKIaBBFqSC1"
			"1JbEUlJBKUE0agu1NLWV/tFWY2klpLX8UVqlBNX+YytaTf+oopbYtYRaa0k0bRJbI/aln3d/OTUu"
			"n2vOXDPnnDlzrufz8Xg9wn3fv2tmrt/MmTmfOed77gYAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA"

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? `GitLab`

- Using a docker image? ghcr.io/inti-cmnb/kicad8_auto_full

**Additional context**
Add any other context about the problem here.

set-soft, 2024-09-10:
Hallo @MarkusSBS !
Are you using KiBot from the *dev* branch? Which tag of the docker image are you using?
It looks like a duplicate of #623

MarkusSBS, 2024-09-10:
ghcr.io/inti-cmnb/kicad8_auto_full

i will test the dev branch

MarkusSBS, 2024-09-10:
dev fixed it

closed 2024-09-10 by MarkusSBS