INTI-CMNB/KiBot #658: Integrate InteractiveHtmlBom

CameronBrooks11, 2024-09-11:
Integration of the InteractiveHtmlBom functionality either as part of docs index or perhaps on the top level of pcb index.

set-soft, 2024-09-12:
I'm not sure what you want. The *ibom* output implements InteractiveHtmlBom. 

In the web page generated by *navigate* you'll find it under **Schematic/BoM** and **PCB/fabrication/assembly**. This is the default *category*, which can be customized.

CameronBrooks11, 2024-09-12:
Thanks for the info...I was trying things about for the first time and did a skim of the source and must've missed that

set-soft, 2024-09-12:
You can find a full list of supported outputs [here](https://kibot.readthedocs.io/en/master/configuration/outputs.html#supported-outputs)

closed 2024-09-12 by set-soft

CameronBrooks11, 2024-09-12:
That’s great, thank you so much!