INTI-CMNB/KiBot #660: [BUG] Using 'var_rename' as 'pre_transform' for variants produces unopenable .kicad_pcb files with exceptionally long line lengths

willpuckett, 2024-09-13:
**Describe the bug**
I started working with variants this week in the hopes of implementing them into a CI/CD pipeline. I am working on a [flippable keyboard](https://github.com/willpuckett/Swoon) and I hope to have test pads that bridge solder jumpers in some variants so that boards assembled for a single side will not need to be hand soldered.

I created three variants, **flip** (the flippable variant for hand soldering), **left**, and **right** (the last two assembled by the board house with test pads to jump the solder jumpers). I ultimately found my way to the 'var_rename' filter, which I am using as a pre_transform on the **left** and **right** variants in swoon.kibot.yml. The **flip** variant has no 'var_rename' pre_transform.

Using `left:Footprint` and `right:Footprint` fields, I was hoped to be able to have the test pads I am using to bridge the jumpers set to a project footprint with a 0.001 mm pad for the flippable version, and then swapped to the 1mm test point footprint when the left or right variant is selected, as [demonstrated here](https://github.com/INTI-CMNB/kibot_variants_arduprog?tab=readme-ov-file#changing-values).

Unfortunately, when I run KiBot with [either the left or right variant selected](https://github.com/willpuckett/Swoon/actions/runs/10856041867), ie using 'var_rename', it produces a kicad_pcb file with exceptionally long line lengths which pcbnew cannot open, and then errors 

```
ERROR:Error loading PCB file. Corrupted? (kibot.gs - gs.py:834)
ERROR:Maximum line length exceeded (kibot.gs - gs.py:834)
```

With some targets, KiBot produced an additional warning shortly before erroring out:

```
WARNING:(W011) Missing default symbol library table (kibot - config.py:520)
```

These errors only occur when the 'var_rename' pre_transform is used. Everything works as expected without it in place, eg the **flip** variant, or **left** or **right** with 'var_rename' disabled.

Being that I am relatively new to KiBot, and exceptionally new to working with variants in it, I initially thought I was experiencing user error, and imagined that I somehow hadn't set an environment variable for my project library paths to resolve correctly. I tried copying them into the output directory, but the errors persisted.

I created a [var_rename demonstration repository](https://github.com/willpuckett/var_rename_test).  To my surprise, it [ran successfully](https://github.com/willpuckett/var_rename_test/actions) with both system and user footprints, indicating that I may have setup the var_rename VARIANT:FIELD columns correctly.

I then enabled additional logging in my kibot devcontainer with `-vvv`. It seems that KiBot successfully completes all the footprint replacements, and then has problems opening the pcb file in pcbnew.

```
... [abridged]
DEBUG:Replacing TP9 using TestPoint:TestPoint_Pad_1.0x1.0mm (kibot - pcb.py:112)
DEBUG:- Lib file /usr/share/kicad/footprints/TestPoint.pretty/TestPoint_Pad_1.0x1.0mm.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing TP5 using TestPoint:TestPoint_Pad_1.0x1.0mm (kibot - pcb.py:112)
DEBUG:- Lib file /usr/share/kicad/footprints/TestPoint.pretty/TestPoint_Pad_1.0x1.0mm.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing J62 using Swoon:3305-0-15-80-47-27-10-0 (kibot - pcb.py:112)
DEBUG:- Lib file /workspaces/Swoon/library/Swoon.pretty/3305-0-15-80-47-27-10-0.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing J29 using Swoon:3305-0-15-80-47-27-10-0 (kibot - pcb.py:112)
DEBUG:- Lib file /workspaces/Swoon/library/Swoon.pretty/3305-0-15-80-47-27-10-0.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing J54 using Swoon:3305-0-15-80-47-27-10-0 (kibot - pcb.py:112)
DEBUG:- Lib file /workspaces/Swoon/library/Swoon.pretty/3305-0-15-80-47-27-10-0.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing J14 using Swoon:3305-0-15-80-47-27-10-0 (kibot - pcb.py:112)
DEBUG:- Lib file /workspaces/Swoon/library/Swoon.pretty/3305-0-15-80-47-27-10-0.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing J21 using Swoon:3305-0-15-80-47-27-10-0 (kibot - pcb.py:112)
DEBUG:- Lib file /workspaces/Swoon/library/Swoon.pretty/3305-0-15-80-47-27-10-0.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing J17 using Swoon:3305-0-15-80-47-27-10-0 (kibot - pcb.py:112)
DEBUG:- Lib file /workspaces/Swoon/library/Swoon.pretty/3305-0-15-80-47-27-10-0.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing J73 using Swoon:0906-2-15-20-75-14-11-0 (kibot - pcb.py:112)
DEBUG:- Lib file /workspaces/Swoon/library/Swoon.pretty/0906-2-15-20-75-14-11-0.kicad_mod (kibot - pcb.py:124)
DEBUG:Replacing J53 using Swoon:3305-0-15-80-47-27-10-0 (kibot - pcb.py:112)
DEBUG:- Lib file /workspaces/Swoon/library/Swoon.pretty/3305-0-15-80-47-27-10-0.kicad_mod (kibot - pcb.py:124)
DEBUG:- Writing updated PCB to /tmp/kibot_1m6a6gw2.kicad_pcb (kibot - gs.py:973)
DEBUG:Copying project `/workspaces/Swoon/swoon.kicad_pro` to `/tmp/kibot_1m6a6gw2.kicad_pro` (kibot.gs - gs.py:527)
DEBUG:Copying project local settings `/workspaces/Swoon/swoon.kicad_prl` to `/tmp/kibot_1m6a6gw2.kicad_prl` (kibot.gs - gs.py:535)
DEBUG:- Loading the temporal PCB (kibot - pcb.py:94)
ERROR:Trace stack: (kibot.gs - gs.py:822)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 209, in load_board
    board = pcbnew.LoadBoard(pcb_file)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 9328, in LoadBoard
    return _pcbnew.LoadBoard(*args)
           ^^^^^^^^^^^^^^^^^^^^^^^^
ERROR:Error loading PCB file. Corrupted? (kibot.gs - gs.py:834)
ERROR:Maximum line length exceeded (kibot.gs - gs.py:834)
```

I copied the output `/tmp/kibot_1m6a6gw2.kicad_pcb` from /tmp and ran `awk '{print length}' erroring.kicad_pcb | sort -n | uniq -c` with the following results:

```

      1 1
      4 3
      1 5
      1 12
      2 14
   ... [abridged for relevancy]
      2 190
      1 191
      2 192
      2 193
      1 575
      1 809
      3 2414
      2 2415
      5 2416
      2 2756
      5 2757
      8 2758
      2 2760
      1 2761
   ... [abridged for relevancy]
      3 9382
      2 9383
      5 9384
      1 9464
      1 9563
      1 13794
      1 17546
      1 53804
      1 573407
      1 578219
      1 1126340
      1 1211504
      1 1218556
```

Hmmm... Three lines with over a million characters 🤔

So I ran the same command on the original kicad_pcb file with the following results:

```
      1 1
    604 2
   7202 3
... [abridged for relevancy]
      2 125
      1 230
     34 257
```

So nothing more than 257 characters.

I opened the [temporal kicad_pcb](https://github.com/willpuckett/Swoon/blob/main/erroring.kicad_pcb) in my editor and the text is readable/not garbled.

I also considered that perhaps somehow the 'var_rename' filter was negatively interacting with the 'fix_rotation' filter I'm using in conjunction with it, but disabling 'fix_rotation' still produces the errors.

**To Reproduce**
Command line options passed to KiBot, configuration files (.kibot.yaml), PCB/Schematic example to reproduce it (Ex: a link to a repo where the files are) https://github.com/willpuckett/Swoon

**Expected behavior**
A clear and concise description of what you expected to happen.
I was hoping that I could produce variants of a board with KiBot both in a devcontainer for testing and in a hosted CI/CD process. 

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`) ✅ Github
- In a GitHub action? Which tag are you using? (Ex `v1`, `v1_k6`, `v2`, `v2_d`, `v2_dk6`, `v2_k6`) ✅v2_8
- Using a docker image? (Ex: `kicad_auto:ki6`, `kicad_auto:latest`, `kicad_auto:dev`, `kicad_auto:dev_k6`, a custom one? can you provide the `Dockerfile`) ✅ using [this docker file](https://github.com/willpuckett/Swoon/blob/main/.devcontainer/Dockerfile)
- Using a desktop system? Which OS, distro and version? ❌

willpuckett, 2024-09-13:
While probably a separate issue, but also for further context, I was also surprised that 'var_rename' replaces rows in which there is no replacement text as well, as it seemed unlike the behavior [outlined here](https://github.com/INTI-CMNB/kibot_variants_arduprog?tab=readme-ov-file#changing-values).

For instance:

|Ref|Footprint|Config|left:Footprint|
|---|---|---|---|
|D1|D_0603|||

when run with variant left produces

|Ref|Footprint|Config|left:Footprint|
|---|---|---|---|
|D1||||

When I discovered this behavior, I copied and pasted all the relevant footprints into the appropriate columns, and examining the schematics KiBot generates with the 'sch_variant' output seems correct.

willpuckett, 2024-09-16:
A couple of further notes...

There are three types of components I am trying to remove from the left and right variants: 3305's, 0906's, and TestPoint Pads. 

- When I remove these three types of items completely from the schematic, all Kicad targets are able to complete successfully.
- When I try and replace ANY of the components with a different footprint using the variant:Footprint field, the targets involving pcbnew fail.
- When I try and replace ANY of the components with an empty value in the variant:Footprint field, all targets involving pcbnew fail.
- When I change the variant:Footprint field to variant:Value, all targets complete successfully, and values are updated in in the resulting schematic/pcb files.

Applying a semi-colon to the end of variant:Footprint field results in:

```
WARNING:(W011) Missing default symbol library table (kibot - config.py:520)
ERROR:In section 'TSP' (pcb_variant): Missing footprint `TestPoint_Pad_1.0x1.0mm;` in `TestPoint` lib (kibot.gs - gs.py:836)
```

Single or double quoting the replacement footprints in the variant:Footprint field results in:

```
WARNING:(W011) Missing default symbol library table (kibot - config.py:520)
ERROR:In section 'TSP' (pcb_variant): Unknown library `"Swoon` (kibot.gs - gs.py:836)
```

While the scope of this problem is, in this example, somewhat limited and it would be possible to replace footprints by hand, it is my hope to use this technique much more broadly in order to create various parts of a keyboard: top plate, bottom plate, and pcb. 

Reducing duplication by implementing variant based footprint replacement would be a boon for simplifying iterations.

set-soft, 2024-09-16:
> Unfortunately, when I run KiBot with [either the left or right variant selected](https://github.com/willpuckett/Swoon/actions/runs/10856041867), ie using 'var_rename', it produces a kicad_pcb file with exceptionally long line lengths which pcbnew cannot open, and then errors

This seems to be related to complex zone fills, now KiBot splits all s-lisp elements (similar to what KiCad 8 does), which fixes the problem.

> While probably a separate issue, but also for further context, I was also surprised that 'var_rename' replaces rows in which there is no replacement text as well, as it seemed unlike the behavior [...]

This is:
- Intended: you might want to use a variant to clear a field
- A fault on your side: you define this field for components that doesn't need it. I understand this is because you are using the KiCad dialog to do massive changes, but KiBot can't know it.

If you define the field only for the components that needs it the problem will go away. You explicitly asking to fill it with an empty value.

In order to test this patch change `v2_k8` to `v2_dk8`

closed 2024-09-16 by set-soft

willpuckett, 2024-09-21:
Thanks so much for the redirect. The variant gerbers outputs are running without error for me in v2_dk8. 

It seems that smd footprints placed on the back copper layer are being output on the front copper after swapping. This also seems to be happening on single sided paste layers.

Really appreciate getting this functionality into KiCad!

commit 4c2917e3 (2024-09-16): [SCH/PCB save][Fixed] Problems with huge constructs

- In particular zone fills that can be bigger than 1 MB
- KiCad seems to support lines of upto 1 MB
- Now we split everything, like KiCad 8 does
Fixes #660