INTI-CMNB/KiBot #661: [BUG] KiBot incorrectly requires a schematic when generating a position file from a PCB

chille, 2024-09-15:
KiBot 1.7.0 running in a docker container (ghcr.io/inti-cmnb/kicad8_auto_full:latest)

When trying to create a position file from a `*.kicad_pcb`, KiBot will give an error:

```
ERROR:No SCH file found (*.sch), use -e to specify one. (kibot.gs - gs.py:836)
```

This is weird, because:
1) All information that is required to create a position file is already in the `*.kicad_pcb` file.
2) It is possible to create a position file via the KiCad PCB Editor. (File -> Fabrication Outputs -> Component Placement)
3) If I uncomment line 465-466 from kiplot.py, then everything will work as expected.

This is a problem when working with a generated panel. The panelize output does not create a *.kicad_sch.

If I try to use -e option with the `*.kicad_sch` from the single board and the `*.kicad_pcb` from the panel then the position file will be created, but will only contain positions for one single instance of the PCB.

chille, 2024-09-15:
This is my workaround for the moment:

I have modified kiplot.py to disable loading of the schematic, by removing lines 465 and 466
```
sed -i -e '465,466d' kiplot.py
```

I replace the kiplot.py inside the Docker container via a volume mount:

```
docker run --rm -it -v /path/to/hacked_kiplot.py:/usr/lib/python3/dist-packages/kibot/kiplot.py:ro ...
```

This does solve the problem for me right now. I have no idea what other thing I broke by doing this. But so far my CI pipeline seems to work.

set-soft, 2024-09-16:
Hi @chille !

This is not true. Just try running KiBot with the following files: [661.zip](https://github.com/user-attachments/files/17012264/661.zip)
The archive contains a config file and a PCB, nothing more. And it works for 1.7.0 and 1.7.1.

And here comes the important detail: **you forgot to post an example**. I have no doubt you get this error, I don't think you are lying, but the configuration is very important. 

If you trigger the variants mechanism then you really need an schematic. Filters and variants needs it.
KiBot can work without an schematic, but only in particular cases.

closed 2024-09-16 by set-soft

chille, 2024-09-17:
I am not able to reproduce the error today. I removed my hacked kiploy.py, and everything seems to work. I have also spent a few hours trying to go back in the git history and trying different revisions for my CI setup. Everything seems to work.

However I just discovered that the error messages are a bit misleading sometimes. Usually when a warning or error occurs the message will be shown after the name and the comment from *.kibot.yaml have been printed to the console. But when doing something that will require a schematic, like exporting a BOM for example, the error message is shown directly without telling the user what step failed.

This makes me think that maybe I did get a position file generated, but had another error in my configuration and misinterpreted the error message to be related to my position file generation.

I might open a separate issue for this some day when I have time to produce a working example and some more information.