INTI-CMNB/KiBot #663: [BUG] Drill output FileNotFoundError error when renaming variants

willpuckett, 2024-09-21:
**Describe the bug**
When processing variants, the drill output attempts to rename a file, but the variant pcb filename is used instead of the original pcb filename. This file does not exist and results in the rename failing. 

```
Traceback (most recent call last):
  File "/usr/local/bin/kibot", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kibot/__main__.py", line 598, in main
    generate_outputs(args.target, args.invert_sel, args.skip_pre, args.cli_order, args.no_priority,
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 675, in generate_outputs
    _generate_outputs(targets, invert, skip_pre, cli_order, no_priority, dont_stop)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 665, in _generate_outputs
    run_output(out, dont_stop)
  File "/usr/local/lib/python3.11/dist-packages/kibot/kiplot.py", line 567, in run_output
    out.run(get_output_dir(out.dir, out))
  File "/usr/local/lib/python3.11/dist-packages/kibot/out_base.py", line 209, in run
    self.options.run(target)
  File "/usr/local/lib/python3.11/dist-packages/kibot/out_any_drill.py", line 183, in run
    os.replace(k_f, f)
FileNotFoundError: [Errno 2] No such file or directory: '/workspaces/Swoon/default/JLCPCB/swoon-PTH.drl' -> '/workspaces/Swoon/default/JLCPCB/swoon-PTH_FLIP.drl'
```

In the above example, the variant was named `kibot_1fhovjbl.kicad_pcb`, so the drill files should have tried to rename `/workspaces/Swoon/default/JLCPCB/kibot_1fhovjbl-PTH.drl' -> /workspaces/Swoon/default/JLCPCB/kibot_1fhovjbl-PTH_FLIP.drl'`

**To Reproduce**
Command line options passed to KiBot, configuration files (.kibot.yaml), PCB/Schematic example to reproduce it Run a drill job on a variant in v2_dk8

**Expected behavior**
KiBot would rename the file successfully without erroring

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`) ✅ Github https://github.com/willpuckett/swoon
- In a GitHub action? Which tag are you using? (Ex `v1`, `v1_k6`, `v2`, `v2_d`, `v2_dk6`, `v2_k6`) ✅ v2_dk8
- Using a docker image? (Ex: `kicad_auto:ki6`, `kicad_auto:latest`, `kicad_auto:dev`, `kicad_auto:dev_k6`, a custom one? can you provide the `Dockerfile`) ✅ ghcr.io/inti-cmnb/kicad8_auto_full:dev
- Using a desktop system? Which OS, distro and version? ❌

set-soft, 2024-09-24:
The above patch solves the problem

closed 2024-09-24 by set-soft

willpuckett, 2024-10-08:
Thanks so much!

commit 1018386b (2024-09-24): [Var Rename][Fixed] PCB name not always restored

Made the drill generation fail

Fixes #663
commit 8228bdd8 (2024-09-25): [Var Rename][Fixed] PCB name not always restored

Made the drill generation fail

Fixes #663