INTI-CMNB/KiBot #664: [BUG] Using var_rename to make footprint substitutions in v2_dk8 on back side results in pads, paste, and silkscreen being placed on front of board

willpuckett, 2024-09-21:
**Describe the bug**
When making variant footprint substitutions using the 'var_rename' filter on components placed on the back side of the board, substituted components are placed in front layers in Gerber outputs. This happens in copper, silk, and paste layers.

**To Reproduce**
Command line options passed to KiBot, configuration files (.kibot.yaml), PCB/Schematic example to reproduce it 
This [GitHub action run](https://github.com/willpuckett/Swoon/actions/runs/10967711128) demonstrates the case (the SWOON_RIGHT artifact has gerbers demonstrating the issue).

**Expected behavior**
When substituting footprints with 'var_rename', the substituted footprints and all their elements will be placed on the same side of the board as they footprint for which they are substituted.

**Environment (please complete the following information):**
Where are you running KiBot:
- In a CI/CD workflow? (Ex: `GitHub`/`GitLab`) ✅ GitHub 
- In a GitHub action? Which tag are you using? (Ex `v1`, `v1_k6`, `v2`, `v2_d`, `v2_dk6`, `v2_k6`) ✅ v2_dk8
- Using a docker image? (Ex: `kicad_auto:ki6`, `kicad_auto:latest`, `kicad_auto:dev`, `kicad_auto:dev_k6`, a custom one? can you provide the `Dockerfile`) ✅ ghcr.io/inti-cmnb/kicad8_auto_full:dev
- Using a desktop system? Which OS, distro and version? ❌

set-soft, 2024-09-24:
@willpuckett:

Which component is the problematic one?

set-soft, 2024-09-24:
Note that you are using a font that isn't included in your project.

set-soft, 2024-09-24:
The above patch should solve the problem, I can't check your case because I don't know what to look for.

closed 2024-09-24 by set-soft

willpuckett, 2024-10-08:
Apologies for the lack of clarity and delay. I just tested and footprint replacement seems to be working as expected in var_rename substitutions. Replaced footprints and silk are showing on correct sides in gerbers. Thanks again!

set-soft, 2024-10-08:
Hi @willpuckett !

Good to know. Note that replacing footprints is relative new. The original idea behind variants was to implement *assembly variants*, this is: 1 PCB for more than 1 products. Replacing footprint goes farther: 1 PCB file for more than one physical PCB (for even more "products").

So if you find other inconsistencies, just be patient and report them providing as much detail as possible.

willpuckett, 2024-10-08:
Thanks again Salvador. 

For context, these kinds of production variants I think will be _really_ useful for the keyboarding community. It's not uncommon for even the most basic keyboard to have 3 different pcbs: **the bottom plate** (just the outline with mounting holes), **the main pcb** (with components and switches), and **a top plate** (larger cutouts for the switches that stabilize them under the force of typing). Most projects create these as separate files, which then must **all** be changed for even small adjustments. Add to that that some boards have variants that add/remove an additional row of keys beside the pinky, or across the top of the board, and a "simple" project can have 6 or more separate pcb files. Your work making production variants functional has now made it possible to consolidate all these down to a single file to maintain. Not only will this accelerate development, but I think it will help creators feel less overwhelmed about updating projects since they only have to manage a single file now and can see everything in one place. 

I'm traveling over the next couple of weeks including heading to PCB West tomorrow where I look forward to sharing your successes with others. I'll try and implement this process in a couple of projects once I return and see if any further issues crop up. The largest thing I can foresee at this point is needing to be able to place traces in footprints so variants could switch between say two mcus of different dimensions, but that would need to start in KiCad itself before making its way over to KiBot. 

Can you help me understand the release schedule? When these features will be in the main k8 docker image and GitHub action?

set-soft, 2024-10-09:
> Can you help me understand the release schedule? When these features will be in the main k8 docker image and GitHub action?

There is no release schedule, releases are done when needed and when they are ready.

I know people wants to use a stable tag, but lamentably doing releases is time consuming. The *dev* images are well maintained and doesn't break often, so you can use them with confidence.

That said, KiBot will have various *maintenance* releases in the near future, just to fix small things. This is because I want to dedicate more time to the GUI.

commit cdbbc23f (2024-09-24): [Var Rename][Fixed] Flipped components replacement

We need to flip every single element, not just the layer for the
footprint.

Fixes #664
commit b4683ac9 (2024-09-25): [Var Rename][Fixed] Flipped components replacement

We need to flip every single element, not just the layer for the
footprint.

Fixes #664