INTI-CMNB/KiBot #669: [BUG] Unable to use custom template in schematic pdf

allardMVP, 2024-09-25:
Fill as much as you can, remove the rest

**Describe the bug**
I pull a custom git repo down with a custom schematic sheet template.  Putting the path into an environment variable in a previous step of the run.

**To Reproduce**

Actual error:

```ERROR:Unable to expand `MVP_TEMPLATES` in `${MVP_TEMPLATES}/MVPtemplate.kicad_wks` (kibot - log.py:179)```

The variable `MVP_TEMPLATES` is defined in a previous step of the workflow.

**Expected behavior**
Expand the environment variables correctly and find the schematic templates I cloned down in a previous step from a separate repo.

**Environment (please complete the following information):**
Running KiBot in github using the latest docker image and KiBot 1.8.0.

set-soft, 2024-09-25:
Hi @allardMVP 

Please provide more information. I see the words "previous step of the workflow" so I can just guess you are using KiBot in a CI/CD environment, but don't know which one and how you pretend to make MVP_TEMPLATES defined in one step to be available in the next.

If using GitHub just read the GitHub docs. You can define environment variables for a step using `env`. So I guess this has nothing to do with KiBot.

Also note that you can define environment variables from the KiBot configuration (environment option in the global options).

allardMVP, 2024-09-25:
@set-soft 

My bad I did not see the env option and for whatever reason defining variables as one would in linux (`export MVP_TEMPLATES=path/to/files`) does not work in CI/CD.

Got it going using the env option.  Thanks for the info.

closed 2024-09-25 by allardMVP

set-soft, 2024-09-26:
I think the problem is that every job runs on a separated virtual machine instance and each step in a separated shell instance. So jobs are fully isolated (and can run in parallel) and steps are quite isolated. You can communicate (pass stuff) between them, but needs work.

Using KiBot as a GitHub action is easy, but the most limited way to use it. Calling KiBot using shell commands in a workflow is a little bit more complicated, but much more flexible, and similar to the way you use it locally.