INTI-CMNB/KiBot #670: [HOW] How can I use the output of panelize with fab?

oliv3r, 2024-09-27:
So I've been very happy with kibot and generating fab outputs from my pcb files.

Now I have a project where I want to do panels (of the same pcb, multi-projects panels are far far out of scope and should be done manually :D)

As I'm integrating `panelize`, I came to the realization, that this will yield me a (kikit) generated panel as output, but the FAB plugin runs in parallel/with the regular PCB as input.

How can I also(or only) generate fab outputs for the panels?

set-soft, 2024-09-27:
This is a candidate for a FAQ ;-)
Currently you must do it in 2 steps:

1. Generate all stuff related to the real project and the panel.
2. Using the panel (and perhaps the original schematic, lamentably you don't get a real schematic for the panel) you generate any fabrication files related to the panel.

This is two separated runs of KiBot.

closed 2024-09-27 by set-soft

oliv3r, 2024-09-27:
> This is a candidate for a FAQ ;-) Currently you must do it in 2 steps:
>
>     1. Generate all stuff related to the real project and the panel.
>
>     2. Using the panel (and perhaps the original [...]

Yeah, was afraid of that. Multiple KiBot runs isn't a big problem, as the pipeline runs kibot a lot of times anyway (in my case). So once I have a penlized PCB file, how do I tell kibot to use that pcb file to generate the FAB stuff?

Hmm, I suppose normally we generate the FAB stuff by running kibot in folderX, and generate outputs in folder `outputs/panel` (for example). So invoking kibot on folder `outputs/panel` with only the FAB plugin solves that.

The only nasty thing is, the 'behavior' is now different, because up until now, every output is always generated from the same input source. And the output location of course depends on the configuration file of 'some' previous job.

Sadly, my current starting idea was to run kibot with a configuration file that does both in one go; so that obviously won't work at all, ever :)

But this is food for thought, thank you.