INTI-CMNB/KiBot #671: [BUG] ERROR:In section 'pdf_pcb_print' (pcb_print): `repeat_for_layer` specified, but nothing to repeat

oliv3r, 2024-09-27:
With the update to 1.8(.2) I now am getting the following error:

```
ERROR:In section 'pdf_pcb_print' (pcb_print): `repeat_for_layer` specified, but nothing to repeat (kibot.gs - gs.py:845)
```

The code leading to this error is:
```
        - sheet: '%ln layer'
          repeat_for_layer: 'F.Cu'
          repeat_layers: 'inners'
          layers:
          - layer: 'F.Cu'
```
([Full code here](https://gitlab.com/ci-includes/kici/-/blob/master/kibot_jobs/kici_pdf.kibot.yaml#L51)).

The PCB in question does not actually have any inner layers, but when discussing this in the past, you hinted me towards this solution, which is generic and should work in both cases, if you only have 2 layers, or more. With 1.8(.2?) a change was introduced, that breaks this behavior.

Are there alternative solutions? The goal ultimately is to have a single configuration file, that works regardless of how many layers are used (though 2 as a minimum is 'okay' I suppose).

set-soft, 2024-09-27:
You are right 1.8.0/1 are more strict.
I relaxed it again for 1.8.2, please try the *dev* code, it should just issue a warning.

closed 2024-09-27 by set-soft

oliv3r, 2024-09-27:
> You are right 1.8.0/1 are more strict. I relaxed it again for 1.8.2, please try the _dev_ code, it should just issue a warning.

Ahh, right. Well I was on v1.8.2, so should I have to wait for v1.8.3 to get a tagged release?

set-soft, 2024-09-27:
> > You are right 1.8.0/1 are more strict. I relaxed it again for 1.8.2, please try the _dev_ code, it should just issue a warning.
> 
> Ahh, right. Well thought I was on v1.8.2, but guess I got it wrong and was actually on v1.8.1 :), so should I have to wait for v1.8.2 to get a tagged release?

Use the *dev* branch or docker images, they contain the last changes.

> Since this is a regression from previous behavior, other then going back to v1.7.0; any plans to tag v1.8.2 anytime soon?

No particular plan to release 1.8.2 soon.

oliv3r, 2024-09-27:
> Use the _dev_ branch or docker images, they contain the last changes.

Yeah, I know I can; but I prefer using tagged images for (my) releases :)

> 
> > Since this is a regression from previous behavior, other then going back to v1.7.0; any plans to tag v1.8.2 anytime soon?
> 
> No particular plan to release 1.8.2 soon.

Hmm, Can I convince you to reconsider? Since a regression was introduced with 1.8.0, 1.8.2's patch/point release would be to fix the regression :)

set-soft, 2024-09-30:
> Yeah, I know I can; but I prefer using tagged images for (my) releases :)

I agree that once a repo is "finished" (no more important changes) the best is to switch to a tag. But for this you'll have to wait.

> Hmm, Can I convince you to reconsider? Since a regression was introduced with 1.8.0, 1.8.2's patch/point release would be to fix the regression :)

Each release takes a lot of time to ensure some consistency. I already released 1.8.1 as a bug fix release, releasing 1.8.2 for a small regression (that can be easily workaround) isn't worth the effort.

oliv3r, 2024-10-09:
> I agree that once a repo is "finished" (no more important changes) the best is to switch to a tag. But for this you'll have to wait.

I'll stick to 1.7.0 for now; which does everything alright, and I do have a work-around for a bug that was fixed in 1.8.x (EDRC failures halting the pipeline on the first error/warning).

> Each release takes a lot of time to ensure some consistency. I already released 1.8.1 as a bug fix release, releasing 1.8.2 for a small regression (that can be easily workaround) isn't worth the [...]

I understand and agree.

I do have a concern though, I was trying `-dev` for something else, and noticed that `repeat-layers` does print a warning:
```
WARNING:(W165) `repeat_for_layer` specified, but nothing to repeat (kibot - out_pcb_print.py:286)
ERROR:Warnings treated as errors (kibot - log.py:139)
```

Was that the change, to move it from nothing, to an error, to a warning?

If W165 is 'unique' to just this case, I can ignore it of course; but the reason I have this warning, is because I wanted a generic YAML for 2 or more layer boards. Your advise was back then, just add `repeat_for_layer`, which will 'ignore' any problems on 2 layer boards.

Maybe kibot should detect the number of layers somehow and adjust the error based thereon?

set-soft, 2024-10-09:
> Was that the change, to move it from nothing, to an error, to a warning?

Yes. 

> If W165 is 'unique' to just this case, I can ignore it of course; 

Yes

> but the reason I have this warning, is because I wanted a generic YAML for 2 or more layer boards. Your advise was back then, just add repeat_for_layer, which will 'ignore' any problems on 2 layer boards.

Yes, and this is the reason for a warning. From KiBot point is hard to know if you did it on purpose or just forgot something. So is best to say "Hey, is this ok?". As you already figured out the best is to reply "Yes, of course" and just mute the warning.

oliv3r, 2024-10-11:
well it isn't yes of course' as this is a template used by many different kinds of projects; but I still am okay with just accepting this as a given.

commit 82e1a0c5 (2024-09-27): [PCB Print][Fixed] `repeat_for_layer` with empty `repeat_layers`

- Is allowed again, this was the old behavior (i.e. 1.7.0)
- We issue a warning

Fixes #671