INTI-CMNB/KiBot #672: [FEATURE] Allow a different output location to be set for panelize:create_preview

oliv3r, 2024-09-27:
The panelize option for kikit works quite nice, and the `create_preview` option is pretty awesome.

However, when creating panels with the following config:
```
kibot:
  version: 1

import:
  - file: 'kici_common.kibot.yaml'

outputs:
  - name: 'panelize'
    run_by_default: @RUN_BY_DEFAULT@
    comment: 'Creates a panel to fabricate various copies of the PCB at once.'
    type: 'panelize'
    dir: 'outputs/panels'
    options:
      configs: @PANELIZE_CONFIG_JSON@
      create_preview: true

...
definitions:
  PANELIZE_CONFIG_JSON: 'kikit_panelize.json'
  RUN_BY_DEFAULT: true
```

the `dir` parameter nicely sets the output location of the kicad panel project, but the preview is not preferred to be stored there as well. Currently `create_preview` takes a boolean; maybe we can also accept a string to set a different dir location, could be relative to dir; e.g. `create_preview: '../png/'` for example.

set-soft, 2024-09-30:
Hi @oliv3r, did you try using `copy_files` output?
I won't allow a string there, this is nice for the YAML, but quite confusing for the GUI.

oliv3r, 2024-10-04:
Hmm, I know about copy files, but I'd want to 'move_files'; You do already have other situations where you allow `([boolean, string, list[strings]])` for different items?

set-soft, 2024-10-05:
> Hmm, I know about copy files, but I'd want to 'move_files'; 

So perhaps we need to add some option to do a move in the copy output ...

> You do already have other situations where you allow `([boolean, string, list[strings]])` for different items?

And you'll realize how annoying is this as soon as you try the (already available) GUI ...

oliv3r, 2024-10-09:
GUI? No lol I only use kibot in CI/CD :D so not aware of these 'problems'.

Alternativily, using `preview_output: /path` (or `output` as indented from `create_preview` which is also done for other options ...