INTI-CMNB/KiBot #673: [BUG] set_text_variables might not like non-1 return values (maybe a bug?)

oliv3r, 2024-09-28:
I'm playing with the panelize to fab workflow, which I'm getting there. Running kibot with panelization (and `baketext`!) to generate an intermediate `panel.kicad_pcb`. Now things get more tricky, probably because the way I'm doing things ;)

In the pipeline, I do not want to know 'where' I am, and actually, I want all of kibot to run 'just as normally' on my project. However, I will override the board, to `--board outputs/panels/new-panel.kicad_pcb`. Because of this, I no longer have a file that is being tracked in git, this would be fine, if I didn't have a general `set_text_variables` (which I fully realize is useless in this case, as this is a generate intermediate file).

So in summary:

```
/workdir/world$  kibot --defs-from-env --stop-on-warnings  --schematic world.kicad_sch --plot-config '.kibot/kici_panelize.kibot.yaml'
```
which generates my panel and then
```
kibot --defs-from-env --no-warn 135,164 --stop-on-warnings  --schematic world.kicad_sch --plot-config '.kibot/kici_fab.kibot.yaml' --board 'outputs/panels/world-panel.kicad_pcb'
```

> where it 135 (duplicate refs) and 164 (inconsistency on sch, which is odd, but could be related to multi-sheet schematics, it comlains about a missmatch on the U1 datasheet field, which doesn't happen for a regular run ...

will fail with the following `set_text_variables`

```
    - name: 'VCS_PCB_HASH'
      command: 'git log -1 --format="%H" "${KIBOT_PCB_NAME}" || echo "Unknown"'
      before: 'Git <'
      after: '>'
```

The strange thing is, if we wrap it in a sub-shell it works fine.

```
    - name: 'VCS_PCB_HASH'
      command: 'echo $(git log -1 --format="%H" "${KIBOT_PCB_NAME}" || echo "Unknown")'
      before: 'Git <'
      after: '>'
```

This is of course not an issue ("Unknown") because in panelize I have set `baketext` to ensure everything remains exactly the same as with a single PCB.

What's even weirder, is that
```
    - name: 'VCS_PCB_DATE'
      command: 'date -d @$(git log -1 --format="%at" "${KIBOT_PCB_NAME}") -Idate || echo "Unknown"'
```
does work (obviously, removing the `|| echo` at the end fails for different reasons, as is expecte.d

The only difference I can note, is that `date -d @bs` fails with error code `1`; but `git log ...` fails with 128. Though I don't get why this is even relevant, because any error is captured by the `|| echo` which shouldn't (ever) produce an error ...

So while I have an ugly work-around, this shouldn't be a problem of course.

set-soft, 2024-09-30:
Hi @oliv3r !

Run KiBot using `-L logfile` and paste here the portion related to the execution of the external command, it will contain more information.

My guess is that something else is failing and you are overlooking it.

oliv3r, 2024-10-04:
I will do so asap!

set-soft, 2024-10-23:
Should I close it @oliv3r ?

oliv3r, 2024-10-24:
I had forgot about this; let me try :)

oliv3r, 2024-10-24:
Yes, I figured it out, somewhere, someone is interpreting 'PCB time, and wants to change it to an iso format. While I could dig deeper, the work-around I have works and does produce the correct outputs.

```
WARNING:(W075) Trying to reformat PCB time, but not in ISO format (Unknown) (kibot.gs - gs.py:311)
```

Or, I'm not reproducing my error correctly :) in either case, there's an error message that helps me understand the problem.

closed 2024-10-24 by oliv3r