INTI-CMNB/KiBot #674: [FEATURE] Add PCB image to bom

oliv3r, 2024-09-29:
I'm using the BOM feature and quite like it. Because I was a bit lazy (lack of logo for now) I used a PCB image instead of the logo and must say I quite like it.

I do think there's still the need of a logo, and I'm not a UX person to make a pretty design, but rolling the Logo and log into the design is not a bad idea?

Currently it looks like this, and I did have to manually scale the image (though afaik kibot can do this already).

[image]

set-soft, 2024-09-30:
What about generating the image first and then the BoM, specifying as logo the image generated in a previous stage?

You can force the order of outputs in many ways (priority, declaration order, command line order).

In the worst case you can run KiBot once to get the image and then again to create the BoM.

oliv3r, 2024-10-04:
I wanted two images :) so one for the logo, e.g. the kibot robot or whatever, as you currently intended.

But then I also want to include the image of the PCB, which indeed I can get from the previous step, so I have it; I just can't specify two images in the bom's, both a logo and a pcb render.

set-soft, 2024-10-05:
And here things start to go forever ...: Side by side or one on top of the other? Which one on each side? Which relative size? And why not three images, or four?

So my recommendation is to compose the image outside KiBot. Use the `convert` command from Image Magick (included in the docker images) to compose the image. Is somehow complex, but you'll find plenty of examples, and you'll be able to compose the image as you like, add text, borders, etc. All can be done with commands.

oliv3r, 2024-10-09:
I agree that then we'll end up 10 images :p 

But I would argue that the logo is good to have; kind of like have 'company style' (like pre-printed paper with company logos') to make things nice, but the PCB image is nice, as it helps to visually identify the project the BOM is for.